SBFspot/SBFspot #236: Active: failed (Result: start-limit-hit) since Sat 2018-08-11 14:48:03 CEST; 26s ago

Thullie, 2018-08-11:
Hi,   i'm getting this error any idea how to fix it ?

● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Sat 2018-08-11 14:48:03 CEST; 26s ago
   Process: 1763 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=0/SUCCESS)
   Main PID: 1763 (code=exited, status=0/SUCCESS)

aug 11 14:48:03 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 11 14:48:03 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 5.
aug 11 14:48:03 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 11 14:48:03 ZONNUC systemd[1]: SBFspotUpload.service: Start request repeated too quickly.
aug 11 14:48:03 ZONNUC systemd[1]: SBFspotUpload.service: Failed with result 'start-limit-hit'.
aug 11 14:48:03 ZONNUC systemd[1]: Failed to start SBFspot Upload Daemon.

sbf-, 2018-08-11:
Can you post your SBFspotUpload.service file?
What is the output of `journalctl -u SBFspotUpload`

Thullie, 2018-08-15:
SBFspotUpload.service file

[Unit]
Description=SBFspot Upload Daemon
After=mysql.service mariadb.service network.target

[Service]
User=domoticz
Type=simple
TimeoutStopSec=10
ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon
Restart=on-success

[Install]
WantedBy=multi-user.target

Thullie, 2018-08-15:
journalctl -u SBFspotUpload

aug 15 11:37:06 ZONNUC systemd[1]: Started SBFspot Upload Daemon.
aug 15 11:37:06 ZONNUC SBFspotUploadDaemon[19888]: Exit signal received.
aug 15 11:37:07 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 15 11:37:07 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 1.
aug 15 11:37:07 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 15 11:37:07 ZONNUC systemd[1]: Started SBFspot Upload Daemon.
aug 15 11:37:07 ZONNUC SBFspotUploadDaemon[19891]: Exit signal received.
aug 15 11:37:08 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 15 11:37:08 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 2.
aug 15 11:37:08 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 15 11:37:08 ZONNUC systemd[1]: Started SBFspot Upload Daemon.
aug 15 11:37:08 ZONNUC SBFspotUploadDaemon[19894]: Exit signal received.
aug 15 11:37:09 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 15 11:37:09 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 3.
aug 15 11:37:09 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 15 11:37:09 ZONNUC systemd[1]: Started SBFspot Upload Daemon.
aug 15 11:37:09 ZONNUC SBFspotUploadDaemon[19897]: Exit signal received.
aug 15 11:37:09 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 15 11:37:09 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 4.
aug 15 11:37:09 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 15 11:37:09 ZONNUC systemd[1]: Started SBFspot Upload Daemon.
aug 15 11:37:09 ZONNUC SBFspotUploadDaemon[19900]: Exit signal received.
aug 15 11:37:10 ZONNUC systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
aug 15 11:37:10 ZONNUC systemd[1]: SBFspotUpload.service: Scheduled restart job, restart counter is at 5.
aug 15 11:37:10 ZONNUC systemd[1]: Stopped SBFspot Upload Daemon.
aug 15 11:37:10 ZONNUC systemd[1]: SBFspotUpload.service: Start request repeated too quickly.
aug 15 11:37:10 ZONNUC systemd[1]: SBFspotUpload.service: Failed with result 'start-limit-hit'.
aug 15 11:37:10 ZONNUC systemd[1]: Failed to start SBFspot Upload Daemon.

sbf-, 2018-08-15:
When you start the daemon manually (with user domoticz), does it run?

Thullie, 2018-08-17:
When i type domoticz@ZON:~/smadata$ systemctl start SBFspotUpload

i get 
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'SBFspotUpload.service'.
Authenticating as: domoticz,,, (domoticz)
Password:
==== AUTHENTICATION COMPLETE ===

Typed in my domoticz password and i see data on the pvoutput page

Thullie, 2018-08-21:
I have have reinstalled a new pi and i got the same error

DavidVM1982, 2018-08-28:
I have the same problem

sbf-, 2018-08-28:
What's in the SBFspotUpload logfile?

Thullie, 2018-08-31:
I have added RestartSec=10 and i see data on pvoutput :)

DavidVM1982, 2018-09-02:
I add RestartSec=10
now a get something else.

journalctl -u SBFspotUpload
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sun 2018-09-02 15:51:22 CEST. --
sep 02 12:21:25 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:26 raspberrypi SBFspotUploadDaemon[422]: Exit signal received.
sep 02 12:21:27 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:27 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi SBFspotUploadDaemon[448]: Exit signal received.
sep 02 12:21:27 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:27 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi SBFspotUploadDaemon[452]: Exit signal received.
sep 02 12:21:28 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:28 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi SBFspotUploadDaemon[455]: Exit signal received.
sep 02 12:21:28 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:28 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: State 'stop-sigterm' timed out. Killing.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Killing process 464 (SBFspotUploadDa) with signal SIGKILL.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'timeout'.
sep 02 15:50:20 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
lines 1-23...skipping...
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sun 2018-09-02 15:51:22 CEST. --
sep 02 12:21:25 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:26 raspberrypi SBFspotUploadDaemon[422]: Exit signal received.
sep 02 12:21:27 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:27 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi SBFspotUploadDaemon[448]: Exit signal received.
sep 02 12:21:27 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:27 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:27 raspberrypi SBFspotUploadDaemon[452]: Exit signal received.
sep 02 12:21:28 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:28 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi SBFspotUploadDaemon[455]: Exit signal received.
sep 02 12:21:28 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 12:21:28 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 12:21:28 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: State 'stop-sigterm' timed out. Killing.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Killing process 464 (SBFspotUploadDa) with signal SIGKILL.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
sep 02 12:21:38 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'timeout'.
sep 02 15:50:20 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 15:50:30 raspberrypi systemd[1]: SBFspotUpload.service: State 'stop-sigterm' timed out. Killing.
sep 02 15:50:30 raspberrypi systemd[1]: SBFspotUpload.service: Killing process 8523 (SBFspotUploadDa) with signal SIGKILL.
sep 02 15:50:30 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
sep 02 15:50:30 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'timeout'.
sep 02 15:50:36 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 15:50:37 raspberrypi SBFspotUploadDaemon[8550]: Exit signal received.
sep 02 15:50:47 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 15:50:47 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 15:50:47 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 15:50:47 raspberrypi SBFspotUploadDaemon[8584]: Exit signal received.
sep 02 15:50:57 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 15:50:57 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 15:50:57 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 02 15:50:57 raspberrypi SBFspotUploadDaemon[8633]: Exit signal received.
sep 02 15:51:07 raspberrypi systemd[1]: SBFspotUpload.service: Service hold-off time over, scheduling restart.
sep 02 15:51:07 raspberrypi systemd[1]: Stopped SBFspot Upload Daemon.
sep 02 15:51:07 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
[... cut ...]
[15:50:20] INFO: Starting Daemon...
[15:50:37] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[15:50:37] INFO: Starting Daemon...
[15:50:37] INFO: Stopping Daemon...
[15:50:47] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[15:50:47] INFO: Starting Daemon...
[15:50:47] INFO: Stopping Daemon...
[15:50:57] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[15:50:57] INFO: Starting Daemon...
[15:50:57] INFO: Stopping Daemon...
[15:51:07] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[15:51:07] INFO: Starting Daemon...

sbf-, 2018-09-04:
@DavidVM1982 When you disable the service and start the daemon manually with `/usr/local/bin/sbfspot.3/SBFspotUploadDaemon` does it work?

DavidVM1982, 2018-09-05:
it works. 
in the log file I get this:
SBFspotUpload log file
[19:39:45] ERROR: Unable to open output file. See syslog for details.
[19:39:55] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[19:39:55] ERROR: Unable to open output file. See syslog for details.
[19:40:05] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[19:40:05] ERROR: Unable to open output file. See syslog for details.
[19:40:15] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[19:40:15] ERROR: Unable to open output file. See syslog for details.
[19:40:26] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[19:40:26] ERROR: Unable to open output file. See syslog for details.
[19:40:31] INFO: Uploading datapoint: 20180905,19:40,24185472,12,,,38,231.19,18,24,0.108,0.111,173.26,216.11 => OK (200)
[19:40:36] INFO: Starting SBFspotUploadDaemon Version 1.1.0

sbf-, 2018-09-05:
OK, now I see...
systemd is only for daemon 2.0 (SBFspot 3.5) - see [documentation](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)

Why is it always restarting? Does it run several times? Crontab?
What is in syslog?

DavidVM1982, 2018-09-06:
I remove at /etc/rc.local
#Start SBFspotUploadDaemon
/usr/local/bin/sbfspot.3/SBFspotUploadDaemon 

sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Thu 2018-09-06 06:53:21 CEST; 2min 28s ago
  Process: 1075 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=0/SUCCESS)
 Main PID: 1075 (code=exited, status=0/SUCCESS)

sep 06 06:52:31 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
sep 06 06:53:21 raspberrypi systemd[1]: SBFspotUpload.service: State 'stop-sigterm' timed out. Killing.
sep 06 06:53:21 raspberrypi systemd[1]: SBFspotUpload.service: Killing process 1077 (SBFspotUploadDa) with signal SIGKILL.
sep 06 06:53:21 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
sep 06 06:53:21 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'timeout'.

at the log file
[06:52:31] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[06:52:31] INFO: Starting Daemon...

sbf-, 2018-09-06:
No, just the opposite: use `/etc/rc.local` to start the daemon and remove systemd
`sudo systemctl stop SBFspotUpload`
`sudo systemctl disable SBFspotUpload`
OR
Upgrade to SBFspot 3.5.0

DavidVM1982, 2018-09-08:
how do you upgrade to SBFspot 3.5.0

sbf-, 2018-11-26:
> how do you upgrade to SBFspot 3.5.0

Take a look at [sbfspot-config](https://github.com/SBFspot/sbfspot-config)

closed 2018-11-26 by sbf-