SBFspot/SBFspot #238: Task  schedule fails on Win10

lampra1, 2018-08-14:
I am trying to schedule a task on Win10 (unfortunately) task sceduler.
In the scheduler actions, i need to specify the program with the extension .exe.
The task fails. Trying to execute the program with the .exe extension in the command line results to path error.
The program executes fine without the .exe extension
I would appreciate any help
```
c:\Program Files\SBFspot.3>SBFspot.exe -v
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Windows (LE) 32 bit with SQLite support
Commandline Args: -v
Error! Could not open file C\?\C:\Program Files\SBFspot.3\SBFspot.cfg

c:\Program Files\SBFspot.3>SBFspot.exe -v -cfg"c:\Program Files\SBFspot.3\SBFspot.cfg"
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Windows (LE) 32 bit with SQLite support
Commandline Args: -v -cfgc:\Program Files\SBFspot.3\sbfspot.cfg
Reading config 'c:\Program Files\SBFspot.3\sbfspot.cfg'
Unable to locate or access the required datafile. Filespec: C\?\C:\Program Files\SBFspot.3\date_time_zonespec.csv
```

sbf-, 2018-08-15:
Is the file `date_time_zonespec.csv` located in `C:\Program Files\SBFspot.3`?

sbf-, 2018-08-15:
Must be some configuration issue on your PC.
Can you start another application, like notepad.exe (in C:\Windows)?

Is it Windows 10 - 32 or 64-bit?

lampra1, 2018-08-15:
This is Windows 10 32bit. To my knowledge there is no special configuration.
Yes I can start an application like notepad. I also included the SBFspot folder in the path (system & user)
It happens bot h in normal or elevated command line.
Any other scheduler suggestions? 
```
c:\Windows>notepad.exe
c:\Windows>cd "c:\Program Files\SBFspot.3"
c:\Program Files\SBFspot.3>SBFspot.exe -v
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Windows (LE) 32 bit with SQLite support
Commandline Args: -v
Error! Could not open file C\?\C:\Program Files\SBFspot.3\SBFspot.cfg

c:\Program Files\SBFspot.3>SBFspot -v
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Windows (LE) 32 bit with SQLite support
Commandline Args: -v
Reading config 'SBFspot.cfg'
Thu Aug 16 00:30:37 2018: INFO: Starting...
sunrise: 06:40
sunset : 20:21
Nothing to do... it's dark. Use -finq to force inquiry.
```

lampra1, 2018-08-16:
I always thought that windows is a waste of time and once again I feel the same.
Here is my workaround for the task scheduler.
The catch is that I need to use cmd to execute SBFspot, then I need to use quotes in the arguments but I should not use quotes in the working directory specification!! 
(I wasted my time for something that I would have finished in 1 minute on linux"
```
<Actions Context="Author">
    <Exec>
      <Command>C:\Windows\System32\cmd.exe</Command>
      <Arguments>/c "C:\Program Files\SBFspot.3\SBFspot"</Arguments>
      <WorkingDirectory>C:\Program Files\SBFspot.3\</WorkingDirectory>
    </Exec>
```

closed 2018-08-16 by lampra1