SBFspot/SBFspot #239: Add inverter temperature to sbfspot.log

mru-git, 2018-08-15:
Is there a simple way to add inverter temperature to sbfspot.log?

My "old" inverter died last week, after only 6 years. The new one gets quite hot, so I thought it might be a good idea to log the temperature.
I used a simple way to log the data into my home automation [www.pimatic.org](https://pimatic.org) by using their log reader plugin. Works fine for all the provided data, writes all the regularily updated values to the pimatic database. But: Is there a way to include inverter temperature? Sorry, if I'm asking a dumb question - no expert, unfortunately.
Any help highly appreciated!

sbf-, 2018-08-15:
It's already in the log...
Not all inverters report internal temperature, so maybe the value is always 0

mru-git, 2018-08-15:
The inverter STP 8000TL-20 reports temperature, the export to csv-files shows the value in the last column as soon as the inverter starts working. So I'm sure it's there. But it doesn't show up in my file sbfspot.log, which looks like this:

> EToday: 0.000kWh
> ETotal: 173.421kWh
> Operation Time: 83.64h
> Feed-In Time  : 82.83h
> String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
> String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
> Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
> Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
> Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
> Total Pac   :   0.000kW

Any hint how temp could be added here?
Sorry again for a possibly stupid question!

sbf-, 2018-08-16:
>But it doesn't show up in my file sbfspot.log

Please take a closer look, it's there
```
SUSyID: 78 - SN: 2100******
Device Status:      Ok
SUSyID: 78 - SN: 2100******
Device Temperature: 64.7..C
SUSyID: 78 - SN: 2100******
GridRelay Status:      Closed
```
Or are you using an old SBFspot?

mru-git, 2018-08-23:
Me stupid.
Still using 3.3.1
Sorry for bothering You.
Will update soon!

closed 2018-08-23 by mru-git