SBFspot/SBFspot #241: Not retrieving particular data

shadi1989, 2018-08-26:
Found in issue in my PVOutput data, a lot of days of june-2018 are missing.

Data uploaded to PVOutput:
01/06, 11/06, 27/06, 28/06, 29/06, 30/06

Data files in smadata directory:
01/06, 11/06, 27/06, 28/06, 29/06, 30/06

After doing a "/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad90" it created:
01/06, 11/06, 21/06, 22/06, 23/06, 24/06, 25/06, 26/06, 27/06, 28/06, 29/06, 30/06
It also skips the other days pretty fast.

I see this in the terminal, looks like it cannot create csv files from those dates.
```
* ArchiveDayData() *
********************
startTime = 5B2D7160 -> 23/06/2018 00:00:00
ExportDayDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 5B2C1FE0 -> 22/06/2018 00:00:00
ExportDayDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 5B2ACE60 -> 21/06/2018 00:00:00
ExportDayDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 5B297CE0 -> 20/06/2018 00:00:00
********************
* ArchiveDayData() *
********************
startTime = 5B282B60 -> 19/06/2018 00:00:00
********************
* ArchiveDayData() *
********************
startTime = 5B26D9E0 -> 18/06/2018 00:00:00
```

But still got 2 issues:
1. data is not uploaded to PVOutput. Guess that has something do to with the free-version only allowing to sync 14days of history data
2. why are there no csv files created? When i check Sunny Portal i can see the data.

sbf-, 2018-08-27:
Most inverters can hold 62 days of daydata (5-min historical data) so that's more or less until 21 june
The month data (totals per day) are kept for a much longer time, so using -am12 will get this data for 1 year.
At PVoutput you can upload them manually.

TerrorSource, 2018-08-30:
@sbf- thanks for the heads up.
I've already exported daily results and imported hem in PVOutput.

closed 2018-08-30 by TerrorSource