SBFspot/SBFspot #242: SBF no longer uploading

jonk999, 2018-08-31:
I have had SBFspot uploading to PVOutput without issue for quite some time running in a Ubuntu VM on a NUC.
On 24/08 I had a power failure during the night and the NUC shutdown. Since coming back online, data has not been uploading to PVOutput.

Looking at the database, there is current data in vwSpotData, however the latest data in vwPvoData, etc is the last reading from the inverter before the power failure.

I am running I believe V3.3.1 of SBFspot using SQLite.

Hoping someone may be able to assist with getting the uploads going again.
I have tried running /usr/local/bin/sbfspot.3/SBFspotUploadDaemon manually to make sure the daemon is running, but still nothing is being uploaded. So, I believe it is probably an issue with the database?

Cheers.

sbf-, 2018-08-31:
and what is the output of
`SBFspot -v5`

jonk999, 2018-09-02:
I think I found the issue. When I logged into the inverter using Sunny Explorer data then started uploading again. So, I'm guessing the inverter lost it's internal time somehow during the power outage - even though it happened at night.
Is there any easy way to get the data for the days that were missed? I now have a 3-4 day gap in pvoutput.

jonk999, 2018-09-03:
That did the trick. Thanks very much for the assistance.
I will be sure to check the time on the inverter when I next lose power.

closed 2018-09-04 by sbf-