SBFspot/SBFspot #245: Any (other) offline visualization tools for SBFspot?

philipptrenz, 2018-09-07:
Hi guys, many thanks for your excellent work! 

I was searching for a offline visualization tool and found https://github.com/KiOui/SunPortal
Based on the work of @KiOui I'm working on more features [here](https://github.com/philipptrenz/sunportal/). I wanted to share this and also ask if there are any other offline visualization tools for @SBFspot I've overlooked?

mincequi, 2021-02-16:
I am working on an App (iOS, Android) to support visualization of SBFspot data.
Initial version is working and displays live data only (historic data visualization is in the working).
See screenshot and let me know, if someone is interested.

[image]

aenne23769, 2021-03-02:
Very interested

sneumeister, 2021-03-07:
For visualization I use 123solar:
https://www.123solar.org/

Runs on same Pi 2B as SBFSpot.
Just modified to feed data from SBFSpot sqlite-database (not via "-123s" option).

closed 2021-05-20 by sbf-