SBFspot/SBFspot #248: Question: Any interest in a fork with MQTT support?

PjotrekSE, 2018-09-25:
I have made a version with mqtt support for myself during winter & spring. If there is a general interest, I can use this winter to generalise it into a fork with this code. Pls let me know if you want this functionality for general use. 
As my *Sunny Tripower STP 6000TL-20* **totally** lacks Bluetooth (See enclosed pics) I also made a header file that lets me link without Bluetooth, I might do such a fork as well if desirable.

Best regards, and thanx for all your efforts.
Peter Kling, Sweden (PjotrekSE)
System Specialist Emeritus

[image]
[image]

webalexeu, 2018-09-25:
Hello,

I'm interested to have MQTT support. Ideally, it would be great to be a module available in SBFspot and not a fork.

Is it possible to test your version? I've also an ethernet only model

Thanks,
Alex

sbf-, 2018-09-25:
Can you explain a bit more what's MQTT support? Until now I never had requests for it.
Compilation without bluetooth however, that sounds great.

PjotrekSE, 2018-09-25:
I quote from 'MQTT Essentials (https://www.hivemq.com/blog/mqtt-essentials-part-1-introducing-mqtt):
MQTT is a Client Server publish/subscribe messaging transport protocol...  
The publish/subscribe pattern (also known as pub/sub) provides an alternative to traditional client-server architecture. In the client-sever model, a client communicates directly with an endpoint.The pub/sub model decouples the client that sends a message (the publisher) from the client or clients that receive the messages (the subscribers). The publishers and subscribers never contact each other directly. In fact, they are not even aware that the other exists. The connection between them is handled by a third component (the broker). The job of the broker is to filter all incoming messages and distribute them correctly to subscribers.
The most important aspect of pub/sub is the decoupling of the publisher of the message from the recipient(subscriber). This decoupling has several dimensions:

Space decoupling: Publisher and subscriber do not need to know each other (for example, no exchange of IP address and port).

Time decoupling: Publisher and subscriber do not need to run at the same time.

Synchronization decoupling: Operations on both components do not need to be interrupted during publishing or receiving.

In summary, the pub/sub model removes direct communication between the publisher of the message and the recipient/subscriber. The filtering activity of the broker makes it possible to control which client/subscriber receives which message. The decoupling has three dimensions: space, time, and synchronization.
<end of quote>

I recommend you read part 2 of this article series, (https://www.hivemq.com/blog/mqtt-essentials-part2-publish-subscribe)

MQTT is very much used in the world of IoT (Internet of Things). In my setup, the MQTT enabled version of SBFspot (sbfmqtt) publishes the values from the inverter to an MQTT broker run on a Raspberry Pi. It can be the same RasPi that sbfmqtt is run on, but it does not have to be, and in my case those are two different RasPis.
Then, by the parking space of my car, I have a Sonoff ESP8266 controller connected by WiFi, that subscribes to this value, and when it receives a value higher than the set point, switches on the car charger, or, when lower, switches it off.

The advantages using MQTT for me is decoupling takers from givers, and enables me to use standard software for the takers and many of the givers. Also, if the charger client reconnects for any reason, the MQTT broker resends the last message. There are many IoT units that are able to listen to MQTT publications, in my case I use Tasmota (https://github.com/arendst/Sonoff-Tasmota). I can also use an intermediary software, NodeRed, to visualize and modify messages to and from the MQTT broker and/or clients, to make user interfaces and presentations. See enclosed pic. It was taken right now in the evening, so no values present, but you will get the idea. I will upload another, more interesting, tomorrow.

[image]

webalexeu, 2018-09-26:
Is sbfmqtt published already?

Cheers

PjotrekSE, 2018-09-26:
No, it is NOT published. It is up to **SBF** to decide if it is desirable. If you want it, you may tell him/her so. Even so, it will take some time. I sent you my work by private message yesterday.

NohWayJose, 2018-09-27:
Yes please.I’m a big fan of MQTT, as I work at IBM Hursley, where MQTT was written (and Node Red). MQTT was the next on my list of projects to try but happy to help. I’m not a programmer but I’m a UX/UI designer and spend my days gathering requirements and designing APIs, commands, graphical UI and so forth. Happy to help with that, if you like.Cheers,Greg

sbf-, 2018-09-29:
@PjotrekSE This looks really nice to switch your car's battery charger, but I wonder how you handle the charging when there's not enough sun. At a certain time the charger should be switched on to avoid you have to take your bike, no?

I'm still interrested how you disable the bluetooth code. You're talking about a header file, but I'm sure this is not enough ;-)
When I can implement this I can make a lot of people happy!

And finally, about MQTT, I think there are some people wanting this feature, so if you have enough time to create a PR, gor for it. 

Thanks anyway.

PjotrekSE, 2018-09-29:
**sbf-**: I will start the work with integration. If you saw my post about cross-compilation, maybe you noticed in the log pic that it was compiled *without* Bluetooth. Here goes again. 
However, as you mention, there are some issues we need to discuss;  I will send you an email, so we can discuss this. And, I'm now going on a sun vacation in Crete for a couple of weeks, but I expect to be bored pretty soon, and will bring my laptop along...

[image]

qurm, 2018-10-07:
Hi Peter, I'd also be interested in MQTT support.

I may be able to help with some testing and feedback...

PjotrekSE, 2018-10-09:
@qurm,@webalexeu,@NoWayJose,s.b.f-:
s.b.f- and I have just started a cooperation to integrate the mqtt functionality into SBFspot. So check back here now and then to find out about the proceedings.

FutureCow, 2018-10-23:
I'm also really interested in MQTT support. If you need any testing let me know!

DanielXYZ2000, 2018-11-03:
Hi, I'm also interested in MQTT support 
+1
Would be great to integrate with [Homeassistant ](https://www.home-assistant.io/) - in the best case including mqtt autodiscovery like implemented in [ESPhomelib MQTT autodiscovery](https://esphomelib.com/esphomeyaml/components/mqtt.html)
see also [Homeassistant MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/)

eterpstra, 2019-02-25:
I am also intereseted for this.

Can you give us an update? I would like to intergrade my old 1600 TL with Bluetooth in Home-assistant via MQTT if possible.

PjotrekSE, 2019-03-14:
Well, MQTT-fans, SBF does not answer to my suggested ways to proceed, so I guess we are on our own here. I have a working version, but as it is not merged with the development of SBFspot, it takes some programming and administration knowledge to use.

webalexeu, 2019-03-14:
Bad news but I'm ready to perform some test even if it requires time

FutureCow, 2019-03-17:
> Well, MQTT-fans, SBF does not answer to my suggested ways to proceed, so I guess we are on our own here. I have a working version, but as it is not merged with the development of SBFspot, it takes some programming and administration knowledge to use.
> Edited: ...and my solution is only for Linux, tested on Raspberry Pi.

I'm also happy to help and test the MQTT version!

sbf-, 2019-03-20:
Oh, I must have missed something...
I remember we had a short e-mail conversation... but since I really don't have a clue what and how you want to implement this, it's difficult to answer.
Why don't you fork my repository?

BarBaar44, 2019-03-27:
Would be awesome to have MQTT-support in SBF. Alternatieve would be to connect SBFspot to Node Red, and then transform to MQTT. Anybody done this already?

@sbf- I think @PjotrekSE explained extensively what MQTT is. It will only improve SBF, please support him!

verstrepeng, 2019-03-30:
@sbf- @PjotrekSE this would be a very nice add-on as MQTT seems to take lead as a major IoT telemetry protocol. There are a lot of tools available for automation and dashboarding.
So i hope you guys can figure it out !

PjotrekSE, 2019-03-30:
Presently in Crete. Will continue work wen back home 5/4

Pinksheepxx, 2019-03-31:
I am also interested. 
I could test MQTT implementation with Tripower 5000TL-20.

Jendemen, 2019-05-14:
Can you please give us an update regarding the status?
I am also very interested in MQTT support.

verstrepeng, 2019-06-12:
Any updates on MQTT support ?
Are the scripts somewhere ?

ced2git, 2019-06-22:
Hi @PjotrekSE, i would love to see support for MQTT in SBFspot so that i'm able to integrate it nicely within my home automation tool. 

Can you please share your code ? I'm running SBFspot on Rpi so it would fit. Best would be to have it inside the master branch of SBF

PjotrekSE, 2019-06-25:
> Hi @PjotrekSE, i would love to see support for MQTT in SBFspot so that i'm able to integrate it nicely within my home automation tool.
> 
> Can you please share your code ? I'm running SBFspot on Rpi so it would fit. Best would be to have it inside the master branch of SBF

@ced2git I might send you an executable for Raspi if you want to test.

ced2git, 2019-06-25:
Hi peter,

Well, if you don't mind, i rather prefer to have a non compiled version
that i will compile myself. Or you create a fork on github that i come and
grab.

Cedric

PjotrekSE, 2019-06-25:
Cedric, @ced2git <https://github.com/ced2git>
I understand your concerns, but I have presently nothing good for forking,
I need to rework it a lot to integrate with SBFspot, and now during summer,
living on a discontinued farm, there is precious little time for such
activities until autumn. Hope for your understanding.
Peter, ( peter xatx solvik ydoty se )

FutureCow, 2019-06-25:
Nice to see some action here, @PjotrekSE you can also share the code u used to compile the version you send. Other may help to get the code base inside sbfspot.

sbf-, 2019-07-29:
OK, here we go... 
I made a first version with MQTT support. It's still in development, so there is room for improvement.
I wrote a small [wiki](https://github.com/SBFspot/SBFspot/wiki/MQTT-(Development-branch)) to guide you a bit through the installation process.
Currently it's only for Linux (Raspberry Pi) but it works on Windows since all development is done with Visual Studio. I just don't publish the compiled executables yet.

For now, SBFspot uses an external program ([mosquito_pub](https://mosquitto.org/)) to publish the data. This is done to make my life easier. If I have some more time it might be possible it is replaced by pure API calls.

Hope you like it. Feel free to comment and give some ideas to improve it.

webalexeu, 2019-07-29:
MQTT: Publishing (sbfspot_305042202) Timestamp:29/07/2019 12:18:37;InvSerial:305042202;InvName:WebalexSMA;InvStatus:Ok;InvTemperature:83,760;InvGridRelay:Closed;EToday:23,066;ETotal:17872,490;PACTot:7906,000;UDC1:471,840;UDC2:471,060;IDC1:8,491;IDC2:8,547;PDC1:4005,000;PDC2:4026,000;
sh: 1: /usr/bin/mosquitto_pub -h test.mosquitto.org -t sbfspot_305042202 -m Timestamp:29/07/2019: not found
sh: 1: InvSerial:305042202: not found
sh: 1: InvName:WebalexSMA: not found
sh: 1: InvStatus:Ok: not found
sh: 1: InvTemperature:83,760: not found
sh: 1: InvGridRelay:Closed: not found
sh: 1: EToday:23,066: not found
sh: 1: ETotal:17872,490: not found
sh: 1: PACTot:7906,000: not found
sh: 1: UDC1:471,840: not found
sh: 1: UDC2:471,060: not found
sh: 1: IDC1:8,491: not found
sh: 1: IDC2:8,547: not found
sh: 1: PDC1:4005,000: not found
sh: 1: PDC2:4026,000: not found
sh: 1: : Permission denied
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto clients installed?
Error 32512 while publishing to MQTT Broker

Unable to publish to MQTT. mosquitto_pub is installed
Thank you for your work

sbf-, 2019-07-29:
It seems I didn't test good enough on RPi...
mqtt.cpp has been modified to fix it. Obviously you have to recompile.

FutureCow, 2019-07-29:
Big Thanks for this work, really great!

I got a error trying to compile SBFSpot:
`mqtt.cpp: In function ‘int mqtt_publish(const Config*, InverterData**)’:
mqtt.cpp:54:23: error: ‘mqtt_command_line’ was not declared in this scope
  boost::replace_first(mqtt_command_line, "{host}", cfg->mqtt_host);
                       ^~~~~~~~~~~~~~~~~
mqtt.cpp:54:23: note: suggested alternative: ‘math_errhandling’
  boost::replace_first(mqtt_command_line, "{host}", cfg->mqtt_host);
                       ^~~~~~~~~~~~~~~~~
                       math_errhandling
make: *** [makefile:116: sqlite/obj/mqtt.o] Error 1
`
Am i doing something wrong?

ced2git, 2019-07-29:
> 
> 
> It seems I didn't test good enough on RPi...
> mqtt.cpp has been modified to fix it. Obviously you have to recompile.

Hi, 

First, many thanks to bring this mqtt support. This will allow me to integrate SBFspot with my home automation software (in my case jeedom) into scenario's.

My comments: 

1) I could not compile on raspbian. I've changed #if defined (LINUX) --> #if defined (linux)
2) The message body is not recognised as a JSON output. I've tried to change the MQTT_MessageFormat to {"{key}":"{value}"} but the message passed to mosquito_sub is not well understood

i've got this error message: 

MQTT: Publishing (sbfspot) {"Timestamp":"29/07/2019 21:14:42"};{"InvSerial":"2120198970"};{"InvName":"SN: 2120198970"};{"InvStatus":"Ok"};{"InvTemperature":"0,000"};{"InvGridRelay":"Information not available"};{"EToday":"16,871"};{"ETotal":"20552,409"};{"PACTot":"0,000"};{"UDC1":"0,000"};{"UDC2":"0,000"};{"IDC1":"0,000"};{"IDC2":"0,000"};{"PDC1":"0,000"};{"PDC2":"0,000"};

Error: Unknown option '21:14:42};{InvSerial:2120198970};{InvName:SN:'.

Use 'mosquitto_pub --help' to see usage.
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto clients installed?

sbf-, 2019-07-29:
@ced2git I tested only on Raspbian (Buster)

sbf-, 2019-07-29:
@ced2git 

> I could not compile on raspbian. I've changed #if defined (LINUX) --> #if defined (linux)

Strange... it compiled on my pi. I'll have look at it

> The message body is not recognised as a JSON output. I've tried to change the MQTT_MessageFormat to {"{key}":"{value}"} but the message passed to mosquito_sub is not well understood

The double quotes might be a problem... I check if this is possible. Is that the format you want? It's not a correct JSON syntax IMO (Missing "[" and "]" characters)

ced2git, 2019-07-29:
Hi,

I have tried various format including “{key}”:”{value}”. Same error

 I need indeed a correct Json format in the payload with a comma to separate the various key-pair and the key and the pair in “”
Regards 

Envoyé d un smartphone...

FutureCow, 2019-07-29:
I agree, a real jSON would be perfect:

> {"Timestamp":"29/07/2019 22:45:17","InvSerial":"2130120959","InvName:SN":2130120959,"InvStatus":"Ok","InvTemperature":0.000,"InvGridRelay":"Information not available","EToday":15.462,"ETotal":17365.697,"PACTot":0.000,"UDC1":0.000,"UDC2":0.000,"IDC1":0.000,"IDC2":0.000,"PDC1":0.000,"PDC2":0.000}

sbf-, 2019-07-29:
For JSON alike syntax, use following settings:

    MQTT_MessageFormat={"{key}":"{value}"},
    MQTT_PublisherArgs=-h {host} -t {topic} -m '{message}'

Important: single quotes around {message}

FutureCow, 2019-07-29:
This is already a bit better, but json only expect a { and the beginning and a } and the end of the json. Also the decimal should be a dot and nog a comma (now you have to many comma's)

FutureCow, 2019-07-29:
You can also use separate topics for every value
sbfspot/InvGridRelay
sbfspot/ETopday
sbfspot/... etc

sbf-, 2019-07-29:
OK, I have some ideas ;-)
The dot/comma is controlled by DecimalPoint in cfg

    # DecimalPoint (comma/point default comma)
    DecimalPoint=point

FutureCow, 2019-07-29:
> OK, I have some ideas ;-)
> The dot/comma is controlled by DecimalPoint in cfg
> 
> ```
> # DecimalPoint (comma/point default comma)
> DecimalPoint=point
> ```

I know that cfg option, but i use the comma for all the rest of my automation...
So a DecimalPointMQTT=point would be a nice option....
Sorry for all the work!

PjotrekSE, 2019-07-29:
Congrats on the implementation, looks to become great!
One little wish: It would be nice to also get sunrise & sunset in the MQTT data.

sbf-, 2019-07-30:
@PjotrekSE 
> It would be nice to also get sunrise & sunset in the MQTT data.

Good idea! No problem.

I'm still interested in how you managed this: 
> I also made a header file that lets me link without Bluetooth, I might include this as well if desirable.

PjotrekSE, 2019-07-30:
> I'm still interested in how you managed this:
> 
> > I also made a header file that lets me link without Bluetooth, I might include this as well if desirable.
I will go back in my backups and have a check. As I remember, I first made a std version minus BT, maybe also w/o databases.  I'm currently repainting my house, but will find the time.

sbf-, 2019-08-01:
I published the latest changes to fix the JSON message format and while I was working on it I added XML too.

It's all explained in the sample SBFspot.cfg but her you have an overview:
Settings for JSON:
```
MQTT_ItemFormat="{key}": {value}
MQTT_ItemDelimiter=comma
MQTT_PublisherArgs=-h {host} -t {topic} -m "{{message}}"
```

Settings for XML:
```
MQTT_ItemFormat=<item name="{key}" value="{value}" />
MQTT_ItemDelimiter=none
MQTT_PublisherArgs=-h {host} -t {topic} -m "<mqtt_message>{message}</mqtt_message>"
```

It's clear you can create your own message format.

The decimal separator is hardcoded "dot" - I don't think it is necessary to have this configurable.

While testing I couldn't connect to `test.mosquitto.org` so I used `broker.hivemq.com` instead.

FutureCow, 2019-08-01:
Thank you again for this hard work!
My output:

> {

	"Timestamp": "01/08/2019 15:06:42",
	"SunRise": "01/08/2019 05:59:00",
	"SunSet": "01/08/2019 21:28:00",
	"InvSerial": 2130120959,
	"InvName": "SN: 2130120959",
	"InvStatus": "Ok",
	"InvTemperature": 50.000,
	"InvGridRelay": "Closed",
	"EToday": 5.616,
	"ETotal": 17394.355,
	"PACTot": 2108.000,
	"UDC1": 164.900,
	"UDC2": 165.210,
	"IDC1": 6.404,
	"IDC2": 6.772,
	"PDC1": 1055.000,
	"PDC2": 1118.000,
	"UAC1": 238.130,
	"IAC1": 8.855,
	"gridfreq": 49.990,
	"opertm": 27158.296,
	"feedtm": 25603.546
}

Valid JSON! data drops into my homeautomation database.

webalexeu, 2019-08-01:
Working well for me as well.
Is it possible to run it as a service that will publish the data in realtime?
To avoid run the script every second

Thanks again for the work

FutureCow, 2019-08-02:
Just one "error" for me. I also use WebSolarLog, so i run SBFSpot with -wsl and -mqtt. But with -mqtt i also get this output(without -v...):
mqtt_command_line=/usr/bin/mosquitto_pub -h {host} -t {topic} -m "{{message}}"
mqtt_command_line=/usr/bin/mosquitto_pub -h {host} -t {topic} -m '{{message}}'
this might be some debugging info, can you put this in -v2-5 or something. Now WebSolarLog isn't working.

sbf-, 2019-08-02:
Yes indeed, it was debugging info. I removed it.
Is WSL still supported? Long time I heard of it.

sbf-, 2019-08-02:
> I also made a donation for a nice Belgian beer, or 2 :)

thanks!

ced2git, 2019-08-03:
Hi,

It works great. :)  The JSON is well recognised but my setup is made of 3
Sunnyboy (i've the option MIS_Enabled=1).  SBFspot publish 3 times on the
same topic (sbfspot in my case)... and my home automation will consider it
as being the same device from which he is getting three times an update at
the same time.

Instead of publishing 3 times on the same topic, a solution for Multi
Inverter setup would be to publish each inverter under a subtopic like
/sbfspot/{serial}.

Cedric

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ced2git, 2019-08-03:
Something like this in mqtt.cpp (i'm not a dev ... sorry)

if (cfg->MIS_Enabled){
std::string mqtt_topic_mis=cfg->mqtt_topic + "\" +
inverters[inv]->DeviceName;
boost::replace_first(mqtt_command_line, "{topic}", mqtt_topic_mis);

else
{
boost::replace_first(mqtt_command_line, "{topic}", cfg->mqtt_topic);
}

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

sbf-, 2019-08-03:
Have a look at the latest example config. There the serial is used in the topic.

ced2git, 2019-08-03:
Yes indeed. It works perfectly.

Regards

eterpstra, 2019-09-24:
Hi,

Do i have to re-install / compile the whole installation again for MQTT support ?

sbf-, 2019-09-24:
> Do i have to re-install / compile the whole installation again for MQTT support ?

Yes, you have to download [dev_mqtt branch](https://github.com/SBFspot/SBFspot/tree/dev_mqtt) and recompile SBFspot

HABuild, 2019-10-20:
Awesome thanks very much.
I am leaving this comment here to help future people.

I have a funny setup and it was fiddly to install following the old windows install docs. I eventually got sbfspot intalled and working with sqlite database.

my setup*

1. my SMA sunny boy 5000TL-20 is bluetooth only
2. I currently connect via Win 10 machine and bluetooth _(the upload daemon isn't working, no log created)_
3. _command line_    > C:\Program Files (x86)\SBFspot.3\SBFspot -v -mqtt
works and exports json to hassio embedded broker
4. Hassio embedded broker is on a separate rPI HassOS 

**probably pretty simple to those with knowledge, but I will detail for new users.** 
To get mqtt to work I had to install mosquitto for windows for 

> C:\program files\mosquitto\mosquitto_pub.exe

this has to be reflected in the SBFspot.cfg  along with,

MQTT_PublisherArgs=-h {host} **-u** **_your_broker_username_here_** **-P** **_your_broker_pw_here_** -t {topic} -m "{{message}}"

you need to add -u lowercase and -P uppercase arguments into your .conf file to allow remote broker authorization, along with broker IP of course.

_now I just need to figure out how to run SBFspot as a timely service and how to display my JSON data in hassio_

BarBaar44, 2019-11-01:
Any luck already in integrating in hassio @HABuild ? Curious on that, since I a at the same point.

For automation, check the wiki, as referenced earlier in the thread,,

HABuild, 2019-11-02:
@BarBaar44 Hiya, yeah I got working overall. SBFspot is via my desktop, so not in hassio directly, but I just pull the mqtt replies into hassio via json_attributes_topic, I have a batch file that runs (SBFspot.exe -finq -mqtt -nocsv) every 5 mins to update the data (which sbfspot makes easy to only run during sunrise-sunset). I also have single button on lovelace if I want to push the command, that works via iotlink script>another.batch. The pvoutput stuff runs concurrently as well. 

`sensor:
  - platform: mqtt
    name: "SMA Output"
    state_topic: "sbfspot"
    value_template: '{{ value_json.status }}'
    json_attributes_topic: "sbfspot"
    
  - platform: mqtt
    name: "SMA Sunny Boy"
    state_topic: "sbfspot"
    value_template: '{{ value_json.InvName }}'
    icon: "mdi:flash"`

[12 more replies]

closed 2020-05-25 by sbf-