SBFspot/SBFspot #249: make does not work on my Raspberry Pi

wil76, 2018-09-25:
I'm trying to setup SBFspot on my raspberry Pi 2.
but nothing happens after this:

[image]

what can I do ?

PjotrekSE, 2018-09-25:
What happens? Does it just hang there? 
What operating system? 
Can you open other terminals to the RasPi, or does the entire RasPi hang?
-When I compiled on older RasPis, sometimes it took a long time compiling some of the sources, including ArchData. Now  I always compile on a new RasPi 3 B+, it is *much* quicker.

wil76, 2018-09-26:
I'm not confirm with compiler on Raspberry.
but obviously "makefile" starts all compiles in parallel. 
this of course does stress a little raspberry strongly.
is there an option to serialize the compile steps.

sbf-, 2018-09-27:
AFAIK there is no parallel compilation...
Anyway, today I released [sbfspot-config](https://github.com/SBFspot/sbfspot-config)
It comes with precompiled binaries for Raspberry Pi running stretch. No need to install development tools and libraries, just run this command and give it a try:
`curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash`

wil76, 2018-09-27:
Greta job, this will make the setup much easier.

my environment is running now.
but I had to compile the objects one after each other.

closed 2018-09-27 by wil76

wil76, 2018-10-06:
It obviously hangs. I have to cancel the putty session. But than the Raspi is still running. I‘m compiling on a Raspi pi 2. I‘m using a Pi 3 B+ for production and don’t want to install tens of packages only for compiling SBFspot.Is it possible to get a compiled version. I‘m fighting since month with a python module to read the inverter data via Bluetooth. But this does not support more than one SMA component. Are there other compile options to further evaluate. I already tried -v. But I‘m not able to investigate the problem
		
		

		Outlook für iOS beziehen

https://github.com/SBFspot/SBFspot/issues/249#issuecomment-424470726

sbf-, 2018-10-06:
See my comment above about sbfspot-config