SBFspot/SBFspot #252: Failed Connection Troubleshoot

danbates2, 2018-10-03:
On one setup I've had no success with the 'identification via boardcast' function.
By setting IP address to 0.0.0.0 for example, no luck.
Setting a fixed IP sometimes works, but for some reason the IP is changing.
Any suggestions?

sbf-, 2018-10-04:
Different inverters and/or different firmware?

I found this python script to discover speedwire devices. Can you test it? (Just change `localInterface` variable to your host's IP address)
Maybe it can help to improve SBFspots discovery method?

``` python
#!/usr/bin/python

#sudo apt-get install python-twisted

#
# Example to perform a speedwire discovery using python
#
from twisted.internet.protocol import DatagramProtocol
from twisted.internet import reactor
from twisted.application.internet import MulticastServer

# Change this address to the address of your local network interface
localInterface = '192.168.1.53'

# Nothing to change below this line
spwMCastAdr='239.12.255.255'
spwPort = 9522

discoveryRequest = '534d4100000402a0ffffffff0000002000000000'
discoveryResponse ='534d4100000402a000000001000200000001'

class MulticastClientUDP(DatagramProtocol):
    def startProtocol (self):
        print "Joining speedwire multicast group."
        self.transport.joinGroup(spwMCastAdr)
        self.transport.setOutgoingInterface(localInterface)

        print "Sending discovery request."
        data = discoveryRequest.decode ('hex')
        self.transport.write(data, (spwMCastAdr, spwPort))

    def datagramReceived (self , datagram, (srcAddress , port)):
        data = datagram.encode('hex')
        if (data.startswith(discoveryResponse)):
            print "Found device: " + srcAddress

def stopReactor():
    print "Discovery finished."
    reactor.stop()

reactor.listenMulticast(spwPort , MulticastClientUDP(), listenMultiple=True)
reactor.callLater(10, stopReactor)
reactor.run()
```

danbates2, 2018-10-12:
I've set today aside to test this.
Let see what happens!

danbates2, 2018-10-12:
Flawless.
The only thing was that it took a very long time to install python-twisted, it looked like a hang but it just needed time.
[image]

closed 2018-10-12 by danbates2