SBFspot/SBFspot #255: issues with SBFspotUpload - compiling and starting as service

mw446391, 2018-10-22:
_pi@raspberrypi-3:~ $ uname -a
Linux raspberrypi-3 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux_

I've installed SBF v3.5.0 today and came across some issues when compiling SBFspotUploadDaemon:

_pi@raspberrypi-3:~ $ sudo apt-get install libcurl3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**Note, selecting 'libcurl4-openssl-dev' instead of 'libcurl3-dev'**
libcurl4-openssl-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi-3:~ $ cd \~/SBFspot/SBFspotUploadDaemon
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/obj//../SBFspotUploadCommon || mkdir -p sqlite/obj//../SBFspotUploadCommon
test -d sqlite/obj//../SBFspot || mkdir -p sqlite/obj//../SBFspot
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ ../SBFspotUploadCommon/PVOutput_x.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE  -o sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o -I../SBFspot
g++ main.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/main.o -I../SBFspot
In file included from ../SBFspotUploadCommon/CommonServiceCode.h:48:0,
                 from main.cpp:41:
**../SBFspotUploadCommon/Configuration.h:73:27: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  ERRLEVEL    m_LogLevel = LOG_INFO_;
                           ^**
g++ ../SBFspotUploadCommon/Configuration.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/../SBFspotUploadCommon/Configuration.o -I../SBFspot
In file included from ../SBFspotUploadCommon/Configuration.cpp:37:0:
**../SBFspotUploadCommon/Configuration.h:73:27: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  ERRLEVEL    m_LogLevel = LOG_INFO_;
                           ^**
g++ ../SBFspotUploadCommon/CommonServiceCode.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/../SBFspotUploadCommon/CommonServiceCode.o -I../SBFspot
In file included from ../SBFspotUploadCommon/CommonServiceCode.h:48:0,
                 from ../SBFspotUploadCommon/CommonServiceCode.cpp:35:
**../SBFspotUploadCommon/Configuration.h:73:27: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  ERRLEVEL    m_LogLevel = LOG_INFO_;
                           ^**
g++ ../SBFspotUploadCommon/PVOutput.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/../SBFspotUploadCommon/PVOutput.o -I../SBFspot
g++ ../SBFspot/db_SQLite.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/../SBFspot/db_SQLite.o -I../SBFspot
g++ sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o sqlite/obj/main.o sqlite/obj/../SBFspotUploadCommon/Configuration.o sqlite/obj/../SBFspotUploadCommon/CommonServiceCode.o sqlite/obj/../SBFspotUploadCommon/PVOutput.o sqlite/obj/../SBFspot/db_SQLite.o -s -o sqlite/bin/SBFspotUploadDaemon -Wl,-Bdynamic -lpthread -lcurl -lsqlite3 
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $_ 

It compiled, but I'm curious if the warning came about because I have 'libcurl4-openssl-dev' installed instead of 'libcurl3-dev'.

Also I couldn't start SBFspotUpload as service - not sure if this is related to the compiler warnings:

_pi@raspberrypi-3:/etc $ sudo nano /usr/local/bin/sbfspot.3/SBFspotUpload.service
pi@raspberrypi-3:/etc $ sudo systemctl enable /usr/local/bin/sbfspot.3/SBFspotUpload.service
Created symlink from /etc/systemd/system/multi-user.target.wants/SBFspotUpload.service to /usr/local/bin/sbfspot.3/SBFspotUpload.service.
Created symlink from /etc/systemd/system/SBFspotUpload.service to /usr/local/bin/sbfspot.3/SBFspotUpload.service.
pi@raspberrypi-3:~ $ sudo systemctl start SBFspotUpload
pi@raspberrypi-3:~ $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled)
   Active: activating (auto-restart) since Mon 2018-10-22 11:50:33 BST; 6s ago
  Process: 8543 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=0/SUCCESS)
 Main PID: 8543 (code=exited, status=0/SUCCESS)_

But launching it from /etc/rc.local works.

Cheers
MW

sbf-, 2018-10-23:
No, 'libcurl4-openssl-dev' is OK and the warnings shouldn't be a problem too... (never saw them on my build system)
For the service start, is there any logging in SBFspotUpload.log?

mw446391, 2018-10-23:
I see lots off messages like this in the SBFspotUpload.log:

_[11:50:33] INFO: Starting SBFspotUploadDeamon Version 1.0.1
[11:50:33] ERROR: Unable to open output file. See syslog for details._

For the same time in syslog, I see this:

_Oct 22 11:50:33 raspberrypi-3 systemd[1]: Starting SBFspot Upload Daemon...
Oct 22 11:50:33 raspberrypi-3 systemd[1]: Started SBFspot Upload Daemon.
Oct 22 11:50:33 raspberrypi-3 SBFspotUploadDaemon: Unable to open output file /tmp/SBFspotUploadDaemon.out: Success
Oct 22 11:50:43 raspberrypi-3 systemd[1]: SBFspotUpload.service holdoff time over, scheduling restart.
Oct 22 11:50:43 raspberrypi-3 systemd[1]: Stopping SBFspot Upload Daemon...
Oct 22 11:50:43 raspberrypi-3 systemd[1]: Starting SBFspot Upload Daemon...
Oct 22 11:50:43 raspberrypi-3 systemd[1]: Started SBFspot Upload Daemon.
Oct 22 11:50:43 raspberrypi-3 SBFspotUploadDaemon: Unable to open output file /tmp/SBFspotUploadDaemon.out: Success
Oct 22 11:50:54 raspberrypi-3 systemd[1]: SBFspotUpload.service holdoff time over, scheduling restart._

And in /tmp, I have:

_pi@raspberrypi-3:/tmp $ ls -al /tmp/SBFspotUploadDaemon.out 
-rw-r--r-- 1 root root 0 Oct 22 19:17 /tmp/SBFspotUploadDaemon.out_

sbf-, 2018-10-23:
It looks like you forgot one step:
`sudo make install_sqlite`

You're still running V1.0.1 - should be V2.0.0
`[11:50:33] INFO: Starting SBFspotUploadDeamon Version 1.0.1`

mw446391, 2018-10-24:
Hmm - I did run the command during the install. I kept a log of the terminal session and I can see this:

```
pi@raspberrypi-3:~/SBFspot/SBFspot $ sudo make install_sqlite
test -d /usr/local/bin/sbfspot.3/ || mkdir -p /usr/local/bin/sbfspot.3/
cp sqlite/bin/SBFspot /usr/local/bin/sbfspot.3/
cp TagList*.txt /usr/local/bin/sbfspot.3/
cp date_time_zonespec.csv /usr/local/bin/sbfspot.3/
cp SBFspot.cfg /usr/local/bin/sbfspot.3/SBFspot.default.cfg
cp *SQLite.sql /usr/local/bin/sbfspot.3/
pi@raspberrypi-3:~/SBFspot/SBFspot $ !po
popd
-bash: popd: directory stack empty
pi@raspberrypi-3:~/SBFspot/SBFspot $ !pu
pushd /usr/local/bin/
/usr/local/bin ~/SBFspot/SBFspot
pi@raspberrypi-3:/usr/local/bin $ cd sbfspot.3/
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ ls -lrt
total 1656
-rwxrwxrwx 1 pi   pi      5766 Feb 28  2017 SBFspot.cfg
-rwxrwxrwx 1 pi   pi      1867 Mar  2  2017 SBFspotUpload.cfg
-rwxr-xr-x 1 root staff 130280 May 12  2017 SBFspotUploadDaemon
-rwxr-xr-x 1 root staff 322372 Oct 22 11:26 SBFspot
-rwxr-xr-x 1 root staff 195142 Oct 22 11:26 TagListDE-DE.txt
-rwxr-xr-x 1 root staff 187283 Oct 22 11:26 TagListEN-US.txt
-rwxr-xr-x 1 root staff 197586 Oct 22 11:26 TagListES-ES.txt
-rwxr-xr-x 1 root staff 201070 Oct 22 11:26 TagListFR-FR.txt
-rwxr-xr-x 1 root staff 195772 Oct 22 11:26 TagListIT-IT.txt
-rwxr-xr-x 1 root staff 189852 Oct 22 11:26 TagListNL-NL.txt
-rwxr-xr-x 1 root staff  35855 Oct 22 11:26 date_time_zonespec.csv
-rw-r--r-- 1 root staff   5935 Oct 22 11:26 SBFspot.default.cfg
-rw-r--r-- 1 root staff   1142 Oct 22 11:26 Update_340_SQLite.sql
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ popd
~/SBFspot/SBFspot
```

Should I rerun the command?

sbf-, 2018-10-24:
No, just stop the running SBFspotUpload process and restart it.
I guess it's just copied over the old version which is still running. Don't know if this is possible in Linux.

mw446391, 2018-10-24:
Ok - got it sorted.

As you rightly identified, I was running SBFspotUpload v1.0.1; I stopped my running SBFspotUpload process and copied the SBFspotUploadDaemon file from the ~/SBFspot/SBFspotUploadDaemon/sqlite/bin directory to /usr/local/bin/sbfspot.3 and then executed:

```
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ sudo systemctl enable /usr/local/bin/sbfspot.3/SBFspotUpload.service
Created symlink from /etc/systemd/system/multi-user.target.wants/SBFspotUpload.service to /usr/local/bin/sbfspot.3/SBFspotUpload.service.
Created symlink from /etc/systemd/system/SBFspotUpload.service to /usr/local/bin/sbfspot.3/SBFspotUpload.service.
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ sudo systemctl start SBFspotUpload
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled)
   Active: active (running) since Wed 2018-10-24 11:30:58 BST; 3s ago
 Main PID: 19660 (SBFspotUploadDa)
   CGroup: /system.slice/SBFspotUpload.service
           └─19660 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Oct 24 11:30:58 raspberrypi-3 systemd[1]: Started SBFspot Upload Daemon.
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ 
```

In the log file I see this now:

```
[11:30:58] INFO: SBFspotUploadDaemon Version 2.0.0
[11:30:58] INFO: Starting Daemon...
[11:31:57] INFO: Uploading datapoint: 20181024,11:30,20200368,1464,,,,239.87 => OK (200)
[11:36:29] INFO: Uploading datapoint: 20181024,11:35,20200511,1716,,,,239.94 => OK (200)
```

Also checked that process runs after a reboot and it works.

Thanks for your help :-)

mw446391, 2018-10-24:
BTW, I tried this and it failed:

```
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $ sudo make install_sqlite
test -d /usr/local/bin/sbfspot.3/ || mkdir -p /usr/local/bin/sbfspot.3/
cp sqlite/bin/SBFspotUploadDaemon /usr/local/bin/sbfspot.3/
cp: cannot create regular file ‘/usr/local/bin/sbfspot.3/SBFspotUploadDaemon’: Text file busy
makefile:98: recipe for target 'install' failed
make: *** [install] Error 1
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $ pushd /usr/local/bin/sbfspot.3

```

Strange that this didn't fail when I originally installed v3.5.0.

When I kill the process and try it again, then it works:

```
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ ps -ef | grep -i sbf
root       619     1 45 Oct23 ?        06:29:45 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon
pi       19074 18616  0 11:23 pts/1    00:00:00 grep --color=auto -i sbf
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ sudo kill -9 619
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ ps -ef | grep -i sbf
pi       19084 18616  0 11:23 pts/1    00:00:00 grep --color=auto -i sbf
pi@raspberrypi-3:/usr/local/bin/sbfspot.3 $ popd
~/SBFspot/SBFspotUploadDaemon
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $ sudo make install_sqlite
test -d /usr/local/bin/sbfspot.3/ || mkdir -p /usr/local/bin/sbfspot.3/
cp sqlite/bin/SBFspotUploadDaemon /usr/local/bin/sbfspot.3/
cp SBFspotUpload.default.cfg /usr/local/bin/sbfspot.3/
pi@raspberrypi-3:~/SBFspot/SBFspotUploadDaemon $ 

```

closed 2020-06-19 by sbf-