SBFspot/SBFspot #256: Failed to initialize Speedwire connection

rucksman, 2018-10-22:
I am running SBFspot since about 2 years now without any problems. All of a sudden it stopped working:

    pi@raspberrypi:/usr/local/bin/sbfspot.3 $ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -nosql
    SBFspot V3.5.0
    Yet another tool to read power production of SMA solar inverters
    (c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
    Compiled for Linux (LE) 32 bit with MySQL support
    Commandline Args: -v -finq -nocsv -nosql
    Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
    Mon Oct 22 21:46:57 2018: INFO: Starting...
    ...
    Inverter IP address: 192.168.XXX.XXX from SBFspot.cfg
    ERROR: Connection to inverter failed!
    Is 192.168.XXX.XXX the correct IP?
    Please check IP_Address in SBFspot.cfg!
    Mon Oct 22 21:47:02 2018: CRITICAL: Failed to initialize Speedwire connection.

Any idea why all of a sudden the connection cant be established? No chances where made at the inverter or the network etc.

sbf-, 2018-10-23:
Inverter must have received a different IP address.
Try with 0.0.0.0 in config file for autodetection (does not always work)

rucksman, 2018-10-26:
Something strange happened, but I do not know what. So in the end I rebooted the inverter and after that I rebooted the raspberry as well, and all of a sudden everything worked as before. It seems as if the inverter has problems with Modbus every few months which can only be solved by rebooting it. Also a reboot of the raspberry is necessary afterwards from my experience.

closed 2018-10-26 by rucksman

fshumeli, 2019-12-08:
I’m having the same problem since 3 weeks now.
Every morning (this is my Crontab setting) I get the following local email:

> Cron <pi@raspi> /usr/local/bin/sbfspot.3/monthdata
> ERROR: Connection to inverter failed!
> Is 192.168.2.170 the correct IP?
> Please check IP_Address in SBFspot.cfg!

192.168.2.170 is Ethernet static address for the inverter. I also tried with its WiFi static address.. no changes!
Both addresses are STATIC in inverter config. Even reboots haven’t fixed this. 

Pinging both static IP’s from the same raspberry gives perfect response times on my LAN (all Ethernet).
HTTP access on both IP’s from any browser will show inverter’s page perfectly as well.

Just ignoring this with > /dev/null 2>&1 would stop the emails, but that’s not what I want.. it used to work for months. Nothing specific happened at any point... 
I am clueless !?

Any insight welcome..

rucksman, 2020-01-03:
Well, I see this still once in a while. I have still no clue what the reason might be. Not an ideal solution for sure, but I simple reboot my Raspberry every night which partly solves the problem.

timaba, 2020-01-11:
I'm having the same problem for 4 weeks now. Looking at sunnyportal.com I saw, that my plant got an update exact one hour before the connection stopped. 

> Firmware-Version des Hauptprozessors | 2.03.4.R | **07.12.2019 00:55:40** |  
> Firmware-Version des Betriebssystems | 3.09.0.R | **07.12.2019 00:55:40** |  
> Gerätename | Home Manager 2 556 | 20.08.2017 16:16:39 |  
> Geräteklasse | Kommunikationsprodukte | 18.11.2015 08:59:10 |  
> Gerätetyp | Sunny Home Manager 2.0 | 18.11.2015 08:59:10 |  
> Softwarepaket | 2.03.4.R | 07.12.2019 00:55:40

> Geräteklasse | Solar-Wechselrichter | 07.01.2008 03:27:18 |  
> Gerätetyp | SB 3600SE-10 | 07.01.2008 03:27:18 |  
> Softwarepaket | 2.05.2.R | 05.01.2008 19:04:11

I don't think, this is an coincidence.

As reported above, Ping is working perfectly. Data-transfer to sunnyportal.com also, just the communication to SBFspot doesn't work at all. And it doesn't matter if it is configurated with IP oder multicast (0.0.0.0).

```
# /usr/local/bin/sbfspot.3/SBFspot -v5 -finq
SBFspot V3.6.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2019, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v5 -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=0.0.0.0
Password=<undisclosed>
MIS_Enabled=0
Plantname=HW12_PV
OutputPath=/home/<undisclosed>/smadata/%Y
OutputPathEvents=/home/<undisclosed>/smadata/%Y/Events
Latitude=<undisclosed>
Longitude=<undisclosed>
Timezone=Europe/Berlin
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=de-DE
BTConnectRetries=10
SQL_Database=<undisclosed>
SQL_Hostname=127.0.0.1
SQL_Username=<undisclosed>
SQL_Password=<undisclosed>
### End of Config ###
Sat Jan 11 14:21:57 2020: INFO: Starting...
sunrise: 08:16
sunset : 16:47
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 999387831 (0x3B9172B7)
Inverter IP address: 192.168.178.31 found via broadcastidentification
ERROR: Connection to inverter failed!
Is 192.168.178.31 the correct IP?
Please check IP_Address in SBFspot.cfg!
Sat Jan 11 14:22:03 2020: CRITICAL: Failed to initialize Speedwire connection.
```

sbf-, 2020-01-11:
I’m afraid your SHM and/or EM is conflicting with SBFspot.

timaba, 2020-03-11:
> I’m afraid your SHM and/or EM is conflicting with SBFspot.

Yes, me too. I have checked with tcpdump/wireshark, the inverter doesn't send multicast packets as discribed in the SpeedWire-docs - but still is pingable.
The EM on the other hand sends its 608 byte packets every second.
I did a reboot on both. Maybe I should say goodbye to the correlation of the EM update (2.03.05R) and SBFspot stopping to work properly and call the SMA-Service.

PvForFuture, 2020-06-17:
Hallo,
ich habe dasselbe Problem (gehabt). Ich wiederhole die Connection einfach solange bis es klappt (alle 10 sec.). Meistens läuft es wieder nach höchstens 5 Versuchen.
Mein cron Job sieht folgendermaßen aus:
 #!/bin/bash
#
counter=0
grepCount=99
return=""
log=/home/pi/Solar/SMA_Spot/logs/PV_for_Future_$(date '+%Y%m%d').log
logTmp=/home/pi/Solar/SMA_Spot/logs/PV_for_Future_TMP.log
while [ $counter -lt 10  -a  $grepCount -ne 0 ]; do
	echo -e $return$counter": ********************** $(date) ********************** from spotdata">$logTmp
	/usr/local/bin/sbfspot.3/SBFspot -d0 -v0 -ad0 -am0 -ae0 -finq -q -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg &>>$logTmp
	cat $logTmp >> $log
	grepCount=$(grep -c -i "ERROR" $logTmp)
	grepCount=$(( $grepCount + $(grep -c -i "CRITICAL" $logTmp) ))
	let counter+=1
	return="\n"
	$(sleep 4)
done