SBFspot/SBFspot #257: Total AC and Generator Values for SMA Sunny

tgoEP, 2018-10-28:
Hi, 

SMA Sunny (4.4.M) is an island Inverter with battery and diesel Generator feature on the grid side and solar on the island side. I got one on my boat (2,2KW solar, 5kw diesel, LG resu 10).. I use sbfspot on rasperry to get charge and load from the inverter via ethernet to schedule generator runs and measure solar power.

My problem is, that while generator running, only the negative charge value is shown on AC SPOT values- I dont get any consumption any more - ..or am able to calculate it somehow.  SMA suppport wrote me, that the overal total transfered by the Sunny is "Inv Pwr ATT".
 I noticed, that in the TAG files in source, lots more parameters are available.

Did anyone think of making these available? Is this easily possible?

Thanks for a tipp :) - I have no idea, in which command I can get this data.

Anyone here with helpful thoughts?

Thanks,

Timo

sbf-, 2018-10-28:
I don't find "Inv Pwr ATT" in the taglist file...
Which one(s) do you want?

tgoEP, 2018-10-29:
Wow thanks for the fast reply.

10901=Metering.GnMs.TotWOut\4598784\Aktuelle Generatorleistung
10906=Metering.WhIn\4602368\Aufgenommene Energie
10907=Metering.WhOut\4602624\Abgegebene Energie

These should do the job. Hopefully.

tgoEP, 2018-10-29:
but maybe its a better enhancement to enable a command line argument to just print out the value of a given tag number? ...then all future tags are supported somehow out of the box. at least a bit.