SBFspot/SBFspot #263: Suddenly no more uploads sqlite3_exec() returned: 'unable to open database file' while executing

broke23, 2018-11-23:
ExportSpotDataToCSV()
Fri Nov 23 18:59:36 2018: ERROR: Unable to open output file /home/pi/smadata/2018/DiNa-Spot-20181123.csv
[2018-11-23 18:59:36.348] Error: sqlite3_exec() returned: 'unable to open database file' while executing
INSERT OR IGNORE INTO Inverters VALUES(2130432209,'SB 5000TL-21 209','SB 5000TL-21','02.81.01.R',0,0,0,0,0,0,'','',0)
[2018-11-23 18:59:36.348] Error: exec_query() returned: 'unable to open database file' while executing
INSERT OR IGNORE INTO Inverters VALUES(2130432209,'SB 5000TL-21 209','SB 5000TL-21','02.81.01.R',0,0,0,0,0,0,'','',0)
[2018-11-23 18:59:36.349] Error: sqlite3_exec() returned: 'unable to open database file' while executing
UPDATE Inverters SET Name='SB 5000TL-21 209',Type='SB 5000TL-21',SW_Version='02.81.01.R' WHERE Serial=2130432209
[2018-11-23 18:59:36.349] Error: exec_query() returned: 'unable to open database file' while executing
UPDATE Inverters SET Name='SB 5000TL-21 209',Type='SB 5000TL-21',SW_Version='02.81.01.R' WHERE Serial=2130432209
[2018-11-23 18:59:36.350] Error: sqlite3_exec() returned: 'unable to open database file' while executing
UPDATE Inverters SET TimeStamp=1542995976,TotalPac=0,EToday=13385,ETotal=15257406,OperatingTime=8292.1,FeedInTime=8230.27,Status='OK',GridRelay='N/A',Temperature=0 WHERE Serial=2130432209
[2018-11-23 18:59:36.350] Error: exec_query() returned: 'unable to open database file' while executing
UPDATE Inverters SET TimeStamp=1542995976,TotalPac=0,EToday=13385,ETotal=15257406,OperatingTime=8292.1,FeedInTime=8230.27,Status='OK',GridRelay='N/A',Temperature=0 WHERE Serial=2130432209
[2018-11-23 18:59:36.350] Error: sqlite3_exec() returned: 'unable to open database file' while executing
INSERT INTO SpotData VALUES(1542995976,2130432209,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,13385,15257406,0,8292.1,8230.27,72.1569,'OK','N/A',0)
[2018-11-23 18:59:36.351] Error: [spot_data]exec_query() returned: 'unable to open database file' while executing
INSERT INTO SpotData VALUES(1542995976,2130432209,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,13385,15257406,0,8292.1,8230.27,72.1569,'OK','N/A',0)
********************
* ArchiveDayData() *
********************
startTime = 5BF734F0 -> 23/11/2018 00:00:00
ExportDayDataToCSV()
Fri Nov 23 18:59:36 2018: ERROR: Unable to open output file /home/pi/smadata/2018/DiNa-20181123.csv
**********************
* ArchiveMonthData() *
**********************
startTime = 5BDADCB0 -> 01/11/2018 12:00:00
ExportMonthDataToCSV()
Fri Nov 23 18:59:37 2018: ERROR: Unable to open output file /home/pi/smadata/2018/DiNa-201811.csv
Reading events: 2018-Nov-01
ExportEventsToCSV()
Fri Nov 23 18:59:37 2018: ERROR: Unable to open output file /home/pi/smadata/2018/Events/DiNa-User-Events-201811-201811.csv
Fri Nov 23 18:59:37 2018: INFO: Done.

No clue what happened today. Didn't change a thing...

sbf-, 2018-11-23:
Looks like the disk is full.

broke23, 2018-11-24:
Thought so too at first glance but it seems not. Though it's stuck at 2.0gb it seems (5,4 available of 7,4.see underneath 

Disk /dev/sda: 7.6 GiB, 8120172544 bytes, 15859712 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: D3E256B7-703A-468D-A646-7AF1EC120A24

Device     Start      End  Sectors  Size Type
/dev/sda1   2048 15859678 15857631  7.6G Linux filesystem

sbf-, 2018-11-25:
Are you able to copy or create files onto that disk?
What is the size of SBFspot.db? Maybe this isn't important as you can't create the CSV files too. 
So I can only think of the partition is either full or it has become read-only.

broke23, 2018-11-25:
Db is only 8,4 mb

df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.4G  1.7G  5.4G  24% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   24M  441M   6% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  51% /boot
tmpfs            93M     0   93M   0% /run/user/1000

sbf-, 2018-11-26:
Only the db should be copied...
Installation goes superfast with [sbfspot-config](https://github.com/SBFspot/sbfspot-config)

broke23, 2018-11-29:
Got a question regarding that sbfspot-config, how do you setup the location ?  It's setup at 51.51/-0.13

but when editing it shows 51.51 only, and when you edit it you see xx.xx/-0.13

How to edit -0.13 ? 

afterwards in config then or this is a small error ?

sbf-, 2018-11-29:
Seems to be bug. I'll have to check it.

sbf-, 2018-12-03:
Fixed in 1.0.12
Now you can enter the location in one line.

closed 2018-12-03 by sbf-