SBFspot/SBFspot #264: Big gaps in database and missing documentation for commandline args

rucksman, 2018-11-26:
I am running sbfspot since almost 2 years now, and in the beginning it worked flawlessly. Since quite some time I have a problem: There are big gaps in my database, sometimes  40 or 50 minutes, sometimes 2 or 3 hours, even whole days or weeks. I did not change anything in my configuration: sbfspot is running on a raspberrypi 3, the database is stored on a small server (both are connected to my network via the same switch). My cronscript:
/usr/local/bin/sbfspot.3/SBFspot -finq
It is scheduled to run every 10 minutes. Do you have any idea what might be wrong here?

So in order to check what is going wrong I wanted to debug. But I did not find any documentation for the various commandline args. Can sombody point me to the place where I can check which args are available and what they are made for?

rucksman, 2018-11-26:
OK, found something related to my problem. I tried to start sbfspot on the commandline:
`/usr/local/bin/sbfspot.3/SBFspot -v5 -d5 -nosql -nocsv`

Result:
```
ethGetPacket()
Timeout reading socket
No data!
ERROR: Connection to inverter failed!
Is 192.168.XXX.XXX the correct IP?
Please check IP_Address in SBFspot.cfg!
Mon Nov 26 15:04:41 2018: CRITICAL: Failed to initialize Speedwire connection.
```

So obviously the inverter needs more time to answerm because after I issued the same command immediately after the first one that causes the error, everything worked fine. So how can I solve this problem?

sbf-, 2018-11-26:
Almost all command line tools have an `-?` option, so does SBFspot
```
pi@raspberrypi:~ $ /usr/local/bin/sbfspot.3/SBFspot -?
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
SBFspot [-options]
 -scan               Scan for bluetooth enabled SMA inverters.
 -d#                 Set debug level: 0-5 (0=none, default=2)
 -v#                 Set verbose output level: 0-5 (0=none, default=2)
 -ad#                Set #days for archived daydata: 0-300
                     0=disabled, 1=today (default), ...
 -am#                Set #months for archived monthdata: 0-300
                     0=disabled, 1=current month (default), ...
 -ae#                Set #months for archived events: 0-300
                     0=disabled, 1=current month (default), ...
 -cfgX.Y             Set alternative config file to X.Y (multiple inverters)
 -u                  Upload to online monitoring system (see config file)
 -finq               Force Inquiry (Inquire inverter also during the night)
 -q                  Quiet (No output)
 -nocsv              Disables CSV export (Overrules CSV_Export in config)
 -nosql              Disables SQL export
 -sp0                Disables Spot.csv export
 -installer          Login as installer
 -password:xxxx      Installer password
 -loadlive           Use predefined settings for manual upload to pvoutput.org
 -startdate:YYYYMMDD Set start date for historic data retrieval
 -settime            Sync inverter time with host time

Libraries used:
        SQLite V3.16.2
        BOOST V1.66.0
```

sbf-, 2018-11-26:
With big gaps you mean in the spot data? For 5-min historic data it should load all the data for that day.
Can you post (or send me by e-mail) 2 complete debug logs (-v5 -d5)
One when it fails and another when it's OK

sbf-, 2018-11-26:
Does it work with sunny explorer (on windows)?

rucksman, 2018-11-27:
> Almost all command line tools have an -? option, so does SBFspot
Sorry, this was obviously too easy ;-)

>With big gaps you mean in the spot data?
If I take a look at the database view vwSpotData, I can see for example for today an entry at 08:10, the next at 10:20, then 10:40 and since then nothing (it's 13:40 right now).

>Can you post (or send me by e-mail) 2 complete debug logs (-v5 -d5)
See my second post yesterday. It seems as if the first connection times out. If I try once again immediately after the error, it usually works.
>Does it work with sunny explorer (on windows)?
Yes, but I do not use it very often. As I am travelling at the moment I cannot test it.

sbf-, 2018-11-27:
I don't think it's a connection timeout...
SBFspot is sending a command that is probably not understood by the inverter (or not expected by the inverter at that time) The inverter doesn't reply and SBFspot waits until it times out.
Maybe it can help if you can provide a Wireshark trace of the communication between Sunny Explorer and your inverter during logon.
There aren't any other SMA devices added to the network recently?

rucksman, 2018-12-01:
No new SMA devices were added. I have really no idea why it worked for a long time without any problem and all of a sudden it stopped working - with no changes in the network or on the inverter...

rucksman, 2018-12-03:
So this is interesting. I installed SBFspot on a different machine and made a test:
``

The output is interesting:
```
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v5 -d5 -nosql -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=192.168.166.162
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/user/SBFspot/smadata/%Y
OutputPathEvents=/home/user/SBFspot/smadata/%Y/Events
Latitude=XX.XXXX
Longitude=XX.XXXX
Timezone=Europe/Brussels
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=3600
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=0
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=de-DE
BTConnectRetries=10
SQL_Database=SBFspot
SQL_Hostname=localhost
SQL_Username=SBFspotUser
SQL_Password=<undisclosed>
### End of Config ###
Mon Dec  3 11:44:51 2018: INFO: Starting...
sunrise: XX:XX
sunset : XX:XX
Connecting to Local Network...
Initializing...
SUSyID: XXX - SessionID: XXXXXXXXX
Inverter IP address: 192.168.XXX.102 from SBFspot.cfg
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 26 00 10 60 65 09 A0
00000020: FF FF FF FF FF FF 00 00 7D 00
00000030: 69 5E 7E 34 00 00 00 00 00 00
00000040: 01 80 00 02 00 00 00 00 00 00
00000050: 00 00 00 00 00 00 00 00
58 Bytes sent to IP [192.168.XXX.102]
ethGetPacket()
MAX_CommBuf is now 600 bytes
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
Received 600 bytes from IP [192.168.XXX.101]
 ==> packet ignored
^C
```
101 is my EnergyMeter. Is it possible that the multicast from the EnergyMeter disturbs SBFspot because it is on the same port? If I issue the same command immediately after I stopped the previous output, everything works fine. Also after waiting some minutes and issuing the same command again, it works fine.

sbf-, 2018-12-03:
Sure this is caused by the SMA energymeter... Difficult to fix I'm afraid.
One possibility is to put the connection code inside a retry loop (Ethernet.cpp) something like this:
``` c++
bool connected = false;
for (int retry = 0; retry < 5; retry++)
{
	writePacketHeader(pcktBuf, 0, NULL);
	writePacket(pcktBuf, 0x09, 0xA0, 0, anySUSyID, anySerial);
	writeLong(pcktBuf, 0x00000200);
	writeLong(pcktBuf, 0);
	writeLong(pcktBuf, 0);
	writeLong(pcktBuf, 0);
	writePacketLength(pcktBuf);

	//Send packet to first inverter
	ethSend(pcktBuf, inverters[0]->IPAddress);

	if ((rc = ethGetPacket()) == E_OK)
	{
		ethPacket *pckt = (ethPacket *)pcktBuf;
		inverters[0]->SUSyID = btohs(pckt->Source.SUSyID);	// Fix Issue 98
		inverters[0]->Serial = btohl(pckt->Source.Serial);	// Fix Issue 98
		connected = true;
	}
}
	
if (!connected)
{
	cerr << "ERROR: Connection to inverter failed!\n";
	cerr << "Is " << inverters[0]->IPAddress << " the correct IP?\n";
	cerr << "Please check IP_Address in SBFspot.cfg!\n";
	return E_INIT;
}

logoffSMAInverter(inverters[0]);

return rc;
```

I can't test it because I have a BT inverter - No speedwire nor energymeter

rucksman, 2019-01-09:
I tried the solution with the loop you proposed, but it did not work as expected. So my intention would be to include a pause in between the retries within the loop. Maybe it would be a good idea to increase the pause in between the retries. Unfortunately I am not a CPP guy, so could you please help?

sbf-, 2019-08-04:
Unable to fix.

closed 2019-08-04 by sbf-