SBFspot/SBFspot #265: Compilation problem

rucksman, 2018-12-03:
Because of the problems described in issue #264 I wanted to install SBFspot on a different machine. But after
`make mysql`
I get this error after some time:
```
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFspot.o
In file included from SQLselect.h:46:0,
                 from SBFspot.cpp:72:
db_MySQL.h:41:25: fatal error: mysql/mysql.h: Datei oder Verzeichnis nicht gefunden
 #include <mysql/mysql.h>
                         ^
compilation terminated.
make: *** [mysql/obj/SBFspot.o] Fehler 1
```

Any idea what goes wrong here?

freekdk, 2018-12-03:
You should install a packet like mysql-devel.

rucksman, 2018-12-03:
You are right... for some reason libmysqld-dev was missing... Thanks!

closed 2018-12-03 by rucksman