SBFspot/SBFspot #268: Daemon not connecting to server

fschepers, 2019-01-26:
Yesterday I installed SBFspot V3.5.0 on a new RPi-0-w. I followed the instructions “Installation Linux SQLite” Also I transferred an existing sqlite database and cfg files from an RPi-3 to the RPi-0.
 
Now SBFspot is running well on the RPi-0. The data from the SMA-inverter is transferred to the SBFspot.db.

Only the Daemon is not uploading. In the logfile is written: 
ERROR: addBatchStatus() returned Couldn't connect to server

I triple checked the cfg files, PVoutput_SID and PVoutput_Key. When I stop the Daemon the message is the same, only the timing is different. See the part of the SBFspotUpload-log file below.
Please advise.

[12:19:07] ERROR: addBatchStatus() returned Couldn't connect to server
[12:19:07] ERROR: addBatchStatus() returned Couldn't connect to server
[12:21:08] ERROR: addBatchStatus() returned Couldn't connect to server
[12:21:08] ERROR: addBatchStatus() returned Couldn't connect to server
[12:23:06] ERROR: addBatchStatus() returned Couldn't connect to server
[12:23:07] ERROR: addBatchStatus() returned Couldn't connect to server
[12:25:08] ERROR: addBatchStatus() returned Couldn't connect to server
[12:25:08] ERROR: addBatchStatus() returned Couldn't connect to server
[12:26:13] INFO: Stopping Daemon...
[12:26:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:27:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:28:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:29:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:30:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:31:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:32:48] ERROR: addBatchStatus() returned Couldn't connect to server
[12:33:27] INFO: SBFspotUploadDaemon Version 2.0.0
[12:33:27] INFO: Starting Daemon...
[12:34:00] ERROR: addBatchStatus() returned Couldn't connect to server
[12:34:04] ERROR: addBatchStatus() returned Couldn't connect to server
[12:36:06] ERROR: addBatchStatus() returned Couldn't connect to server
[12:36:07] ERROR: addBatchStatus() returned Couldn't connect to server
[12:38:06] ERROR: addBatchStatus() returned Couldn't connect to server
[12:38:06] ERROR: addBatchStatus() returned Couldn't connect to server
[12:40:07] ERROR: addBatchStatus() returned Couldn't connect to server
[12:40:07] ERROR: addBatchStatus() returned Couldn't connect to server

sbf-, 2019-01-29:
This is a CURL error...
Are you sure you can connect to pvoutput.org?

fschepers, 2019-01-29:
Thanks for your reaction SBF. I think you are right.
After uploading my issue to GitHub I remembered that during installation of SBFspot I had a problem with downloading the sourcecode (wget –N https://github.com/SBFspot/SBFspot/archive/V3.5.0.tar.gz)

Result:

> pi@RPi-0-w:~$ wget https://github.com/SBFspot/SBFspot/archive/V3.5.0.tar.gz
--2019-01-29 17:41:32--  https://github.com/SBFspot/SBFspot/archive/V3.5.0.tar.gz
Resolving github.com (github.com)... 140.82.118.3, 140.82.118.4
Connecting to github.com (github.com)|140.82.118.3|:443... failed: Network is unreachable.
Connecting to github.com (github.com)|140.82.118.4|:443... failed: Network is unreachable.

I think my mentioned issue is caused by the same ‘unreachable’ problem. (SBF called CURL)
The RPi-0 is installed with Debian 9 (Stretch). Using wget I also tried to download something else from a https-page. Result connection! (watch the IPv4 and IPv6 addresses):

> pi@RPi-0-w:~$ wget https://www.knmi.nl/nederland-nu/weer/waarschuwingen/noord-holland
--2019-01-29 17:43:01--  https://www.knmi.nl/nederland-nu/weer/waarschuwingen/noord-holland
Resolving www.knmi.nl (www.knmi.nl)... 2001:610:178:fd::47, 145.23.253.47
Connecting to www.knmi.nl (www.knmi.nl)|2001:610:178:fd::47|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27742 (27K) [text/html]
Saving to: ‘noord-holland’ noord-holland 100%[===========================================>]  27.09K  --.-KB/s    in 0.01s
2019-01-29 17:43:02 (2.25 MB/s) - ‘noord-holland’ saved [27742/27742]

fschepers, 2019-02-04:
UPDATE:
The RPI’s have a static IP. I configured it in network/interfaces.
The lines in network/interfaces were:
```
interface wlan0
static ip_address=192.168.178.244/24
static router=192.168.178.1
static domain_name_servers=208.67.220.220
```
I found in the file etc/network/interfaces a ‘s’ missing after the word router. I placed the s. And I put an IPv6 DNS address because the problem looked to me IPv4 / IPv6 related:
```
interface wlan0
static ip_address=192.168.178.244/24
static routers=192.168.178.1
static domain_name_servers=208.67.220.220 2001:B88:1202::10
```
After a reboot (and a while…) the RPi-zero was running fine! A could ping to pvoutput.org.

But: My other RPi (3B+) - same first configuration – worked as well! Pinging to pvoutput.org was possible. I am confused. Something in de router?

closed 2019-02-04 by fschepers