SBFspot/SBFspot #272: myql

htttito, 2019-02-11:
Hi! I hope someone can help :-)
I want to test sbfspot with mysql.
My problem is: sbfspot writes no data to db and I get this error:
ExportSpotDataToCSV()
[2019-02-11 20:41:22.479] Error: exec_query() returned : Table 'SBFspot.Inverters' doesn't exist
Executed Statement: INSERT IGNORE INTO Inverters VALUES(304918980,'STP 5000TL-20 980','STP 5000TL-20','02.52.00.N',0,0,0,0,0,0,'','',0)
[2019-02-11 20:41:22.479] Error: exec_query() returned : Table 'SBFspot.Inverters' doesn't exist
Executed Statement: UPDATE Inverters SET Name='STP 5000TL-20 980',Type='STP 5000TL-20',SW_Version='02.52.00.N' WHERE Serial=304918980
[2019-02-11 20:41:22.479] Error: exec_query() returned : Table 'SBFspot.Inverters' doesn't exist
Executed Statement: UPDATE Inverters SET TimeStamp=1549914082,TotalPac=0,EToday=5374,ETotal=35848986,OperatingTime=21691.8,FeedInTime=21348,Status='OK',GridRelay='N/A',Temperature=0 WHERE Serial=304918980
[2019-02-11 20:41:22.480] Error: [spot_data]exec_query() returned : Table 'SBFspot.SpotData' doesn't exist
Executed Statement: INSERT INTO SpotData VALUES(1549914082,304918980,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,5374,35848986,0,21691.8,21348,0,'OK','N/A',0)
********************

htttito, 2019-02-11:
I forget: I use ubuntu-server 18.04

sbf-, 2019-02-12:
`Table 'SBFspot.SpotData' doesn't exist`

Did you run `CreateMySQLDB.sql` script?

htttito, 2019-02-13:
Hi, 
No, how can i run these script? I don't find anything about it.

htttito, 2019-02-13:
OK, thank you for the tipp!
I found something about it.

closed 2019-02-16 by htttito