SBFspot/SBFspot #273: mysql no Power=0 and TimeStamp

htttito, 2019-02-16:
Hi again!
There is no Power=0 in vwDaydata or somewhere?
The Table stop a little bit before Power=0 .
There is also no Timestamp like in .csv for Power=0
(It is all after sunset)

Thanks!

htttito, 2019-02-16:
The Timestamp in DayData for example "1550334300" what does it stand for?

sbf-, 2019-02-16:
It's a UNIX timestamp (https://www.unixtimestamp.com/index.php)
What do you mean with

> There is no Power=0 in vwDaydata or somewhere?

htttito, 2019-02-16:
Hi thanks!
I mean that SbfSpot dont write zero values in the mysql Table!

htttito, 2019-02-16:
You can see that there is missing Timestamp and Power in all tables, after sunrise.
Look at https://pv.tito.dynu.net/ at the end of the site...

sbf-, 2019-02-19:
yes, that's by design. To save space in the db

htttito, 2019-02-19:
Thanks!
Is there a possibility to change this for mysql?
In csv there are zero values....

sbf-, 2019-02-19:
The zero's in csv is to be compatible with SMA's SunnyExplorer
Have a look at db_MySQL_Export.cpp
In the db_SQL_Export::day_data() function I'm looking for start and end of production.
Comment out these sections and define first_rec, last_rec

htttito, 2019-02-20:
OhOh my php.....
The code to find first and last_rec  is very good.
I only need one entry after sunset with Power=0 for my website.
It is not so important, its only to show it right ;-) 
If you have time...can you help me with the code?
Thanks :-)!

sbf-, 2019-02-20:
Try this (didn't test it)
Change
``` C
// Store data from first to last record
for (unsigned int idx = first_rec; idx <= last_rec; idx++)
```
To
``` C
// Store data from first to last record+1
for (unsigned int idx = first_rec; idx <= last_rec+1; idx++)
```
This should add 1 record with timestamp and 0 Watts

htttito, 2019-02-20:
It look good, i will see it exactly tomorrow!
VERY VERY COOL!
Thanks :-)

closed 2019-03-14 by sbf-