SBFspot/SBFspot #274: no data geupload with SBFspotUploadDaemon

Downhilll, 2019-02-18:
I have a issue with the upload data to PV output.
the configiration file seems like fine. In the log register /var/log/sbfspot.3/SBFspotUploadDaemon is no data uploding
[12:17:18] INFO: SBFspotUploadDaemon Version 2.0.0
[12:17:18] INFO: Starting Daemon...
[13:49:44] INFO: Stopping Daemon...
[13:49:48] INFO: SBFspotUploadDaemon Version 2.0.0
[13:49:49] INFO: Starting Daemon...
what do i wrong?

INFO:

# SBFspotUpload.cfg - Created Mon 18 Feb 15:02:33 CET 2019 with sbfspot-config V1.0.12
# For detailed configuration info, refer to SBFspotUpload.default.cfg
#
LogDir=/var/log/sbfspot.3
PVoutput_SID=2100086175:****
PVoutput_Key=2abccc4897cc3cb672be11d1a54fc7e5af******
SQL_Database=/home/pi/smadata/SBFspot.db
# End of Config

Downhilll, 2019-02-18:
pi@bovenpi:/usr/local/bin/sbfspot.3 $ ps -ef | grep -i sbf
pi         363     1  0 13:49 ?        00:00:03 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon
pi        5199  5195  0 15:25 ?        00:00:00 /bin/sh -c /usr/local/bin/sbfspot.3/daydata
pi        5200  5199  0 15:25 ?        00:00:00 /bin/bash /usr/local/bin/sbfspot.3/daydata
pi        5202  5200  0 15:25 ?        00:00:00 /usr/local/bin/sbfspot.3/SBFspot -v -ad1 -am0 -ae0 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
pi        5227   966  0 15:25 pts/0    00:00:00 grep --color=auto -i sbf
pi@bovenpi:/usr/local/bin/sbfspot.3 $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-18 13:49:48 CET; 1h 36min ago
 Main PID: 363 (SBFspotUploadDa)
   CGroup: /system.slice/SBFspotUpload.service
           └─363 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Feb 18 13:49:48 bovenpi systemd[1]: Started SBFspot Upload Daemon.

sbf-, 2019-02-19:
Is SBFspot running correctly? Logfile?
Is there valid data in the DB?

Downhilll, 2019-02-19:
Yes, SBFspot works perfectly, with 2 Csv files Esch day. the data contained all sensordetails as temperatuur, grid voltage, and Total data.

sbf-, 2019-02-19:
CSV files are not the proof of a perfect job. The daemon uses the sql db and if this one is empty, you'll never have data at pvoutput ;-)

Some logging of sbfspot might help (use -v5)

Show us the output of
`select * from vwPvoData limit 200;`

Downhilll, 2019-02-19:
pi@bovenpi:/usr/local/bin/sbfspot.3 $ sqlite3 /home/pi/smadata/SBFspot.db
SQLite version 3.16.2 2017-01-06 16:32:41
Enter ".help" for usage hints.
sqlite> select * from vwPvoData limit 200;
Error: no such table: vwPvoData
sqlite> .quit
pi@bovenpi:/usr/local/bin/sbfspot.3 $ 

pi@bovenpi:/usr/local/bin/sbfspot.3 $ sudo ./SBFspot -v5 -finq
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:09:F9:B1
Password=<undisclosed>
MIS_Enabled=0
Plantname=harryphpnew
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.37
Longitude=4.9
Timezone=Europe/Amsterdam
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Tue Feb 19 22:40:44 2019: INFO: Starting...
sunrise: 07:49
sunset : 17:58
Connecting to 00:80:25:09:F9:B1 (1/10)
Connecting to 00:80:25:09:F9:B1 (2/10)
Connecting to 00:80:25:09:F9:B1 (3/10)
Connecting to 00:80:25:09:F9:B1 (4/10)
Connecting to 00:80:25:09:F9:B1 (5/10)
Connecting to 00:80:25:09:F9:B1 (6/10)
Connecting to 00:80:25:09:F9:B1 (7/10)
Connecting to 00:80:25:09:F9:B1 (8/10)
Connecting to 00:80:25:09:F9:B1 (9/10

I see that er is no table vwPvoData. How can i this resolved?

Downhilll, 2019-02-19:
the table vwspotdata contains everything:
i@bovenpi:/usr/local/bin/sbfspot.3 $ sqlite3 /home/pi/smadata/SBFspot.db
SQLite version 3.16.2 2017-01-06 16:32:41
Enter ".help" for usage hints.
sqlite> select * from vwPvoData;
Error: no such table: vwPvoData
sqlite> select * from vwspotData;
2019-02-19 18:01:03|2019-02-19 18:00:00|SN: 2100086175|SB 4000TL-20|2100086175|0|0|0.0|0.0|95.23|95.12|0|0|0|0.0|0.0|0.0|234.72|0.0|0.0|0|0|0|6814|40673263|50.07|40188.5|38875.7|67.5|OK|Open|23.5
2019-02-19 17:56:03|2019-02-19 17:55:00|SN: 2100086175|SB 4000TL-20|2100086175|0|0|0.0|0.0|242.01|241.95|0|0|0|0.0|0.0|0.0|235.83|0.0|0.0|0|0|0|6814|40673263|49.97|40188.4|38875.7|67.8|OK|Open|24.1

sbf-, 2019-02-20:
Very strange this view doesn't exist...
Try to create it manually (but maybe other tables and/or views are missing too)

``` SQL
CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              dd.Power AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;
```

Downhilll, 2019-02-23:
Now that the sun is shining again, look again in the log file. now there appeared to be a too high data dump. This again gedeleted. Now the data appears to be uploaded again to PVoutput.
Problem solved. So ff try to get that high value there definitively. Thank everyone.
### 
__[14:46:31]_ ERROR: Uploading 100 datapoints, starting with 20190206,15:20,40564883,132,,,22.2,236.03 Bad request 400: Power value [486$
[14:47:31] INFO: Uploading 100 datapoints, starting with 20190206,15:20,40564883,132,,,22.2,236.03 => OK (200)
[14:48:32] INFO: Uploading 100 datapoints, starting with 20190207,15:10,40571765,516,,,27.3,236.73 => OK (200)
[14:49:31] INFO: Uploading 100 datapoints, starting with 20190208,14:55,40574341,228,,,24.9,239.57 => OK (200)
[14:50:32] INFO: Uploading 100 datapoints, starting with 20190209,14:40,40580063,348,,,27.3,237.48 => OK (200)
[14:51:31] INFO: Uploading 100 datapoints, starting with 20190210,14:45,40582624,132,,,24.3,236.31 => OK (200)
[14:52:31] INFO: Uploading 100 datapoints, starting with 20190221,14:50,40685903,588,,,26.9,239.08 => OK (200)
[14:53:31] INFO: Uploading 100 datapoints, starting with 20190222,13:45,40689184,480,,,31.3,238.34 => OK (200)
[14:54:31] INFO: Uploading 31 datapoints, starting with 20190223,12:20,40696143,2460,,,32.4,243.61 => OK (200)
[14:56:31] INFO: Uploading datapoint: 20190223,14:55,40702360,2112,,,37.4,243.13 => OK (200)
[15:01:31] INFO: Uploading datapoint: 20190223,15:00,40702534,2088,,,37.4,242.7 => OK (200)_

closed 2019-02-23 by Downhilll