SBFspot/SBFspot #275: SBFspotUpload error

migila1, 2019-02-21:
How can I upgrade the database? I tried some things:
- install sqlite3 tell's me i got the newest software already installed.
- apt-get update
- apt-get upgrade
- find a solution online, cant find it
without the correct result.

What database has to be upgraded to version 1? I have installed 3.5.0 this week, as a fresh install.

```
root@raspberrypi:/home/pi# sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-02-21 21:13:49 CET; 5min ago
  Process: 28434 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=1/FAILURE)
 Main PID: 28434 (code=exited, status=1/FAILURE)

Feb 21 21:13:49 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
Feb 21 21:13:49 raspberrypi SBFspotUploadDaemon[28434]: Upgrade your database to version 1
Feb 21 21:13:49 raspberrypi systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
Feb 21 21:13:49 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
Feb 21 21:13:49 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.

```

migila1, 2019-02-21:
Can it have to do with the schema version? That's suppose to be version 1 ? How can i check this? :)

sbf-, 2019-02-22:
I'm afraid this is a misleading error meaning "I don't find your database"
Can you have a look at your config file?

migila1, 2019-02-22:
/home/pi/SBFspot/SBFspot/SBFspot.cfg has: SQL_Database=/home/pi/smadata/SBFspot.db
and
/home/pi/SBFspot/SBFspotUploadDaemon/SBFspotUpload.cfg has the same SQL_Database=/home/pi/smadata/SBFspot.db
The file exists on that location. SBFspot is configured in Domoticz and is working.

migila1, 2019-02-22:
This codes work for me. So the database should exist isn't it?

(this are not my results)
`sqlite3 SBFspot.db
SQLite version 3.16.2 2017-01-06 16:32:41
Enter ".help" for usage hints.
sqlite> select * from config;
SchemaVersion|1
sqlite>.quit`

`cd /home/pi/smadata
sqlite3 SBFspot.db
SQLite version 3.7.13 2012-06-11 02:05:22
Enter ".help" for instructions
Enter SQL statements terminated with a ";"
sqlite> select * from vwspotdata;
2014-08-26 10:32:08|2014-08-26 10:30:00|SB4000TL|SB 4000TL-20|2100276197|127|56|0.37|0.339|345.3|166.73|178|0|0|0.769|0.0|0.0|232.98|0.0|0.0|183|178|0.0|332|14969473|49.96|15967.7|15502.0|64.7059|OK|Closed|36.84
sqlite> select * from inverters;
2100276197|SB4000TL|SB 4000TL-20|03.01.05.R|1409041928|178|0.332|14969.5|15967.7|15502.0|OK|Closed|0.0
sqlite> .quit`

sbf-, 2019-02-23:
DB looks ok.
What happens if you start daemon from command line (without systemctl)
```/usr/local/bin/sbfspot.3/SBFspotUploadDaemon -c /home/pi/SBFspot/SBFspotUploadDaemon/SBFspotUpload.cfg```

closed 2019-03-20 by sbf-