SBFspot/SBFspot #276: Setting konverter time- Problem with locale?

agricolaXIII, 2019-02-24:
Hello,
I'v got a problem with the konverter time. Can I correct it with SBFSpot?
When I try 
_/usr/local/bin/sbfspot.3/SBFspot -v -settime_
so I got the error "Failed to get current plant time (-1)".
Can it be a problem with the locale de_DE? I had change the file SBFspot.cfg:
**************************
Locale=de-DE
DateTimeFormat=%d.%m.%Y %H:%M:%S
DateFormat=%d.%m.%Y
TimeFormat=%H:%M:%S
**************************
The output from the command date on the command line of the Raspi:
**************************
pi@raspberry2:~/sbfspot.3 $ date
So 24. Feb 19:09:52 CET 2019
**************************
The full output of the request:
*************************
pi@raspberry2:~/sbfspot.3 $ /usr/local/bin/sbfspot.3/SBFspot -v -settime
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -settime
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Sun Feb 24 19:11:36 2019: INFO: Starting...
sunrise: 07:05
sunset : 17:33
Connecting to 00:80:25:34:14:B2 (1/10)
Initializing...
SUSyID: 125 - SessionID: 840526777 (0x32196BB9)
SMA netID=01
Serial Nr: 7... (213...)
BT Signal=65.5%
Logon OK
Failed to get current plant time (-1)
******************************************************
Is there any idea what is going on?

Sincerely
Jens

sbf-, 2019-02-24:
No, locale is not the issue...
Setting the inverter time with SBFspot does not always work
You will need sunnyexplorer to perform this task (Windows only)

agricolaXIII, 2019-02-25:
OK, thanks for your reply.
So I'll try this in the next days...
:-) Jens

closed 2019-03-14 by sbf-