SBFspot/SBFspot #277: Connecting to inverters: CRITICAL: bthConnect() returned -1

CJau777, 2019-03-03:
SBFspot seems to have trouble connecting to my inverters from a Raspberry Pi (1, Model B, bluetooth dongle added). This was running fine, but I needed to reinstall for various reasons and in the process updated to the latest SBFSpot.

I've been using the installer script at https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config, it gets through the config and gives the below output.

```Running SBFspot...
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with MySQL support
Commandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Sun Mar  3 17:58:01 2019: INFO: Starting...
sunrise: 07:06
sunset : 19:49
Connecting to 00:80:25:A4:87:AF (1/10)
Connecting to 00:80:25:A4:87:AF (2/10)
Connecting to 00:80:25:A4:87:AF (3/10)
Connecting to 00:80:25:A4:87:AF (4/10)
Connecting to 00:80:25:A4:87:AF (5/10)
Connecting to 00:80:25:A4:87:AF (6/10)
Connecting to 00:80:25:A4:87:AF (7/10)
Connecting to 00:80:25:A4:87:AF (8/10)
Connecting to 00:80:25:A4:87:AF (9/10)
Connecting to 00:80:25:A4:87:AF (10/10)
Sun Mar  3 17:58:11 2019: CRITICAL: bthConnect() returned -1
```
Running hcitool scan will show the inverters.
```
pi@raspberrypi:~/sma $ hcitool scan
Scanning ...
        00:80:25:A4:5D:59       SMA002d SN: 2001900854 SN2001900854
        00:80:25:A4:87:AF       SMA002d SN: 2001900861 SN2001900861
```

Any suggestions how I can debug the connection issue would be great.

sbf-, 2019-03-03:
Try to change BTaddress to other inverter (00:80:25:A4:5D:59) in SBFspot.cfg

CJau777, 2019-03-03:
Gives a similar response:

```
pi@raspberrypi:~/sma $ /usr/local/bin/sbfspot.3/SBFspot -v -ad0 -ae0 -sp0 -nocsv -nosql -finq
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with MySQL support
Commandline Args: -v -ad0 -ae0 -sp0 -nocsv -nosql -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Mon Mar  4 09:04:24 2019: INFO: Starting...
sunrise: 07:06
sunset : 19:47
Connecting to 00:80:25:A4:5D:59 (1/10)
Connecting to 00:80:25:A4:5D:59 (2/10)
Connecting to 00:80:25:A4:5D:59 (3/10)
Connecting to 00:80:25:A4:5D:59 (4/10)
Connecting to 00:80:25:A4:5D:59 (5/10)
Connecting to 00:80:25:A4:5D:59 (6/10)
Connecting to 00:80:25:A4:5D:59 (7/10)
Connecting to 00:80:25:A4:5D:59 (8/10)
Connecting to 00:80:25:A4:5D:59 (9/10)
Connecting to 00:80:25:A4:5D:59 (10/10)
Mon Mar  4 09:04:34 2019: CRITICAL: bthConnect() returned -1

CJau777, 2019-03-03:
Update - just rebooted the Pi, and it seems to be connecting now.  This was the first reboot since a significant upgrade, so perhaps there was a new kernel/module that weren't being used or something.

closed 2019-03-03 by CJau777