SBFspot/SBFspot #278: Failed to open database - stopping daemon

fschepers, 2019-03-03:
Hello,

For about three weeks I use SBFspot with a database in MariaDB. (After using sqlite for a few years.)
It works smooth and fast. I like it! 
This morning the daemon stopt at 08:29:33. See logfile below.
I re-started the daemon with 

> sudo systemctl start SBFspotUpload

and the daemon is running smooth again.

It looks to me that the daemon was overreacting by stopping itself after a faillure in connecting.

Is there a way to make the daemon "autorestart"?

Hava a nice day!

```
[08:15:36] INFO: Uploading 2 datapoints, starting with 20190303,08:10,19967253,0,,,,223.97,223.97,221.23,0,0,21.30 => OK (200)
[08:20:36] INFO: Uploading datapoint: 20190303,08:20,19967257,36,,,,181.38,181.38,177.98,37,35,22.00 => OK (200)
[08:25:36] INFO: Uploading datapoint: 20190303,08:25,19967259,24,,,,185.23,185.23,182.56,38,36,22.60 => OK (200)
[08:29:33] ERROR: Failed to open database: Can't connect to MySQL server on '192.168.178.210' (113 "No route to host")
[08:29:33] INFO: Can't connect to MySQL server
[08:29:44] INFO: SBFspotUploadDaemon Version 2.0.0
[10:46:03] INFO: SBFspotUploadDaemon Version 2.0.0
[10:46:03] INFO: Starting Daemon...
[10:46:10] INFO: Uploading 28 datapoints, starting with 20190303,08:30,19967263,48,,,,185.28,185.28,182.58,41,39,23.10 => OK (200)
[10:50:36] INFO: Uploading datapoint: 20190303,10:50,19967585,252,,,,214.72,214.72,213.03,143,135,28.00 => OK (200)
[10:55:36] INFO: Uploading datapoint: 20190303,10:55,19967607,264,,,,214.72,214.72,208.86,137,129,28.20 => OK (200)
````

sbf-, 2019-03-03:
You can try to modify your .service file`

`Restart=on-error`
or
`Restart=always`

fschepers, 2019-03-04:
Thanks for your reply.

In the default .service file the option is:
`Restart=on-success`
For me this looks a bit confusing...

When I change it to 
_Restart=on-error_
or
_Restart=always_
and after running `sudo systemctl daemon-reload` I get in _**both**_ cases:
` [/usr/local/bin/sbfspot.3/SBFspotUpload.service:9] Failed to parse service restart specifier, ignoring: on-error`

For the moment I put the option back to _Restart=on-success_ and made a bash-script to check and restart if needed the Daemon. A cronjob execute the script every hour.

closed 2019-03-14 by sbf-