SBFspot/SBFspot #281: SBFspotUploadDaemon

OscarAarts, 2019-03-12:
Good day,

I have a question about SBFspotUploadDaemon,
I created yesterday my RasPi with SBFspot 3.5
I used the manual from  https://github.com/SBFspot/SBFspot/wiki ... nux-SQLite
The automatic mode wont work, because i changed the default password from my inverter.

Now the system logs the data in the file sbfspot.db (I also can check them)
But the data wont be sent to PVoutput.
There i get a no data

Everything is running as user pi.
In the log from the upload i see
Mar 11 17:26:20 raspberrypi systemd[1]: Started SBFspot Upload Daemon.

There is no message that some datapoints are uploaded.

Oscar

sbf-, 2019-03-14:
S/N is too big to fit in SQLite's int data type, causing to insert a negative serial number.

sbf-, 2019-03-20:
This is continued [here (Dutch)](https://www.zonstraal.be/forum/viewtopic.php?f=38&t=21851)

sbf-, 2019-03-20:
Fixed in V3.5.1

closed 2019-03-20 by sbf-