SBFspot/SBFspot #284: Question regarding cronjob to retrieve monthly data

H4ndl3, 2019-03-30:
In the wiki the cron job to retrieve monthly data is configured to run daily at 05:55 in the morning ([link](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite#%EF%B8%8F-automation)). However, during winter the inverter won't be up at this time in my area because it is still dark then. Why was 05:55 being chosen?

HarvsG, 2019-04-01:
It used the `-finq` option which wakes the inverter even if there is no sunlight

H4ndl3, 2019-04-11:
@sbf- just to be sure: Will there be a problem if the `monthdata` cronjob runs at the same time as a `daydata` cronjob?

sbf-, 2019-04-12:
No problem, but it's overkill to run it every 5min as only 1 record will be added per day.

closed 2019-04-12 by H4ndl3