SBFspot/SBFspot #288: Last Data not read from inverter

zekirma, 2019-04-11:
When I compare the data from Sunny Portal and the CSV file, the last data from the inverter are not read. 
Exemple from yesterday:
Sunny Portal values:
19:15 0,974 
19:30 0,804 
19:45 0,655 
20:00 0,443 
20:15 0,126 
20:30 0,021 
20:45 0,001 
21:00  
21:15  

And CSV File
10/04/2019 19:15:00;SB3.0-1AV-40 567;Sunny Boy 3.0;1992111567;102,000;783,000;0,440;3,160;230,100;247,100;865,000;0,000;0,000;3,590;0,000;0,000;241,150;0,000;0,000;885,000;860,000;0,000;20,209;47,725;49,990;58,111;56,739;0,000;Ok;Closed;0,000
10/04/2019 19:35:00;SB3.0-1AV-40 567;Sunny Boy 3.0;1992111567;81,000;564,000;0,350;2,280;226,900;246,700;638,000;0,000;0,000;2,650;0,000;0,000;240,770;0,000;0,000;645,000;625,000;0,000;20,458;47,973;50,000;58,444;57,072;0,000;Ok;Closed;0,000
10/04/2019 20:00:00;SB3.0-1AV-40 567;Sunny Boy 3.0;1992111567;50,000;185,000;0,220;0,840;225,000;221,000;227,000;0,000;0,000;0,940;0,000;0,000;241,810;0,000;0,000;235,000;184,000;0,000;20,637;48,152;50,020;58,861;57,489;0,000;Ok;Closed;0,000

So the last data are not read. Why?

sbf-, 2019-04-12:
10/04/2019 19:15:00
10/04/2019 19:35:00
10/04/2019 20:00:00

It's not only the latest data, but also in between...
Probably bad communication between SBFspot and inverter - Try with verbose logging (-v5)
Daydata (every 5min) should be OK

zekirma, 2019-04-12:
When I start manualy it doesn't work the first time; I make a ping. It take long time to receive the first response the I retry manuly and it work. 
Here the log.

pi@SBFspot:~ $ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Apr 12 17:19:22 2019: INFO: Starting...
sunrise: 06:56
sunset : 20:30
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 979874730 (0x3A67B3AA)
Inverter IP address: 192.168.2.12 from SBFspot.cfg
ERROR: Connection to inverter failed!
Is 192.168.2.12 the correct IP?
Please check IP_Address in SBFspot.cfg!
Fri Apr 12 17:19:27 2019: CRITICAL: Failed to initialize Speedwire connection.pi@SBFspot:~ $ /usr/lping 192.168.2.12
PING 192.168.2.12 (192.168.2.12) 56(84) bytes of data.
40 bytes from 192.168.2.12: icmp_seq=7 ttl=64 (truncated)
40 bytes from 192.168.2.12: icmp_seq=8 ttl=64 (truncated)
40 bytes from 192.168.2.12: icmp_seq=9 ttl=64 (truncated)
40 bytes from 192.168.2.12: icmp_seq=10 ttl=64 (truncated)
40 bytes from 192.168.2.12: icmp_seq=11 ttl=64 (truncated)

--- 192.168.2.12 ping statistics ---
11 packets transmitted, 5 received, 54% packet loss, time 10222ms
rtt min/avg/max/mdev = 8.840/28.169/82.149/27.463 ms
^Cpi@SBFspot:~ $ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Apr 12 17:19:45 2019: INFO: Starting...
sunrise: 06:56
sunset : 20:30
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 831132914 (0x318A14F2)
Inverter IP address: 192.168.2.12 from SBFspot.cfg
Logon OK
SUSyID: 342 - SN: 1992111567
Device Name:      SB3.0-1AV-40 567
Device Class:     Solar Inverters
Device Type:      Sunny Boy 3.0
Software Version: 01.03.17.R
Serial number:    1992111567
SUSyID: 342 - SN: 1992111567
Device Status:      Ok
SUSyID: 342 - SN: 1992111567
Device Temperature: 0.0Â°C
SUSyID: 342 - SN: 1992111567
GridRelay Status:      Closed
SUSyID: 342 - SN: 1992111567
Pac max phase 1: 3000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 342 - SN: 1992111567
Energy Production:
	EToday: 12.894kWh
	ETotal: 84.074kWh
	Operation Time: 83.62h
	Feed-In Time  : 81.73h
SUSyID: 342 - SN: 1992111567
DC Spot Data:
	String 1 Pdc:   0.279kW - Udc: 250.70V - Idc:  1.110A
	String 2 Pdc:   0.284kW - Udc: 245.80V - Idc:  1.150A
SUSyID: 342 - SN: 1992111567
AC Spot Data:
	Phase 1 Pac :   0.545kW - Uac: 241.37V - Iac:  2.260A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.531kW
SUSyID: 342 - SN: 1992111567
Grid Freq. : 50.00Hz
SUSyID: 342 - SN: 1992111567
Current Inverter Time: 12/04/2019 17:19:46
Inverter Wake-Up Time: 08/04/2019 10:15:33
Inverter Sleep Time  : 12/04/2019 17:19:46
********************
* ArchiveDayData() *
********************
startTime = 5CAFB8E0 -> 12/04/2019 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 5CA1EF30 -> 01/04/2019 13:00:00
Reading events: 2019-Apr-01
Fri Apr 12 17:19:47 2019: INFO: Done.
pi@SBFspot:~ $

sbf-, 2019-04-12:
11 packets transmitted, 5 received, 54% packet loss, time 10222ms doesn't look so good...
You should fix these network issues first.

closed 2019-05-05 by sbf-