SBFspot/SBFspot #290: Etoday and temperature missing from STP6.0 inverter

ot3, 2019-04-16:
Etoday and temperature are shown as ZERO in below's SBFspot log:

```
pi@eon:[~]: /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -cfgSTP6.0-3AV-40.cfg
SBFspot V3.5.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -finq -nocsv -cfgSTP6.0-3AV-40.cfg
Reading config '/usr/local/bin/sbfspot.3/STP6.0-3AV-40.cfg'
Tue Apr 16 15:34:54 2019: INFO: Starting...
sunrise: 06:41
sunset : 20:33
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 947831743 (0x387EC3BF)
Inverter IP address: 192.168.178.110 from SBFspot.cfg
Logon OK
SUSyID: 378 - SN: 3002969XXX
Device Name:      East/West
Device Class:     Solar Inverters
Device Type:      Sunny Tripower 6.0
Software Version: 02.13.07.R
Serial number:    3002969XXX
SUSyID: 378 - SN: 3002969XXX
Device Status:      Ok
SUSyID: 378 - SN: 3002969XXX
Device Temperature: 0.0°C
SUSyID: 378 - SN: 3002969XXX
GridRelay Status:      Closed
SUSyID: 378 - SN: 3002969XXX
Pac max phase 1: 6000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 378 - SN: 3002969XXX
Energy Production:
        EToday: 0.000kWh
        ETotal: 32.460kWh
        Operation Time: 10.86h
        Feed-In Time  : 10.48h
SUSyID: 378 - SN: 3002969XXX
DC Spot Data:
        String 1 Pdc:   2.159kW - Udc: 415.48V - Idc:  5.199A
        String 2 Pdc:   1.823kW - Udc: 418.03V - Idc:  4.363A
SUSyID: 378 - SN: 3002969XXX
AC Spot Data:
        Phase 1 Pac :   1.285kW - Uac: 233.73V - Iac:  5.493A
        Phase 2 Pac :   1.289kW - Uac: 234.83V - Iac:  5.491A
        Phase 3 Pac :   1.304kW - Uac: 237.15V - Iac:  5.506A
        Total Pac   :   3.870kW
SUSyID: 378 - SN: 3002969XXX
Grid Freq. : 50.01Hz
SUSyID: 378 - SN: 3002969XXX
Current Inverter Time: 2019-04-16 15:34
Inverter Wake-Up Time: 2019-04-16 15:06
Inverter Sleep Time  : 2019-04-16 15:34
********************
* ArchiveDayData() *
********************
startTime = 5CB4FEE0 -> 16/04/2019 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 5CA1EF30 -> 01/04/2019 13:00:00
Reading events: 2019-Apr-01
Tue Apr 16 15:34:54 2019: INFO: Done.
```

The same code runs fine with my old SB3000-TL21 inverter.

sbf-, 2019-04-17:
For device temp: not all inverters report their internal temperature, so STP6.0 might be one of them.
For EToday, afaik only STP6.0 has this issue. If someone can test this with Sunny Explorer (Windows)
If SE shows EToday, I might be able to fix it, otherwise i'll have to look for a workaround.
@fonske Any news on this?

sbf-, 2019-04-17:
No, it's not related. However, there is a [workaround](https://github.com/SBFspot/SBFspot/issues/137)
You will need to modify the vwPvodata
```
CASE WHEN dd.Power > 8000
    THEN 0
    ELSE dd.Power
END
```

sbf-, 2019-04-17:
> This starts to look like a firmware issue in the inverter and I've reported this to SMA

Please let us know the outcome. Really curious ;-)

ot3, 2019-04-17:
Thanks for the quick response. I've updated the firmware to version: 2.13.33.R and this has the following effect:

1. daily yield is now available
2. temperature is still not available

I've found the culprit (**476232**); see below:
```
sqlite> select * from vwPvoData where V2 > 10000 order by Timestamp desc limit 10;
2019-04-17 06:25:00|East/West|Sunny Tripower 6.0|3002969XXX|39685|476232|||||||||||
```
Problem is I don't know how to remove this record as it's a view. I've seen your solution where you change the schema but I'd first like to try a simple delete as this problem may not recur with the latest firmware. How can I do this?

P.S. will of course let you know the outcome of the SMA service call

ot3, 2019-04-17:
In the mean time I've tried this:
```
sqlite> select * from daydata where Power > 10000 order by Timestamp desc limit 10;
1555475100|3002969674|39685|476232|
sqlite> delete from daydata where power> 10000;
sqlite> select * from daydata where Power > 10000 order by Timestamp desc limit 10;
sqlite> DROP VIEW IF EXISTS vwPvoData;
sqlite>
sqlite> CREATE VIEW vwPvoData AS
   ...>        SELECT dd.Timestamp,
   ...>               dd.Name,
   ...>               dd.Type,
   ...>               dd.Serial,
   ...>               dd.TotalYield AS V1,
   ...>               CASE WHEN dd.Power > 10000
   ...>               THEN 0
   ...>               ELSE dd.Power
   ...>               END AS V2,
   ...>               cons.EnergyUsed AS V3,
   ...>               cons.PowerUsed AS V4,
   ...>               spot.Temperature AS V5,
   ...>               spot.Uac1 AS V6,
   ...>               NULL AS V7,
   ...>               NULL AS V8,
   ...>               NULL AS V9,
   ...>               NULL AS V10,
   ...>               NULL AS V11,
   ...>               NULL AS V12,
   ...>               dd.PVoutput
   ...>          FROM vwDayData AS dd
   ...>               LEFT JOIN vwAvgSpotData AS spot
   ...>                      ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
   ...>               LEFT JOIN vwAvgConsumption AS cons
   ...>                      ON dd.Timestamp = cons.Nearest5min
   ...>         ORDER BY dd.Timestamp DESC;
sqlite> select * from daydata where Power > 10000 order by Timestamp desc limit 10;
sqlite> select * from daydata where Power > 10000 order by Timestamp desc limit 10;
1555475100|3002969XXX|39685|476232|

```
The erroneous record is initially empty but as soon as SBFspot runs again, it is repopulated.

sbf-, 2019-04-17:
> The erroneous record is initially empty but as soon as SBFspot runs again, it is repopulated.
That's normal, but it will not upload to PVO because of the changed view.
But beware, P=10000 is still too high for PVO as it refuses everything above 8450 ("too high for system size [8450]")

> daily yield is now available
That's good news. As I wrote before: not all inverters report temperature, so there is nothing to do about it.

@fonske
It seems a FW update solves the EToday issue

ot3, 2019-04-17:
> 
> 
> > The erroneous record is initially empty but as soon as SBFspot runs again, it is repopulated.
> > That's normal, but it will not upload to PVO because of the changed view.

It did initially but upon retesting PVoutput started accepting data. All data has been successfully uploaded.

> > But beware, P=10000 is still too high for PVO as it refuses everything above 8450 ("too high for system size [8450]")

This I don't understand: I thought PVoutput had a 50% margin?

> 
> > daily yield is now available
> > That's good news. As I wrote before: not all inverters report temperature, so there is nothing to do about it.

We'll let SMA chew on this.
> 
> @fonske
> It seems a FW update solves the EToday issue

Many thanks for your support.

fonske, 2019-04-17:
I am on 2.13.33.R and can't find a newer firmware for the SMA tripower 6.0
But daily yield is still 0

It doesn't bother me, because PVoutput upload works.
Not clear how you're problem was solved, if on the same firmware

ot3, 2019-04-17:
Can't explain that. I've reported 3 problems to SMA of which the lack of daily yield was one. I'll report back if and when I get an answer from SMA.

ot3, 2019-04-20:
Further analysis of sbfspot.db has shown that although both PVoutput and the WebUI show the daily yield (Etoday), this value is not stored in the database:
```
sqlite> select Etoday from SpotData order by Timestamp desc limit 10;
9470
0
9425
0
9380
0
9335
0
9291
0
sqlite>
```
The first number shown is for my old SB3000 inverter while the zero's are for the new STP6 inverter. I assume that both PVoutput and the WebUI use a different strategy to obtain Etoday by subtracting the current Etotal from yeterday's Etotal. Is this an approach that SBFspot could emulate?

sbf-, 2019-04-22:
SBFspot takes Total Yield from daydata table, not from spotdata. PVOutput calculates power and generation from this value (over 5min time span)
Spotdata is only used for temperture (if available) and AC/DC voltage and current.
Don't know if it is really necessary to calculate EToday?

ot3, 2019-04-22:
It's not _necessary_ to calculate EToday but I find it quite useful, e.g. to calculate how much of the targeted yield has been reached for the day.

scotte, 2019-08-01:
For what it's worth, my Inverter - SB7.7-1SP-US-40 - doesn't have device temperature or EToday either. Perhaps they either no longer provide that data in newer inverters or it's in a different place in the SpeedWire protocol blobs. For my purposes this isn't a problem (pvoutput figures out daily production just fine, and I use ambient air temperature from my weather station), but this does look to be some change by SMA. Also curious to hear if SMA responds...

sbf-, 2019-08-02:
@scotte For temperature OK, not every inverter supports it.
EToday is something else... It's important data I think, no?
I would like to know if SunnyExplorer reports it, and if that is the case we might be able to get it out of the data stream. If not, there must be a way to calculate it.

sbf-, 2019-08-04:
@scotte what firmware version do you have? V2.13.33.R should solve the etoday issue. See above  ...

scotte, 2019-08-04:
Sorry, I didn't mean to suggest EToday isn't important - just that it wasn't really bothering me that much at the moment, but it yes not having it is not ideal. I missed the note above re firmware version - I have `02.04.81.R` which is the latest available, at least as an over-the-air update, for my inverter at the moment. If I have anything useful to add, I'll update here.

ot3, 2019-08-06:
In the end - since SBFSpot does not make the device temperature available - I wrote a program to extract the data via the ModBus API. Therefore I can't think of a reason for SBFSpot not to use the same approach.

sbf-, 2019-08-06:
The reason is I don't have a ModBus capable inverter. SBFspot was BT only in the beginning. Thanks to some other developers/testers we were able to make it work with speedwire. Feel free to share your code. I hope it's c++ using libmodbus and not python ;-)

ot3, 2019-08-06:
Unfortunately in this case, it's Python. Here's the code:
```
#!/usr/bin/python3
# -*- coding: utf-8 -*-

__version__     = '1.0.1'
__appname__     = 'log-inverter-temperature.py'
__project__     = 'PV'
__description__ = 'periodically load data from an SMA PV inverter and store the results in InfluxDB'
__status__      = 'production'

import os
import sys
import time
import argparse
import logging
import logging.handlers
import pendulum
from common_logging import config_logging
from influxdb import InfluxDBClient
from pymodbus.client.sync import ModbusTcpClient as ModbusClient

class Settings:
  def __init__(self,
               unit=3,
               register=30953,  # internal temperature,
               influx_host='eon',
               database='energy',
               inverter1_serial='2130216XXX',  # SMA SB3000-TL21    - Araminta Station South
               inverter2_serial='3002969XXX',  # SMA STP6.0-3AV-40  - Araminta Station East/West
               ):
      self.unit = unit
      self.register = register
      self.influx_host = influx_host
      self.database = database
      self.inverter1_serial = inverter1_serial
      self.inverter2_serial = inverter2_serial

settings = Settings()

def get_arguments():
    # setup argument parsing
    argparser = argparse.ArgumentParser(description=__description__)
    argparser.add_argument('--logfile', '-lf', help='log file:')
    argparser.add_argument('--loglevel', '-ll', default='WARNING', help='log level: {VERBOSE, DEBUG, INFO, MONITOR, WARNING, ERROR, CRITICAL}')
    # get command line arguments
    cmd_args = argparser.parse_args()
    config_logging(cmd_args)  # configure the logger

def store_measurements(serial, internal_temperature):
    points = []
    dts = int(time.time())
    point = {
        "measurement": "sma",
        "time": dts,
        "fields": {
            'internal_temperature': internal_temperature,
        },
        "tags": {
            'serial': serial,
        },
    }
    points.append(point)
    points2influx(points, settings.database)

def points2influx(points, database):
    if points == None:
        return
    else:
        logging.verbose(points)
        client = InfluxDBClient(database=database)  # init InfluxDB
        client.write_points(points,  # and write to InfluxDB
                            time_precision='s',
                            database=database,
                            retention_policy=None,
                            tags=None,
                            batch_size=None)

def main():
    get_arguments()
    logging.monitor('- START %s - VERSION %s', __appname__, __version__)
    client = ModbusClient('192.168.178.110', port=502)
    client.connect()
    while True:
        rr = client.read_holding_registers(settings.register, 2, unit=settings.unit)
        internal_temperature = rr.registers[1] / 10  # convert deci-°C to °C
        logging.monitor('- %s°C', internal_temperature)
        if internal_temperature > 0:
            store_measurements(settings.inverter2_serial, internal_temperature)
        time.sleep(60)
    client.close()

if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        logging.warning("- PROGRAM TERMINATED AT USER'S REQUEST")
        try:
            sys.exit(0)
        except SystemExit:
            os._exit(0)
```

joanwa, 2019-09-06:
a bit off topic: @ot3 do you mind making your log-inverter-temperature.py available on github? Couldn't find it in your repos and would like to fork it.

Thanks, Johannes

ot3, 2019-09-08:
I've never created a github repo and a such am not familiar with the required steps. Feel free to use the code as shown above.

knights1, 2019-12-28:
> Thanks for the quick response. I've updated the firmware to version: 2.13.33.R and this has the following effect:
>
> 1. daily yield is now available
> 2. temperature is still not available
>
> I've found [...]

I know this is an old thread, but exactly my problem: @ot3 : did you got a response from SMA regarding the empty EToday value or do you have found a fix?

My SB3.0-41 is on the latest firmware and does not show EToday (and temperature).

ot3, 2019-12-29:
I never heard back from SMA but will give it another try early next year as they've completely revamped their service portal. As for the data: I'm now using the ModBus interface and that gives me all the data I need.

SyntaxError-404, 2020-01-17:
> > Thanks for the quick response. I've updated the firmware to version: 2.13.33.R and this has the Where did you find the firmware 2.13.33.R ? I cannot find it.

knights1, 2020-01-18:
> I never heard back from SMA but will give it another try early next year as they've completely revamped their service portal. As for the data: I'm now using the ModBus interface and that gives me all the data I need.

Hi ot3, did you perhaps already sent out a message Towards sma?

knights1, 2020-02-02:
> I never heard back from SMA but will give it another try early next year as they've completely revamped their service portal. As for the data: I'm now using the ModBus interface and that gives me all the data I need.

How do you use the modbus interface? And any update on SMA?

ot3, 2020-02-02:
I haven't taken the time yet to contact SMA. The Python code I use was posted by me in this forum on 2019-08-06.

knights1, 2020-02-02:
just save the code and run it with python right?

pi@raspberrypi:~ $ python bla.py
Traceback (most recent call last):
  File "bla.py", line 17, in <module>
    from common_logging import config_logging
ImportError: No module named common_logging

any ideas about the error?

ot3, 2020-02-03:
Yes, you will need to change the code to work in your environment (IP numbers, etc.) I'm using my own logging module which you will have to replace with standard logging. I shared this code "as-is" and it's not productized; you need some python experience to make the required changes.

Wiebo, 2020-05-11:
> Thanks for the quick response. I've updated the firmware to version: 2.13.33.R 

Does someone have version the Firmware version 2.13.33.R for SMA Tripower available?

sbf-, 2020-06-03:
@scotte Does SunnyExplorer reports EToday or not?

scotte, 2020-06-03:
I have no idea - my inverter doesn't have bluetooth, so I don't have or use SunnyExplorer.

sbf-, 2020-06-03:
Sunny Explorer works with both Bluetooth and Speedwire (webconnect)

knights1, 2020-06-04:
It does.

sbf-, 2020-06-04:
Nice! Then it should be possible to fix it.
Would you be so kind to make a Wireshark trace of the communication between SE and inverter?

knights1, 2020-06-04:
Sorry, havent got the time atm. I fixed the etoday by logging Howard
grafana by using a formula. If you want this, let me know!

scotte, 2020-06-05:
> Sunny Explorer works with both Bluetooth and Speedwire (webconnect)

I did not know that! The website says it only supports bluetooth! - `Free PC software for monitoring the PV plant via Bluetooth`

I'll install this and try to get a tcpdump over the next few days.

sbf-, 2020-06-05:
Yes, I know... I'm pretty sure they want to force people with web connected devices to their Sunny Portal.
Thanks for your help.

scotte, 2020-06-12:
Unfortunately Sunny Explorer couldn't find my inverter - so either the 7.7US-40 isn't supported, or it doesn't work over wifi. I did not try plugging directly into ethernet on the inverter. Oddly, Sunny Explorer seems to only work via broadcast - I couldn't find any way to configure the inverter's IP address directly as used by SBFSpot or to access the onboard web interface. *shrug*

sbf-, 2020-06-13:
Thanks for testing...
I'll try to calculate it from daily data.

knights1, 2020-06-14:
What tool do you use for this? I got some experience with grafana and
calculating the etoday from etotal.  If you also use this, let me know.

Regards

sbf-, 2020-06-14:
I'm the developer of SBFspot, so I will calculate EToday inside SBFspot, unless someone else can provide a communication trace between Sunny Explorer and inverter.

mincequi, 2021-01-24:
I own a SMA STP10.0-3AV-40 and had exactly the same issue.
Updating Firmware helped.

There is also an official statement from SMA:
https://www.photovoltaikforum.com/thread/136374-sma-sunny-tripower-10-0-stp10-0-3av-40-tagesertrag-per-modbus-auslesen/?postID=1841356#post1841356

Apparently, SMA reacted to user reactions about missing day yield.

sbf-, 2021-01-24:
Thanks for the update @mincequi

Syonyk, 2021-01-24:
Posting to follow.

I have three Sunny Boy US inverters (two 6.0s and a 3.0).

All are running US firmware 03.01.25.R

I have no temperature or EToday values reported either.  I have Sunny Explorer installed in a VM and can communicate with the inverters, but I don't know where to look to see if it's reporting daily production values.  If it reports them somewhere, I'm happy to provide tcpdumps.

The web interface certainly does on the dash overview, but I'm not sure which json file it comes from.

mincequi, 2021-01-24:
Day yield should be shown at the overview tab of your inverter within Sunny Explorer. Besides the total yield and power.
If day yield is missing there, it is likely, that your inverter does not support it.

Syonyk, 2021-01-24:
This is what I see in the inverter overview tab in Sunny Explorer.

[image]

However, if I access the web interface directly, I see daily production values.

[image]

So clearly the inverter *has* them.

But if Sunny Explorer isn't showing them, it sounds like there's no real hope for getting them reported properly in SBFspot?

sbf-, 2021-01-25:
> So clearly the inverter _has_ them.

Not so sure about that. If this was the case it should be visible in SE

[image]

Most probably it is calculated by the web interface (DayYield = Current TotalYield - Today Starting TotalYield)

> But if Sunny Explorer isn't showing them, it sounds like there's no real hope for getting them reported properly in SBFspot?

Must be possible with above calculation...

Syonyk, 2021-01-25:
Interesting.  That could be the case, I suppose.  I'll look at the SQL tables and see if I can generate that - I'm playing around with SunPortal and I can't see daily production numbers.

closed 2021-03-18 by sbf-

Syonyk, 2021-03-18:
Tested with commit https://github.com/SBFspot/SBFspot/commit/931e0ce3f0f2a38c17a2894c9696096d8aed845d just now - seems to generate valid EToday data on my inverters, which weren't reporting it before, so, confirming this fix works.

Thank you very much!

```
...
Energy Production:
        EToday: 8.769kWh
        ETotal: 1887.163kWh
        Operation Time: 1334.67h
        Feed-In Time  : 1303.90h
SUSyID: 413 - SN: xxxxxxxx75
Energy Production:
        EToday: 8.811kWh
        ETotal: 1943.578kWh
        Operation Time: 1408.14h
        Feed-In Time  : 1309.67h
SUSyID: 413 - SN: xxxxxxxx11
Energy Production:
        EToday: 3.009kWh
        ETotal: 843.890kWh
        Operation Time: 1327.07h
        Feed-In Time  : 1188.03h
SUSyID: 413 - SN: xxxxxxxx2
...
```

sbf-, 2021-03-19:
Thanks for the feedback.
@ot3 for inverter temperature there is no fix, but you have your own solution.

silliewimons, 2022-08-07:
@ot3 - i did get in touch with SMA-help-desk about reading the "temperature" from SMA inverters/. . . .the devices

SMA-help-desk informed that, from their knowledge, no SMA inverters provides reporting temp. towards SunnyPortal
  --since being aware that you found a solution using "modbus
     "SMA-help-desk was asking to get more information  eg  screen-shots of the result  any useull additional information
??  please confirm that your python code can be distributed to SMA-help-desk
??  please inform, from your knowledge/eperience, which temperature you can extract (ambient | in-cabinet | module | . . . ) 
  
SMA-help-desk agrees that this summer-period is exceptionel and several inverters should report -derating-due-to-temp-
  
background - during this summer the envornment temperature was often higher then 40°C, (measured) 
- inverters installed on roof, (with or without protection against sun-heating) should encounter in-cabinet temperature of more then 50°C  AND  several inverters should have activated derating conditions 

@sbf- can you please deliver any kind of additional information that might be usefull about this item
   in view of further contact with SMA-help-desk

kind regards  sillie

@sillie - follow on

silliewimons, 2022-08-07:
For information, my SunnyBoy SB3300 has active-cooling 
the fan is activated once that 'cooling-block with power-electronics' reaches 40°C
with the fan on maximum speed (DC-input to inverter 4000Wattt) inside temp. of inverter does get up to 53°C 
there is no derating-event in the event log
 - one can measure the inside temp with ir-temp-meter (most IR-health care allow to measure objcet-surface temp.)
kr  sillie

ot3, 2022-08-07:
> @ot3 - i did get in touch with SMA-help-desk about reading the "temperature" from SMA inverters/. . . .the devices
>
> SMA-help-desk informed that, from their knowledge, no SMA inverters provides [...]

Regarding my Python code:

1.  I've never claimed that the inverter temperature will be published to Sunny Portal; I'm publishing to MQTT and my own database/dashboard
2. The code is already published near the beginning of this thread. This will only be a solution for you if you can modify the code and run in your own environment.

I hope that this will clarify my solution.

silliewimons, 2022-08-07:
@ot3  thank you  for  the complete answer        including the graph 

??when??   i will get information from SMA   i'll keep you (and sbf) posted
kr sillie

commit 931e0ce3 (2021-03-18): Fix #290 Etoday and temperature are shown as ZERO from STP6.0 inverter