SBFspot/SBFspot #292: SBFspotUploadDaemon build failed on Raspberry Pi

jogi19, 2019-05-14:
Hello, I’ve installed SBFspot on my Raspberry PI and it works really nice. Now I want to use SBFspotUploadDaemon to upload to PVOutput. I did all the steps according to https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite .

_pi@raspberrypi ~/SBFspot/SBFspotUploadDaemon $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/obj//…/SBFspotUploadCommon || mkdir -p sqlite/obj//…/SBFspotUploadCommon
test -d sqlite/obj//…/SBFspot || mkdir -p sqlite/obj//…/SBFspot
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ main.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE -O2 -o sqlite/obj/main.o -I…/SBFspot
In file included from …/SBFspotUploadCommon/…/SBFspot/SBFspot.h:46:0,
_ from …/SBFspotUploadCommon/PVOutput.h:38,_
_ from …/SBFspotUploadCommon/CommonServiceCode.h:38,_
_ from main.cpp:41:_
_…/SBFspotUploadCommon/…/SBFspot/boost_ext.h:42:0: warning: "GLIBC_HAVE_LONG_LONG" redefined [enabled by default]
/usr/include/features.h:350:0: note: this is the location of the previous definition
In file included from …/SBFspotUploadCommon/CommonServiceCode.h:48:0,
_ from main.cpp:41:
…/SBFspotUploadCommon/Configuration.h:73:27: error: ISO C++ forbids initialization of member ‘m_LogLevel’ [-fpermissive]
…/SBFspotUploadCommon/Configuration.h:73:27: error: making ‘m_LogLevel’ static [-fpermissive]
…/SBFspotUploadCommon/Configuration.h:73:27: error: ISO C++ forbids in-class initialization of non-const static member ‘m_LogLevel’
cc1plus: warning: unrecognized command line option “-Wno-unused-local-typedefs” [enabled by default]
makefile:106: recipe for target ‘sqlite/obj/main.o’ failed
make: *** [sqlite/obj/main.o] Error 1_

As workaround I changed line 73 in SBFspotUploadCommon/Configuration.h

ERRLEVEL m_LogLevel = LOG_INFO_;
to
ERRLEVEL m_LogLevel ;

At least it worked for me, but I don't know if this is the solution for everyone

sbf-, 2019-05-14:
What version of gcc do you have?

jogi19, 2019-05-14:
pi@raspberrypi ~ $ gcc --version
gcc (Debian 4.6.3-14+rpi1) 4.6.3

closed 2020-04-27 by sbf-