SBFspot/SBFspot #293: Raspberry Pi - Built libicui18n.so.57: undefined reference

JaseSmith78, 2019-05-18:
I am receiving 
```"/usr/local/bin/sbfspot.3/SBFspot: symbol lookup error: /usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined symbol: _ZNK6icu_5711Replaceable11hasMetaDataEv" ```
when trying to run daydata that i had already installed, so i tried downing the latest and rebuilding and now i am receiving this when i try to build.

Linux pi-3 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
g++ (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516

```
pi@pi-3:~/SBFspot/SBFspot $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ sqlite/obj/boost_ext.o sqlite/obj/misc.o sqlite/obj/sunrise_sunset.o sqlite/obj/SBFNet.o sqlite/obj/CSVexport.o sqlite/obj/Ethernet.o sqlite/obj/EventData.o sqlite/obj/ArchData.o sqlite/obj/SBFspot.o sqlite/obj/TagDefs.o sqlite/obj/Bluetooth.o sqlite/obj/db_SQLite.o sqlite/obj/db_SQLite_Export.o -s -o sqlite/bin/SBFspot -Wl,-Bdynamic -lpthread -lbluetooth -lboost_date_time -lboost_system -lboost_filesystem -lboost_regex -lsqlite3
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `vtable for icu_57::UnicodeStringAppendable'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::Replaceable::hasMetaData() const'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::UnicodeStringAppendable::~UnicodeStringAppendable()'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::Replaceable::clone() const'
collect2: error: ld returned 1 exit status
makefile:119: recipe for target 'sqlite/bin/SBFspot' failed
make: *** [sqlite/bin/SBFspot] Error 1
```

sbf-, 2019-05-18:
Try `make clean` first

JaseSmith78, 2019-05-18:
Same error

```
pi@pi-3:~/SBFspot/SBFspot $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/boost_ext.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/EventData.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/Bluetooth.o
g++ db_SQLite.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/db_SQLite.o
g++ db_SQLite_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_SQLITE -o sqlite/obj/db_SQLite_Export.o
g++ sqlite/obj/boost_ext.o sqlite/obj/misc.o sqlite/obj/sunrise_sunset.o sqlite/obj/SBFNet.o sqlite/obj/CSVexport.o sqlite/obj/Ethernet.o sqlite/obj/EventData.o sqlite/obj/ArchData.o sqlite/obj/SBFspot.o sqlite/obj/TagDefs.o sqlite/obj/Bluetooth.o sqlite/obj/db_SQLite.o sqlite/obj/db_SQLite_Export.o -s -o sqlite/bin/SBFspot -Wl,-Bdynamic -lpthread -lbluetooth -lboost_date_time -lboost_system -lboost_filesystem -lboost_regex -lsqlite3
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `vtable for icu_57::UnicodeStringAppendable'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::Replaceable::hasMetaData() const'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::UnicodeStringAppendable::~UnicodeStringAppendable()'
//usr/lib/arm-linux-gnueabihf/libicui18n.so.57: undefined reference to `icu_57::Replaceable::clone() const'
collect2: error: ld returned 1 exit status
makefile:119: recipe for target 'sqlite/bin/SBFspot' failed
make: *** [sqlite/bin/SBFspot] Error 1
```

sbf-, 2019-05-18:
So, SBFspot worked and suddenly it stopped? Did you change something? Some updates maybe?
It looks like some system files (libicui18n.so.57) are missing or corrupt.
Just an idea, if you have a spare SD card, you can reinstall it completely. Will take some time, I know ;-)

JaseSmith78, 2019-05-19:
It does seem to be an issue with my setup - I reinstalled all packages using the below and recompiling works - I am still waiting for an upload to run to confirm that is working

```
dpkg --get-selections | grep -v deinstall | awk '{print $1}' > list.log
awk '$1=$1' ORS=' ' list.log > newlist.log
sudo apt clean
sudo apt-get install --reinstall $(cat newlist.log)
```

closed 2019-05-25 by sbf-