SBFspot/SBFspot #294: Logs are going to /home/pi/smadata

mk-au, 2019-05-22:
In 3.4.0:
The logs for SBFspot went to /var/log/sbfspot.3/
The logs for SBFspotUpload went to /home/smadata/logs/

In 3.5.1:
The logs for SBFspot now go to /home/smadata/logs/
The logs for SBFspotUpload went to /home/smadata/logs/

Should these not both go to /var/log/sbfspot.3/ ?

sbf-, 2019-05-22:
Yes, since 3.4.0 all logs should go to `/var/log/sbfspot.3/`

For SBFspot, its location is defined in the daydata script:
```
#!/bin/bash
#
log=/var/log/sbfspot.3/MyPlant_$(date '+%Y%m%d').log
/usr/local/bin/sbfspot.3/SBFspot -v -ad1 -am0 -ae0 >>$log
```

For SBFspotUpload daemon, the loglocation is taken from its config file:
`LogDir=/var/log/sbfspot.3`

mk-au, 2019-05-23:
@sbf- thanks for that... based on the timestamp it looks like the update script modified those files. In that case not necessarily a bug if the location of all logs for 3.5.1 is intended to be /home/pi/smadata/logs/

Thanks :)

closed 2020-07-08 by sbf-