SBFspot/SBFspot #295: SBFspotUploadDeamon - libcurl4 - OPENSSL_3 not found

Mikebert4, 2019-05-26:
I have the following issue from the SBFspot uploader deamon - having googles around, it looks like an issue that's occured in the jump from libcurl3 to libcurl4 - but I might be wrong.

does anyone else have the issue, or know if a workaround?

` SBFspotUploadDaemon[936]: /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/bin/sbfspot.3/SBFspotUploadDaemon)`

Thanks

Mikebert4, 2019-05-27:
Ok... this was solved by re-compiling SBFspotUpload with the libcurl4-openssl-dev package installed.

I guess I must have built it against libcurl3 and then apt-get upgraded myself into this corner... oh well.. it's here if others see the same!

closed 2019-05-27 by Mikebert4

TerrorSource, 2019-07-12:
@Mikebert4 how did you solve this exactly?

Got the same issue and im not the only one:
https://forum.pvoutput.org/t/sbfspot-fresh-install-curl-openssl-3-not-found/2071

Mikebert4, 2019-07-12:
Hi!

Really sorry, I usually get back to all my posts when I solve an issue!

This was solved by re-compiling SBFSPOT UPLOAD myself with the
libcurl4-openssl-dev package installed.

I guess I must have built it against libcurl3 and then apt-get upgraded
myself into this corner... oh well.. it's here if others see the same!

TerrorSource, 2019-07-12:
@Mikebert4 Thanks for the reply.

Was already busy with installing/configuring by hand via https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite

This does work. the auto-config script doesn't right now.

sbf-, 2019-07-12:
I was able to reproduce the issue and as a workaround I recompiled the daemon for Raspbian buster.
You can find it here: https://github.com/SBFspot/SBFspot/releases/edit/V3.5.2
Download SBFspotUploadDaemon_buster and rename it to SBFspotUploadDaemon
Replace the file in /usr/local/bin/sbfspot.3
Reinstall the service:

```
sudo systemctl disable /usr/local/bin/sbfspot.3/SBFspotUpload.service
sudo systemctl enable /usr/local/bin/sbfspot.3/SBFspotUpload.service
sudo systemctl start SBFspotUpload
sudo systemctl status SBFspotUpload

● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-07-12 22:50:26 CEST; 13min ago
 Main PID: 2887 (SBFspotUploadDa)
    Tasks: 1 (limit: 2200)
   Memory: 1.5M
   CGroup: /system.slice/SBFspotUpload.service
           └─2887 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Jul 12 22:50:26 sbfspot systemd[1]: Started SBFspot Upload Daemon.
```

AlainVeGe, 2019-07-27:
@sbf- Can't find SBFspotUploadDaemon_buster. Did you remove it?

sbf-, 2019-07-27:
The correct link is https://github.com/SBFspot/SBFspot/releases/tag/V3.5.2

Abrastreve, 2019-08-05:
Thanks for the workaround SBF.

My question is: should we still execute the instruction <sudo apt-get install libcurl3-dev> as described in the intallation manual?

The problem is that I do not get <SBFspotUpload.service> in the /usr/local/bin/sbfspot.3/ directory when I follow the manual.

sbf-, 2019-08-05:
I still have to review the docs

ChrisV78, 2019-11-15:
> I was able to reproduce the issue and as a workaround I recompiled the daemon for Raspbian buster.
> You can find it here: https://github.com/SBFspot/SBFspot/releases/tag/V3.5.2
> Download [...]

Thnx! Your workaround works like a charm(and easy to find with Google) after sbfspot-config did it's thing and the only error I got was this :)

23x7x365, 2019-12-24:
Hi, I’ve tried the work around on Buster but it doesn’t appear to work when using MariaDB. Daemon reports ignoring ‘sql_hostname’ and then says it can’t open SQLite dB....

Depechie, 2019-12-28:
@sbf- when I download the work around buster I get following exception when trying to start the upload deamon:
`Failed at step EXEC spawning /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: Permission denied`

Any idea?

DreiBaer, 2020-02-01:
> 
> 
> Hi, I’ve tried the work around on Buster but it doesn’t appear to work when using MariaDB. Daemon reports ignoring ‘sql_hostname’ and then says it can’t open SQLite dB....

The same to me. Is there an additional file for MySQL/MariaDB users?

DreiBaer, 2020-02-04:
1.
Install the libcurl4 libraries:
`sudo apt-get install libcurl4-openssl-dev`

2.
First install the Maria DB development files.
```
sudo apt-get update
sudo apt install libmariadb-dev
```

3.
Download the source code from SBFspot. e.g.: https://github.com/SBFspot/SBFspot/archive/V3.5.2.zip
```
cd ~
mkdir download
cd download
wget https://github.com/SBFspot/SBFspot/archive/V3.5.2.zip
```

4.
Unzip the source code
`unzip V352.zip`

5.
Change to the directory of the daemon:
`cd downloads / SBFspot-3.5.2 / SBFspotUploadDaemon`
In this file
`nano ../SBFspotUploadCommon/../SBFspot/db_MySQL.h`
the line
`#include <mysql / mysql.h>`
in
`#include <mariadb / mysql.h>`
to change.

6.
Then we build the upload daemon:
`make mariadb`
And install it:
`sudo make install_mariadb
`

7.
Last but not least, the upload service has to be restarted:
`sudo systemctl start SBFspotUpload`

If you want, you can still check whether everything went well.
`sudo systemctl status SBFspotUpload.service`

deagel, 2020-02-16:
Hi,
For me the same. 

> > Hi, I’ve tried the work around on Buster but it doesn’t appear to work when using MariaDB. Daemon reports ignoring ‘sql_hostname’ and then says it can’t open SQLite dB....
> 
> The same to me. Is there an additional file for MySQL/MariaDB users?

Is it possible to create a buser SBFspotUploadDaemon that supporting mariadb / mysql and add it to sbfspot-config?

obduser, 2020-03-30:
> I was able to reproduce the issue and as a workaround I recompiled the daemon for Raspbian buster.
> You can find it here: https://github.com/SBFspot/SBFspot/releases/tag/V3.5.2
> Download [...]

I just switched to buster and had the same problem because I still use V3.5. An upgrade to V3.6 is still pending. Thank you for this solution