SBFspot/SBFspot #296: Where is the missing data?

ot3, 2019-06-07:
I had a Bluetooth failure today on thye connection to one of my inverters so wanted to load the missing data. From the log file below it looks like SBFspot loads the missing data but I can't find it in the database.

I used the following query:
`SELECT * FROM SpotData where TimeStamp >= 1559890800 order by TimeStamp asc LIMIT 120;`
This does not show any data from the relevant period for the relevant inverter.
Where is the data hiding? Am I using the wrong table or the wrong command?

```
pi@eon:[~/smadata]: /usr/local/bin/sbfspot.3/SBFspot -startdate:20190607 -v3 -finq -nocsv -cfgSB3000TL-21.cfg -ad1 -am1
SBFspot V3.5.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -startdate:20190607 -v3 -finq -nocsv -cfgSB3000TL-21.cfg -ad1 -am1
Reading config '/usr/local/bin/sbfspot.3/SB3000TL-21.cfg'
Configuration settings:
BTAddress=00:80:25:A6:1C:A2
Password=<undisclosed>
MIS_Enabled=0
Plantname=Araminta Station - South
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=51.44
Longitude=5.61
Timezone=Europe/Amsterdam
CalculateMissingSpotValues=1
DateTimeFormat=%Y-%m-%d %H:%M
DateFormat=%Y-%m-%d
TimeFormat=%H:%M
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=3600
DecimalPoint=dot
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=1
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=3
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Fri Jun  7 12:48:44 2019: INFO: Starting...
sunrise: 05:23
sunset : 21:49
Connecting to 00:80:25:A6:1C:A2 (1/3)
Initializing...
SUSyID: 125 - SessionID: 952266677 (0x38C26FB5)
SMA netID=03
Serial Nr: 7EF8859F (2130216XXX)
BT Signal=69.4%
Logon OK
Local Host Time: 2019-06-07 12:48
Plant Time     : 2019-06-07 12:48 (+1 sec)
TZ offset      : 3600 sec - DST: On
Last Time Set  : 2019-06-07 05:00
Time was already adjusted today
SUSyID: 138 - SN: 2130216XXX
Device Name:      SN: 2130216XXX
Device Class:     Solar Inverters
Device Type:      SB 3000TL-21
Software Version: 02.55.03.R
Serial number:    2130216XXX
SUSyID: 138 - SN: 2130216XXX
Device Status:      Ok
SUSyID: 138 - SN: 2130216XXX
[... 180 lines cut ...]
2019-06-07 12:10 : 19958.982kWh - 2652.000W
2019-06-07 12:15 : 19959.207kWh - 2700.000W
2019-06-07 12:20 : 19959.429kWh - 2664.000W
2019-06-07 12:25 : 19959.651kWh - 2664.000W
2019-06-07 12:30 : 19959.875kWh - 2688.000W
2019-06-07 12:35 : 19960.100kWh - 2700.000W
2019-06-07 12:40 : 19960.315kWh - 2580.000W
2019-06-07 12:45 : 19960.540kWh - 2700.000W
======
**********************
* ArchiveMonthData() *
**********************
startTime = 5CF25AB0 -> 01/06/2019 13:00:00
SUSyID: 138 - SN: 2130216XXX
2019-06-01 : 19871.735kWh - 21.922kWh
2019-06-02 : 19893.320kWh - 21.585kWh
2019-06-03 : 19906.225kWh - 12.905kWh
2019-06-04 : 19921.058kWh - 14.833kWh
2019-06-05 : 19932.082kWh - 11.024kWh
2019-06-06 : 19951.862kWh - 19.780kWh
======
Reading events: 2019-Jun-01
Fri Jun  7 12:48:47 2019: INFO: Done.
pi@eon:[~/smadata]:
```

sbf-, 2019-06-07:
spotdata table contains the actual data which is not kept in the inverter. So in case of connetion issues this data is lost.
The historical (every 5min) is kept for approx 2 months in the inverter and is stored in daydata table
`SELECT * FROM DayData where TimeStamp >= 1559890800 order by TimeStamp asc LIMIT 120;`
should give you the results

ot3, 2019-06-07:
Many thanks. I had assumed that since the inverter also stores data with 1s resolution, SBFspot would also extract these. Obviously I didn't think this through.

closed 2019-06-07 by ot3

sbf-, 2019-06-07:
The inverter does not store data with 1s resolution, only average data over 5min.

ot3, 2019-06-07:
Mine apparently does: I just downloaded yesterday's "/Data/One-second values/" via the GUI and this file has 58209 records covering the period between  06.06.2019 05:51:11 and 06.06.2019 22:01:20. Looks like one record/second to me. 

I tried to copy a few lines into this message but this crashed Firefox.

MattG40, 2020-06-21:
My Sunny Tripower 4.0 also - would it be possible to get 1s values with SBFspot?