SBFspot/SBFspot #301: Adding more registers

KiavashS, 2019-06-29:
Hi, I'm using the Windows build and have little code or modbus skills. Is there a way to add more of the available inverters registers to the redirected data to the database?

I'm using this on my sunny island as a monitoring system to automate some other tasks. Some of the additional data I'd like are the total PV power, generator power, current consumption by the house and others. I believe the registers are available, but can they be added to the extract for SBFspot?

sbf-, 2019-07-07:
SBFspot is using SMAdata2+ protocol instead of modbus
There is already limited support for battery inverters (see vwBatteryData in the database)

KiavashS, 2019-07-09:
Thanks. I've found that view and it's very useful, but is there possibility to add additional fields? Eg, TotalLoadCurrent, GeneratorCurrent and more?

sbf-, 2019-07-09:
Should be possible, yes.
Do you have time to test? I don't have a battery inverter, so I can't test the code myself.

KiavashS, 2019-07-09:
I'm no good at coding anything but sure I could put time aside to test it thoroughly.

KiavashS, 2019-07-23:
@sbf- any progress? I have the equipment and time to help test and anything else I'm capable of.

sbf-, 2019-07-25:
No, not yet. I don't have much time atm

eagle360, 2020-03-17:
I have an SBS6.0-US I would like to gather additional data from as well.  If you could point me in the right direction to adding additional parameters I would take a crack at it.    If I had a device that talked to the SBS I would just snoop the traffic and see what it is doing but I don't know of any devices that query the SBS for this data using SMAdata2+.  I sent you a recent donation (March 7) and an e-mail about RS-485 Off-Grid control (just today).  You can use that e-mail to contact me.  Thanks.

sbf-, 2020-03-18:
For the SBS6.0 you will need SunnyExplorer (windows only) From there you can trace the traffic between the device and SE
For RS485 you can’t use sbfspot nor SE. Maybe [this](https://github.com/ardexa/sma-rs485-inverters) can help