SBFspot/SBFspot #303: SBFspotUploadDaemon Version 2.0.0 - Uploads no data

JaseSmith78, 2019-07-06:
Raspberry Pi 3
Updated as today (gcc (Raspbian 8.3.0-6+rpi1) 8.3.0)
Fresh download of source and compile today

have set loglevel to debug, no additional data logged 
not sure how to check whats being added to the DB, but csv's are created and contain data
-----
[16:52:14] INFO: SBFspotUploadDaemon Version 2.0.0
[16:52:14] INFO: Starting Daemon...

sbf-, 2019-07-06:
A few things to check:
Is sbfspot running every 5 minutes?
Check the time of your inverter.
Is there data in the db? Use vpvodata to check.
Post the output of sbfspot -v5

JaseSmith78, 2019-07-06:
Yes, the cron job is setup (the csv's are being populated with the expected timestamps)

I'll attach the output and the csv, I'm not sure what you mean by vpvodata though

[debug.txt](https://github.com/SBFspot/SBFspot/files/3365029/debug.txt)
[Number13-20190707.txt](https://github.com/SBFspot/SBFspot/files/3365030/Number13-20190707.txt)

JaseSmith78, 2019-07-06:
I will assume vpvodata is a typo, and i found the below 
I assume this will be the cause - I do at least the the CSV's so i can manually upload 

sqlite> select * from vwPvoData;
Error: database disk image is malformed

sbf-, 2019-07-07:
Oh, you have a corrupt database file...
You can try to [recover](https://www.sysinfotools.com/blog/fix-error-sqlite-database-disk-image-malformed/) it or else start with a complete new one.
Personally, I take a daily backup (zipped) of the file during the night.

JaseSmith78, 2019-07-07:
Would it be possible to use the csv's to add the missing data back in to the db so the upload daemon can upload it (its about 1 1/5 - 2 months worth of data, i could upload the csvs by hand, but not all the captured data will saved that way) - I'm happy to do the work, but can you help with what the mapping between the csv and the sql table, and if i update one row, if i need to update a row in another table?

sbf-, 2019-07-08:
2 months is about the max an SMA inverter keeps in memory... You can retrieve all the historical data, except spot values (AC & DC voltage/current etc)
Use this command:
```
SBFspot -finq -v -ad62 -am2 -ae2
```

closed 2019-07-28 by sbf-