SBFspot/SBFspot #304: Getting data from both Sunny Boy and Sunny Island

KiavashS, 2019-07-09:
Hi,

I can't find any info about this too extracting data from 2 sources. Below indicates that multiple inverters are supported but since each inverter has a different password, how can I include both (1 X Sunny Boy, 1 X Sunny Island)? 

Thanks.

----------------------------------------------------------

# Multiple IP addresses can be provided (comma separated)
IP_Address=192.168.1.22

# User password (default 0000)
Password=0000

# MIS_Enabled (Multi Inverter Support: Default=0 Disabled)
# +------------+-------+-------------+
# | #Inverters | NetID | MIS_Enabled |
# +------------+-------+-------------+
# |      1     |   1   | Don't Care  |
# +------------+-------+-------------+
# |      1     |   >1  |      0      |
# +------------+-------+-------------+
# |      >1    |   >1  |      1      |
# +------------+-------+-------------+
MIS_Enabled=0

KiavashS, 2019-07-09:
Nevermind this issue. Problem is solved.

I tested by changing the password for both inverters to be the same. This now works. Although I'd like to suggest a feature upgrade to allow different passwords if the option is there.

Thnkyou. :)

sbf-, 2019-07-09:
With 2 devices with different passwords you can use 2 config files and run SBFspot twice (use -cfg command line parameter)

closed 2019-07-14 by sbf-