SBFspot/SBFspot #305: SBFspotUploadDaemon[3415]: Unable to write to logfile

obduser, 2019-07-10:
Hello,

When I started the upload daemon, I get the following error message:

`SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-07-10 07:17:52 CEST; 7s ago
 Main PID: 3415 (SBFspotUploadDa)
   CGroup: /system.slice/SBFspotUpload.service
           └─3415 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Jul 10 07:17:52 pvlogger systemd[1]: Started SBFspot Upload Daemon.
Jul 10 07:17:52 pvlogger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190710.log]
Jul 10 07:17:

Can anyone help ?

sbf-, 2019-07-10:
Can you actually write to /mnt/usb/smadata/logs ?

obduser, 2019-07-14:
Hello sbf,

sorry for the late answer.

I do not understand why, but despite the error message, the file is written.

> [15:20:45] INFO: Uploading datapoint: 20190714,15:20,45917652,1416,,,55,234.15 => OK (200)
[15:25:45] INFO: Uploading datapoint: 20190714,15:25,45917803,1812,,,55,235.32 => OK (200)
[15:30:46] INFO: Uploading datapoint: 20190714,15:30,45917962,1908,,,55.1,236.16 => OK (200)
[15:35:45] INFO: Uploading datapoint: 20190714,15:35,45918083,1452,,,55,234.12 => OK (200)
[15:40:45] INFO: Uploading datapoint: 20190714,15:40,45918185,1224,,,55,234.44 => OK (200)
[15:45:45] INFO: Uploading datapoint: 20190714,15:45,45918285,1200,,,55.1,235.02 => OK (200)
[15:50:45] INFO: Uploading datapoint: 20190714,15:50,45918392,1284,,,55.3,232.59 => OK (200)

> Jul 14 15:40:45 logger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190714.log]
Jul 14 15:45:45 logger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190714.log]
Jul 14 15:45:45 logger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190714.log]
Jul 14 15:50:46 logger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190714.log]
Jul 14 15:50:46 logger SBFspotUploadDaemon[3415]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190714.log]

obduser, 2019-07-17:
Hello,

today, i saw the following entry at the syslog:

> SBFspotUploadDaemon[388]: Unable to write to logfile [/mnt/usb/smadata/logs/SBFspotUpload20190717.log]

but still data was written to the log file.

Can you actually write to /mnt/usb/smadata/logs ? --> yes, as user pi

closed 2019-08-11 by sbf-