SBFspot/SBFspot #307: SBFspot upload daemon libcurl compatibility

Hindosh, 2019-07-14:
jul 14 13:48:14 raspberrypidomo systemd[1]: Started SBFspot Upload Daemon.
jul 14 13:48:14 raspberrypidomo SBFspotUploadDaemon[25224]: /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found
jul 14 13:48:14 raspberrypidomo systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
jul 14 13:48:14 raspberrypidomo systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.

TerrorSource, 2019-07-14:
did you try removing libcurl3, installing libcurl4 and re-compile via the manuall install guide?

Hindosh, 2019-07-14:
no used the config installer 
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash

TerrorSource, 2019-07-14:
@Hindosh try the manual guide. it worked for me with the manual guide on a Pi 4 4gb.

sbf-, 2019-07-14:
There is a temporary workaround described [here](https://github.com/SBFspot/SBFspot/issues/295)

Hindosh, 2019-07-14:
"You can find it here: https://github.com/SBFspot/SBFspot/releases/edit/V3.5.2"
seems inaccessible via the browser (404 error)
can you send a directlink to: Download SBFspotUploadDaemon_buster 
thnx

closed 2019-08-11 by sbf-