SBFspot/SBFspot #308: Merge db file

TerrorSource, 2019-07-14:
Hi,

I moved to a new Pi with SBFSpot.
Copied all the csv files (2014-2019) to the new pi but im stuck with the SVFspot.db file.
Old one: 13.791 kb
New one: 904kb

Already run
/usr/local/bin/sbfspot.3/SBFspot -ad300
/usr/local/bin/sbfspot.3/SBFspot -am60
/usr/local/bin/sbfspot.3/SBFspot -ae7

But looks like old data from 300d and older is not picked up and put into the db file.
How can i merge them by hand?

Couldnt find a how-to online.

sbf-, 2019-07-14:
Most inverters only keep 62 days of daydata. The -ad300 is the limit if for some inverters this should be possible.
Just copy the old db to the new pi... simple as that.

TerrorSource, 2019-07-14:
@sbf- Thanks.

Uploaded the db file to the new pi and re-run all 3 command's.
DB file grew from 22.527kb to 22.601kb

closed 2019-07-14 by TerrorSource