SBFspot/SBFspot #310: Historic data (2)

Myoldpasswordbobharris1943, 2019-07-23:
I have installed SBFspot on my Raspberry Pi and successfully recovered data from my SMA Sunny Boy inverter and uploaded it to PVOutput. The SBFspot.db database became corrupted and I am trying to rebuild it.
Using the -ad# option I can recover old daydata, but they appear in an odd sequence in the table. The values within a day are in increasing time order (every 5 minutes from around 06:00 to around 21:30), but the daily records appear in decreasing date order (22 July, 21 July, 20 July ......). Is this correct, or have I perhaps got a setting wrong somewhere?
I would like them to appear as a steadily progressing sequence (but of course with gaps corresponding to zero output at night). 
Please can anybody help? If necessary I can provide more info. Please let me know which.
Bob

sbf-, 2019-07-24:
To sort the data, use [ 'order by' or 'order by desc'](http://www.sqlitetutorial.net/sqlite-order-by/)
``` SQL
SELECT * FROM vwDayData
ORDER BY Timestamp
```

sbf-, 2019-07-24:
Make a backup on a regular base by copying SBFspot.db (during the night, when there is no production, so nothing is written to it)
I do this daily (zipped and rolling backup with date added to the file. Don't overwrite each day the backup because you can overwrite it with a corrupt one)

Myoldpasswordbobharris1943, 2019-07-25:
Thank you for your reply. Upload to PVOutput is not a problem and the data
all seem to fit.
I want to use the DayData column in a Python script though and need to have
the Timestamp values in ascending order. The ORDER BY command does not
alter the order of the Timestamp values (see attached figure).
The steps in the sawtooth correspond to individual days and the values
correspond exactly with the day duration and inverter active periods. The
downward trend shows the reverse order of the days. I'm not sure about the
jumps at the ends.

I have in the meantime solved the problem as far as my Python script is
concerned, by sorting the DayData record:
                  sma.sort(key = Lambda sma: sma[0]).
This also sorts the Timestamp values correctly, but not of course in
SBFspot.db.

I still don't understand whether SBFspot's import of old data is meant to
be in reverse date order or if that is due to my setup in some way.

In any case, thanks for your help.
Bob

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Myoldpasswordbobharris1943, 2019-07-25:
> Yes, when using -ad# option the data is imported in reverse order (today,
> yesterday,...) This is by design but shouldn't cause any issues because it
> can be sorted like you want.
> How and where do you see unsorted data? In which program?
> BTW, I can't see any pictures attached.

Here is the png file again. I see this behaviour both directly in DB
Browser for SQLite and if I read DayData into a Python script and plot
Timestamp vs _rowid_. From what you wrote this is the correct behaviour.
i wonder if there is a way to re-order the Timestamp values in SBFspotdb
itself. Applying your SQL (in DB Browser for SQLite), but replacing
vwDayData with DayData does not seem to do anything.
Bob

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

sbf-, 2019-07-25:
> i wonder if there is a way to re-order the Timestamp values in SBFspotdb
itself.

Sure... this will create a new table DayData_Sorted with all data ordered, most recent at the top

``` SQL
create table DayData_sorted as select * from DayData order by timestamp desc;
```

Myoldpasswordbobharris1943, 2019-07-25:
Worked perfectly. Now I need to learn some SQL to reverse it so the
Timestamp is increasing (I only started with my RPi and inverter a couple
of months ago).
Thanks a lot for your help!
Bob

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Myoldpasswordbobharris1943, 2019-07-25:
Solved. I left out the desc. Obvious once I thought about it.
Bob

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2019-08-03 by sbf-