SBFspot/SBFspot #312: database and CSV files not updating after going from single to multiple inverters

schmidtl4, 2019-07-25:
After adding a new inverter yesterday (going from 1 to 2) I'm observing the following:

In the SQLite database, the new inverter has been added to Inverters table and spot data is being recorded in the spotdata table.  However there are no DayData or MonthData entries for the new inverter.

The CSV files are not being updated any more for either inverter.  The last entry in the {{plantname}}-Spot-20190724.csv file is just before the config file was changed. The config file changes were limited to adding the IP addresses and changing MIS_Enabled from 0 to 1.

Clearly both inverters are connected as evidenced by the Spot data entries, inverter table entries and the results of a command line -finq -v3 inquiry.  Why are some tables and the CSV files not being updated?

sbf-, 2019-07-25:
Did you check the time of your inverters? Requesting data for today when the time is more than a day behind will never work.
If this is OK, create a full debug log (-d5 -v5)
This will generate a lot of data and reveal an error maybe.
BTW, MIS_Enabled is for Bluetooth devices only (with NetID) - In your case it has no effect

schmidtl4, 2019-07-25:
Thanks - I'll check the time on the inverters, but the times recorded in the SpotData table are in sync - i.e. the time stamps for the two inverters are the same and the Pac values are very similar (as they both have the same number of panels connected to them).  Is there logic to say that the spotvalues would have identical timestamps and still suspect that inverter time was the cause of the lack of data in the other tables?

sbf-, 2019-07-26:
Yes, it depends of the  CSV_Spot_TimeSource setting.

schmidtl4, 2019-07-27:
Where would I find that setting?

drdiesel, 2020-06-02:
I am having a similar issue on a fresh install, one inverter is working fine and uploading to pvoutput fine and the other isn't.  This was working fine on a previous install but on the fresh while upgrading to spot3.6 I started having issues.

Working inverter = 1990056838
https://pvoutput.org/intraday.jsp?id=69593&sid=61850&dt=20200602

Not working inverter = 1990056252
https://pvoutput.org/intraday.jsp?id=69593&sid=68277&dt=20200602

Not exactly sure what all to attach, please let me know if further info will help.  I don't see a method to make the upload log more verbose?

Last couple upload log entires:

[10:05:30] INFO: Uploading datapoint: 20200602,10:05,4086900,1944,,,0,124.675 => OK (200)
[10:05:30] INFO: Uploading datapoint: 20200602,10:05,16496625,4092,,,0,125.49 => OK (200)
[10:10:30] INFO: Uploading datapoint: 20200602,10:10,4087066,1992,,,0,124.54 => OK (200)
[10:10:30] INFO: Uploading datapoint: 20200602,10:10,16496982,4284,,,0,125.42 => OK (200)
[10:15:30] INFO: Uploading datapoint: 20200602,10:15,4087237,2052,,,0,124.69 => OK (200)
[10:15:30] INFO: Uploading datapoint: 20200602,10:15,16497353,4452,,,0,125.58 => OK (200)

I have verified the API key and System IDs are all correct.
Thanks for the help

[spotdebugD5V5.log](https://github.com/SBFspot/SBFspot/files/4717648/spotdebugD5V5.log)
[SBFspotcfg.log](https://github.com/SBFspot/SBFspot/files/4717652/SBFspotcfg.log)
[SBFspotUloadcfg.log](https://github.com/SBFspot/SBFspot/files/4717654/SBFspotUloadcfg.log)

sbf-, 2020-06-03:
Inverter: Time is taken from inverter
Computer: Time is taken from host (in case inverter time is wrong)

closed 2020-06-03 by sbf-