SBFspot/SBFspot #313: Bad request 400: Power value [15538XXXX] too high for system size [1212]

fritserasmus, 2019-07-27:
I just need some pointers please.

I know what the error message means, so I just want to edit the record OR just delete the record.

I edited the file:  MyPlant-Spot-20190725.csv by removing the bad record but I still get the error message.

Which records are actually uploaded? What should I delete - Which file should I edit.
Is it perhaps loading data from the SQL database?

Any help will be greatly appreciated

sbf-, 2019-07-27:
All data for PVoutput is coming from SQL database... CSV files are not needed.
Take a look at [this thread](https://github.com/SBFspot/SBFspot/issues/137) for a workaround

fritserasmus, 2019-07-29:
Thanks for the link.
I understand what to do.

Can I connect to SQL Lite with "DB Browser (SQLite)" installed on my Windows PC and then do the necessary on the RPi db?
Would you mind to give me some pointers as I do not know how to connect to the db.
command-line connection is not something I tried before

fritserasmus, 2019-07-29:
[(https://github.com/SBFspot/SBFspot/issues/137#issuecomment-515994667)]

sbf-, 2019-07-29:
`DROP View` is a SQL command, so you have to start SQLite3 first:

```
pi@sbfspot:~/smadata $ sqlite3 SBFspot.db 
SQLite version 3.27.2 2019-02-25 16:06:06
Enter ".help" for usage hints.
sqlite> select * from daydata where power >= 155380000;
sqlite> 
```
155380000 is the too big value
To delete it, execute
```
delete from daydata where power >= 155380000;
```
To change the view, follow the 'drop view' procedure

Take a backup first, just by copying SBFspot.db - Just in case ;-)
Best is outside production hours, or make sure SBFspot and daemon are not writing to it.

fritserasmus, 2019-07-29:
Thanks,

Got it to work.

Thanks for the help.

It uploaded only 763 records July 25th to the middle of July 27th. (Middle July 27 to EOD July 29th still not loaded)
Is there an upload limit circa 760 records, or did it just stop because no power was produced?

sbf-, 2019-07-29:
The only limit I know is 60 requests/hour

fritserasmus, 2019-07-30:
@sbf- ,
Thank you very much for your help.
I did everything like explained to the best of my knowledge, but must have done something wrong.

Updates work intermittent and leave gaps.

I am using ver: 3.31 and Ver 3.4 with enhanced install and config is available.

I decided to rather redo the install.

Is there a way to recover some of the missing records from the current RPi, or shall I just accept the gap and rebuild on a new SD card?

What is your opinion?

sbf-, 2019-07-30:
I have no idea what's happening, but reinstalling is maybe a bit drastic. You will lose all your historical data.
Can you send me the db via wetransfer so i can have a closer look?
Is there something in the log files?

fritserasmus, 2019-07-30:
what email address can I use to sned the files?

sbf-, 2019-07-30:
s.b.f at skynet dot be

fritserasmus, 2019-07-30:
done

fritserasmus, 2019-07-30:
It's getting late here. I will wait another 5 minutes and then log off and check-in tomorrow.

Thanks for the help so far

fritserasmus, 2019-07-30:
[SBFspotUpload20190729.log](https://github.com/SBFspot/SBFspot/files/3448840/SBFspotUpload20190729.log)
The last few entries show where it started uploading again

fritserasmus, 2019-07-30:
[SBFspotUpload20190730.log](https://github.com/SBFspot/SBFspot/files/3448847/SBFspotUpload20190730.log)
And this morning only this

fritserasmus, 2019-07-31:
The GAP is now growing bigger and bigger.
There is no feed into pvoutput.org.

If I rebuild SBFspot on another microSD and keep the old one and have a spare RPi (Wich I have) plus a local backup of the database,
is there a way I can import old records into the new database? (With a SQL query /update)

sbf-, 2019-07-31:
> is there a way I can import old records into the new database? (With a SQL query /update)

Yes, with `.dump` command: http://www.sqlitetutorial.net/sqlite-dump/

You need only the `INSERT INTO ....` commands

fritserasmus, 2019-08-01:
@sbf- 
Given my situation, if you were in my shoes, (limited programming knowledge/noob) would you have also consider re-install and then insert and update the previous database into the new database?

sbf-, 2019-08-01:
I should start trying to understand why it isn't working ;-)

I had a look at your db and all pvoutput flags in daydate are set (meaning the upload was done and uploaddaemon is OK)
It seems SBFspot isn't running every 5 minutes because there is missing data for some days:

```
select distinct date(timestamp,'unixepoch','localtime') AS Date, Serial
from daydata
order by 1 desc

Date		Serial
2019-07-30	2130438237
2019-07-27	2130438358
2019-07-27	2130438237
2019-07-26	2130438358
2019-07-26	2130438237
2019-07-25	2130438358
2019-07-25	2130438237
2019-07-24	2130438358
2019-07-24	2130438237
2019-07-23	2130438358
2019-07-23	2130438237
```

On PVO there is nothing for today nor yesterday, so SBFspot is completely down.
To solve this, you should check why SBFspot isn't running anymore (add some logging with -v5 -d5)
Did you reboot the Pi already? I suppose you did
You can run it to get the missing days (-ad5 is last 5 days):

```
./SBFspot -d5 -v5 -ad5
```

If this runs OK, daemon should start with uploading the missing records (see log)
The spotdata will be missing as it was not recorded in the DB

Running integrity check shows the db is OK
```
pragma integrity_check;
```
Hope this helps

fritserasmus, 2019-08-02:
@sbf- 
Thanks for your reply.
I did not want to just ask without trying.

I was able to confirm the missing data in the database.

I tried to run the SBFSpot to add the missing data, but whatever I tried I got: "No such file or directory"

I thought I just have to find that program.
Not familiar with the commands:
So Help find, man -k find, "https://www.linode.com/docs/tools-reference/tools/find-files-in-linux-using-the-command-line/" I could not find the file and nowhere it runs.
So after about spending 5 hours on this I gave up and have to ask you for instructions as I just do not know .

sbf-, 2019-08-02:
SBFspot is located in /usr/local/bin/sbfspot.3 so either

```
cd /usr/local/bin/sbfspot.3
./SBFspot -d5 -v5 -ad5
```
or
```
/usr/local/bin/sbfspot.3/SBFspot -d5 -v5 -ad5
```

fritserasmus, 2019-08-03:
@sbf- ,

Thank: Success-

A lot of screens and some updates.
Lots of new entries in PVOutput, but a lot is missing still.

I have two inverters and I log each separately and then combine in another.
What is really interesting is :

Only one inverter is uploading to PVoutput.

hcitool scan reports both inverters.

To me, it seems like the SD card causes issues. 

I am going to rebuild SBFSpot from the latest version (New SD card and new RPi - Just to check) and once / if it runs fine, will check in for help to import old data (Currently in the db) I will do the flash and config in production time and switchover non-production hours

sbf-, 2019-08-11:
What’s the status?

commit 527b2a4b (2021-12-11): Fix various issues Bad request 400: Power value too high for system size:
Fix #137
Fix #313
Fix #381
Fix #384
closed 2019-08-19 by sbf-