SBFspot/SBFspot #314: Warning from systemctl status SBFUploadDaemon

freekdk, 2019-08-01:
When performing the command in the title, the following warning is shown:
/etc/systemd/system/SBFspotUpload.service:10: Failed to parse service restart specifier, ignoring: on-succes
This file has been created using the example on the Wiki.

sbf-, 2019-08-01:
On Raspberry Pi? Raspbian Version?

freekdk, 2019-08-01:
Yes on Raspberry Pi 2B, but on openSUSE Tumbleweed.

scotte, 2019-08-01:
Just to be sure - double check that you don't have a typo - is last "s" in "success" missing? From the error message pasted above, that might be the case. The full line should look like:

```
Restart=on-success
```

sbf-, 2019-08-01:
> Failed to parse service restart specifier, ignoring: on-succes

Is that a typo?
It should be `Restart=on-success` with double 's'

freekdk, 2019-08-02:
Indeed, the double 's' was a single one. Now all is OK.

freekdk, 2019-08-02:
Forgot to say: Many thanks.

closed 2019-08-02 by freekdk

sbf-, 2019-08-02:
Geen probleem... soms zit het in details ;-)