SBFspot/SBFspot #318: Higher sampling rate

mahadi, 2019-08-11:
Is it actually possible to increase the sampling rate from e.g. 5 minutes to 10 seconds? Or is this a restriction by the inverters? (They dont provide the data more often)? 

I know from https://www.openhab.org/addons/bindings/smaenergymeter/ that the energy meter uses network broadcasts which can be read. If the inverters use the same technique, would it be possible to sample it in a similar way?

I am not doing any exports to PVOutput.org, but i am storing the values locally for my visualisation.

sbf-, 2019-08-11:
It's a restiction by SBFspot. There is too much overhead for login/logout. To retrieve spot data every 5 à 10 seconds the connection should be left open.
An energy meter does not require a login/logout, it just publishes its data which can be captured by any device on the network.
However, leaving the connection open is already a feature request for quite some time. I'll try give it some priority.

PjotrekSE, 2019-08-12:
I am already running it continuously logged in. It was really simple. But
it seems my inverter only allows 60 querys per hour, so maximum seems to be
once per minute.
I am considering rereading the config file (where other timing could be
stored) by sending SIGHUP, that way I could change the timing according to
the current need, i.e. once per minute when monitoring charging the car,
and once every five minutes at other times.
I already terminate the program by sending SIGTERM.
PjotrekSE

mahadi, 2019-08-12:
Ok, how did you change the config file to achieve that?
I wonder if the ["live" view](https://www.sunnyportal.com/FixedPages/HoManLive.aspx) in SunnyPortal also logs in every 10 seconds to get the current values?

PjotrekSE, 2019-08-12:
@mahadi
That is not only a change to the config file. It involves minor program
additions, and an addition to the configuration code.

mahadi, 2019-08-12:
Would you like to share it? :)

PjotrekSE, 2019-08-12:
@mahadi
Maybe your inverter is different from mine. As I said, mine only allows 60
logins / hour.
I will prepare a patch file for the changes I made. But there is no reread
of the config file - that is, again as I said, not yet implemented.
Don't hold your breath, though, I'm pretty choked as it is.
PjotrekSE

technimad, 2020-05-26:
Implementation seems pretty ok. https://github.com/SBFspot/SBFspot/compare/master...PjotrekSE:feature/runcont 
Any chance this will be merged?

PjotrekSE, 2020-05-26:
> Implementation seems pretty ok. master...PjotrekSE:feature/runcont
> Any chance this will be merged?

I don't think so. Sbfspot rejected my solution(s) and went another route.

technimad, 2020-05-26:
Too bad. @SBFspot what's the route you are taking?

I'm dockerizing this tool, to use in [a project](https://gitlab.com/technimad/emonitor/) to visualise both energy generated by PV and readouts of a smart meter. Higher resolution readings would be very welcome in that project. 

I'd like to contribute back, I'm interested in what would fit the direction of the project.

sbf-, 2020-05-27:
I prefer to have the same features for both windows and linux versions. So, if someone provides a PR developed for linux, someone has to compile/test it for windows. In most cases it's me ;-)
IMO, the aproach of PjotrekSE by changing configuration parameters at runtime is overkill (for 99.99% of the userbase). That's the main reason why his code won't get merged as is.
Also important: writing every 2 à 5 seconds a record to the Spotdata table (instead of every 300 sec) will blow up the SQLite db file.

fasihi01, 2020-06-06:
Hi i've added a parameter "-continuousMqtt" in my fork. I messed up pretty much because of line formatting, the change is not that big. It's working for me :)

https://github.com/fasihi01/SBFspot

PjotrekSE, 2020-06-06:
Well, hope it worx. You must be very careful with the mallocs not to cause
a memory leak.
Have a look at my clone at github.

fasihi01, 2020-06-06:
I have absolutely no experience in programming cpp, to be honest. I just played around until it was working.

Difference between your and my solution: you put half of the program in a loop, while I only want the mqtt-part to run every 5 seconds until someone issues a sigint.

I will have a close look on the stability of my virtual machine and if it starts freaking out, i guess i have created a memory leak :P

PjotrekSE, 2020-06-06:
Correct me if I'm wrong, but if you only loop the mqtt part, will it not
send the same values every time?
Would you not want to get new values from the inverter as well?

fasihi01, 2020-06-06:
You‘re right, i poll new data before the mqtt-call.

technimad, 2020-06-08:
Thanks for your perspective. 

> I prefer to have the same features for both windows and linux versions. So, if someone provides a PR developed for linux, someone has to compile/test it for windows. In most cases it's me ;-)
> IMO, [...]

I can see that multi-platform is important for this project. The change of config parameters by signalling the daemon is a neat trick, yet it seems very much tied to the use case of PjotrekSE. 

I can't help with multi-platform development in this project. But I could try to remove (most of) the signalling from PjotrekSE fork.

PjotrekSE, 2020-06-08:
If that's the case, the simplest way may be to use preprocessor directives
to keep that part in linux, and exclude in all other.
They impose little if any load on the system.

MartinRinas, 2020-06-11:
I'm super interesting in higher intervals as well, I'd like to connect SFBspot to openwb.de - an open source wall box for EV charging. Ideally I'd like to send the current power of the inverter every 10s via MQTT.

fasihi01, 2020-06-11:
My solution is up and running since i wrote the patch. Maybe not elegant, bit dont care.

MartinRinas, 2020-06-27:
@fasihi01 has your fork the latest version? Tried to use yours but ran into issues, there was a typo in a file name which was easy to fix, but I also run into this one. Haven'T had to look into this closer, wonder if you could publish your latest version.
SBFspot.cpp:(.text.startup+0x20c8): undefined reference to `continuousMqtt(InverterData**, Config*)'

PjotrekSE, 2020-06-27:
No, it's an old one.

fasihi01, 2020-06-27:
I‘ll have a look on monday. Sorry for your inconvenience.

fasihi01, 2020-06-27:
I just forcefully pushed my current running version. Please check if it's working for you now.

MartinRinas, 2020-06-28:
yes it does build, thank you so much!

MartinRinas, 2020-06-28:
@fasihi01 build was successulf, I was able to get & publish my values to MQTT. How do I now enable the continuousMQTT mode, tried using -continouosMQTT parameter but that wasn't recognized. I also tried to work out how that would be called but didn't see that either. Is the SBFSpot.cpp & .h not being updated by any chance?

`pi@raspbKeller:/usr/local/bin/sbfspot.3 $ ./SBFspot -finq -nocsv -continuousMqtt
SBFspot V3.6.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2019, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -finq -nocsv -continuousMqtt
Invalid argument: -continuousMqtt
Use -? for help`

MartinRinas, 2020-06-28:
thank you, the updated branch works perfect.

technimad, 2020-06-28:
Do you notice any memory leaks?
I've been using @PjotrekSE 's fork with a 20 sec interval. Output to sqlite. And this setup seems to leak, starting at 6 Mb up to more than 600 Mb at the end of the day.

It doesn't seem to leak every iteration (output copied from running with option -d1):
```
RSS - 6756 kB
    Shared Memory - 4868 kB
    Private Memory - 1888 kB

RSS - 7004 kB
    Shared Memory - 4868 kB
    Private Memory - 2136 kB

RSS - 7268 kB
    Shared Memory - 4868 kB
    Private Memory - 2400 kB

RSS - 7268 kB
    Shared Memory - 4868 kB
    Private Memory - 2400 kB

RSS - 7516 kB
    Shared Memory - 4868 kB
    Private Memory - 2648 kB

RSS - 7516 kB
    Shared Memory - 4868 kB
    Private Memory - 2648 kB
```

MartinRinas, 2020-06-28:
no issues so far, running -nosql variant. Process is running for about an hour, 5MB right now. Will check tomorrow and let you know.

technimad, 2020-06-29:
> Do you notice any memory leaks?
> I've been using @PjotrekSE 's fork with a 20 sec interval. Output to sqlite. And this setup seems to leak, starting at 6 Mb up to more than 600 Mb at the end of the day.

It's about 600Mb in 24 hours. Running for about 12 hours now, with a memory usage of 250MB.

PjotrekSE, 2020-06-29:
I remember now that I found some memory leaks - or not correctly disposed
of memory allocations in the original code. That was of course no problem
there, since that code was only run once, and then finished, closing all
memory allocations. I remember the leak you mention, but was not able to
find it, probably for being in one of the libraries. It all comes to mind
now, I tested as you did, with memory increasing, and then suddenly going
down again, presumably when a library re-initialized itself. I decided I
can live with that.

MartinRinas, 2020-06-29:
no issues for me after running through the night , unless I'm looking at the wrong place:

```
pi@raspbKeller:~ $ cat /proc/17010/smaps | grep -i rss |  awk '{Total+=$2} END {print Total/1024" MB"}'
5.48828 MB
pi@raspbKeller:~ $ cat /proc/17010/smaps | grep -i pss |  awk '{Total+=$2} END {print Total/1024" MB"}'
4.13965 MB
```

sure enough the inverter didn't produce anything during the night, however I do run with -finq so that shouldn't make a difference. Lets see if that changes today.

technimad, 2020-06-29:
> I remember now that I found some memory leaks - or not correctly disposed of memory allocations in the original code. That was of course no problem there, since that code was only run once, and then [...]

I did some digging and it looks like this one line change fixes the leak: https://gitlab.com/technimad/sbfspot-high-resolution/-/compare/feature%2Fruncont...feature%2Fruncont-fix-memoryleak?w=1#369e0cc3de8634da5bbf75ffc16879a2b1a31446_652_652 

I've removed line 652 (please ignore the changes my editor made while fixing the encoding) and now memory usage RSS is around 7Mb after 2 hours of running (20 sec interval, bluetooth, output to sqlite).

sbf-, 2020-06-29:
> I did some digging and it looks like this one line change fixes the leak: [...]

I'm afraid your SQL db functions will not work anymore... but I guess you don't use them.

technimad, 2020-06-29:
> I'm afraid your SQL db functions will not work anymore... but I guess you don't use them.

Data is still written to sqlite after this change. The db variable is defined on line 338: https://gitlab.com/technimad/sbfspot-high-resolution/-/blob/feature/runcont-fix-memoryleak/SBFspot/SBFspot.cpp#L338

sbf-, 2020-06-30:
@technimad 
OK, that's indeed a memory leak... when executing inside a loop it becomes very important.
Thanks for the find/fix

silliewimons, 2021-03-19:
dear users of sbspot,  within the discussiob about the higher sampling rate, ons should keep in mind that:
 -SMA products do use  two communication-systems,
    -a- the operational / production channel, that is used by energymanager / home-manager to control the operation
          of a system,  -  for the need of 'energy-flow' / 'energy-management' that channel must operates at high rate,
    -b- display channel, this is the interaction that is uesd by "sunny-explorer", "sbfspot", . . . .
       !!  if you look in the full-details on the log-file by sbfspot at 'seconds'info, then you will see that the time-stamp associated 
           with the "data-records" retrieved from inverter for: "AC-module"; -"DC-module"; "comm-module"; -. . .  is not exact
           the same,  there is a slight time-difference on "the sample-time"    for the purpose of 'display' - that is no problem,
       !!  the refresh-rate of the information in the display channel, on my older SMA-sunnyBoy is around 5seconds,
       !!  sma-has limited to parallel access by "users" towards the data-channel at TWO

 - the above illustrates that there is no advantage in increasing the "sampling rate" below ~6seconds,
           it helps to understand that SMA-design has limited the number of consecutive interrogations to 60
           developping an 'own' or 'open-source' EnergyManagementSystem is OK, but it should use outputs from EnergyMeter's
           there are several 'open-source' projects for this
   
 - to be clear - i fully support an evolution for SBFspot to include an option for repeated sameples,
          for instance by using the commad-line parameter -sp#  where # is used to indicate the numer of repeats
          where # can have the valuses:  '0';  '1';  '2' up to '60'  in the config file one can add an interval parameter
          in my opinion following limits are usefull: 
          -interval-time:  5seconds should be the smallest interval  (max) 90seconds
          -the combined result of #repeat's and interval should be lower then 59min59sec
   - scripts provide a better tool for all repeat functions above 60seconds
   
kind regards   wim simons

FutureCow, 2022-03-02:
@MartinRinas ; @fasihi01 
Can you update the code to the latest version?

technimad, 2023-04-18:
I’ve been running the high-res port for a few years now, almost without issues. Sources: https://gitlab.com/technimad/sbfspot-high-resolutionContainer: https://gitlab.com/technimad/emonitor/container_registry/1209381 v1.1 is the most recent and stable. Only issue I see is communication is sometimes dropped without warning. Causing no data to be collected. I’ve fixed this by using a healthcheck on the container: https://gitlab.com/technimad/emonitor/-/blob/master/utils/templates/docker-compose.yml.template#L15

Is there any progress for leaving the connection open to achieve higher sampling rates (eg each 10s)? Is there a different solution?

MartinRinas, 2023-04-18:
I've used @fasihi01's fork with -continouosMQTT successfully for some months before I upgraded my SMA inverter with the speedwire (modbus) interface to read data directly.

technimad, 2023-04-18:
> > Container: [https://gitlab.com/technimad/emonitor/container_registry/1209381 v1.1](https://gitlab.com/technimad/emonitor/container_registry/1209381%C2%A0v1.1) is the most recent and stable.
> 
> How do I get a docker container out of this?

```
docker pull registry.gitlab.com/technimad/emonitor/sbfspot-highres:v.1.1
```

evanes68, 2025-02-23:
I have been using @fasihi01 version to continuously send data to mqtt, but have had some issues. Sometimes when there is no more sun the code keeps sending an incorrect value instead of zero. I have patched the code to address this issue. I have also made a Dockerfile to compile an image to run on a Synology NAS with success. I suppose the best option would be to get the code changes into the base repository, there must be other users who would be happy with a live update on a local mqtt server?

sbf-, 2025-03-12:
> I suppose the best option would be to get the code changes into the base repository, there must be other users who would be happy with a live update on a local mqtt server?

If you create a PR I'll merge it in my repository