SBFspot/SBFspot #322: No upload to PVoutput

eterpstra, 2019-08-16:
So I have setup a raspberry pi which was used before for another inverter. I installed it yesterday on a new sma 7000 tl inverter. When i run the upload scripts the status is ok. The log shows only:

SBFspotUploadDaemon
Starting upload..

And that's it.

I can see that the database is filled with data when I run the command manually.

Has this something to do because the pi was first used for another inverter ? I already created a new empty SQL database.

sbf-, 2019-08-16:
Is the inverter time correct?
./SBFspot -v5

eterpstra, 2019-08-16:
How can I check that ? I have to use that command?

eterpstra, 2019-08-16:
Ow...

Current Inverter Time: 19/10/2001 23:04:16                                                                                                                                                                                                
Inverter Wake-Up Time: 19/10/2001 21:01:15                                                                                                                                                                                                
Inverter Sleep Time  : 19/10/2001 23:04:17

sbf-, 2019-08-16:
Try with -settime 
But that doesn't always work. In that case you need sunny explorer

eterpstra, 2019-08-16:
Hmm  what do you mean with -settime? I can change the time with a commandline ?

eterpstra, 2019-08-16:
Ow nevermind. with -settime i am getting this error:

Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'                                                                                                                                                                                     
-settime is only supported for Bluetooth devices

closed 2019-08-21 by sbf-