SBFspot/SBFspot #323: SBFspot and MariaDB/MySQL

rjpost, 2019-08-16:
Hi, i was hoping to find some help. I have installed SBFspot through sbfspot-config on raspberry pi3 with Buster. 
During installation i ran into problems because the database and user weren't created. I managed to resolve this by installing mariadb-server separate, then create the SBFspot database and create a user with access to the database. When running th installer again i simply used the existing config and modified the database name and username accordingly. It all runs through like a charm.

When i run /usr/local/bin/sbfspot.3/SBFspot -v
i get the correct response and results ( i think)
----------response----------------
SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Aug 16 09:47:56 2019: INFO: Starting...
sunrise: 06:25
sunset : 21:03
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 827491857 (0x31528611)
Inverter IP address: 192.168.1.108 from SBFspot.cfg
Logon OK
SUSyID: 409 - SN: 3004182771
Device Name:      SN: 3004182771
Device Class:     Solar Inverters
Device Type:      Sunny Boy 3.6 AV-41
Software Version: 01.01.32.R
Serial number:    3004182771
SUSyID: 409 - SN: 3004182771
Device Status:      Ok
SUSyID: 409 - SN: 3004182771
Device Temperature: 0.0°C
SUSyID: 409 - SN: 3004182771
GridRelay Status:      Closed
SUSyID: 409 - SN: 3004182771
Pac max phase 1: 3680W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 409 - SN: 3004182771
Energy Production:
        EToday: 0.000kWh
        ETotal: 122.133kWh
        Operation Time: 125.20h
        Feed-In Time  : 119.73h
SUSyID: 409 - SN: 3004182771
DC Spot Data:
        String 1 Pdc:   0.480kW - Udc: 297.94V - Idc:  1.607A
        String 2 Pdc:   0.535kW - Udc: 370.19V - Idc:  1.448A
SUSyID: 409 - SN: 3004182771
AC Spot Data:
        Phase 1 Pac :   0.929kW - Uac: 236.78V - Iac:  4.012A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.929kW
SUSyID: 409 - SN: 3004182771
Grid Freq. : 49.99Hz
SUSyID: 409 - SN: 3004182771
Current Inverter Time: 16/08/2019 09:47:56
Inverter Wake-Up Time: 16/08/2019 06:15:47
Inverter Sleep Time  : 16/08/2019 09:47:56
ExportSpotDataToCSV()
[2019-08-16 09:47:57.552] Error: Can't open SQLite db [/VAR/lib/mysql/SBFspot]: 'unable to open database file'
********************
* ArchiveDayData() *
********************
startTime = 5D55D5E0 -> 16/08/2019 00:00:00
ExportDayDataToCSV()
**********************
* ArchiveMonthData() *
**********************
startTime = 5D42C630 -> 01/08/2019 13:00:00
ExportMonthDataToCSV()
Reading events: 2019-Aug-01
ExportEventsToCSV()
Fri Aug 16 09:47:58 2019: INFO: Done.

----end repsonse -----

However i get a Can't open SQLite db error. Which i find strange because i ran the config with option for mariadb. I tried changing SBFspot.cfg so it point to the database location: /var/lib/mysql/SBFspot.

this doenst work. At this time i do see filled csv files with data in smdata folder so it all does appear to work however the mysql database still remains emtpy.

Could anyone help me out to see what im missing / doing wrong. 

all help is appreciated .

ronald

sbf-, 2019-08-16:
That is because the MySQL config is not complete. Missing hostname / user / password
You have only dbname configured

sbf-, 2019-08-16:
Sorry, I reviewed your issue and now I see this:
Compiled for Linux (LE) 32 bit with SQLite support
So you are using the binary for sqlite.

rjpost, 2019-08-16:
Okay but what am i doing wrong then. I used the sbfspot-config  to install and selected the mariadb option in the configuration during the condif process. How can it then have been compiled with the binary for SQL?  Does sbfspot-config support installing SQLite? how should i move forward when i want this to run with mysql?

Thanks for the quick response.

sbf-, 2019-08-16:
Don’t know what’s going on. I should find some time to retest it...
You can manually download and untar the files from
https://github.com/SBFspot/SBFspot/releases/download/V3.5.2/sbfspot-mysql-arm-linux.tar.gz
Btw mysql and mariadb use the same executable.

23x7x365, 2020-04-05:
I have exactly the same problem and I can't find sufficient information in the documentation to be able to resolve this. SBFSpotUpload doesn't work with MySQL/MariaDB when it's been installed with the config tool and there isn't sufficient information to guide me through compiling my own :-(

sbf-, 2021-01-11:
Transferring this issue to sbfspot-config