SBFspot/SBFspot #324: Is it possible to upload 3 inverters to PVoutput.

fonske, 2019-08-16:
I succesfully read out 3 inverters, with SBFspot.

Only problem i have is when uploading to PVoutput, only the 2 inverters are uploaded.
And not the third.
Serial nr: 1930044415 and 57091

################################
### PVoutput Upload Settings ###
################################
#PVoutput_SID (PVoutput_System_ID)
#Map inverters to PVoutput System ID's
#PVoutput_SID=SerialNmbrInverter_1:PVoutput_System_ID_1,SerialNmbrInverter_2:PVoutput_System_ID_2
#e.g. PVoutput_SID=200212345:4321
PVoutput_SID=1930144727:66127,1930121218:66128,1930044415:57091

Maybe it is not implemented in SBFspot?
I use now the (new) version with mqtt support (thank you for that!)
That all works fine, except the third inverter upload.

sbf-, 2019-08-16:
No, this is not a limitation of sbfspot. Are you sure about serial numbers and system ids?  
What is in the daemon logging?

fonske, 2019-08-17:
Now working.
I had to reload the SBFspotUpload service.

sbf-, 2019-08-17:
Never forget the obvious 😀

closed 2019-08-17 by sbf-