SBFspot/SBFspot #325: How to import missing records to SBFspot db

ArnoldSchw, 2019-08-20:
SBFspot is successfully running more than a year. All data is visible at PVoutput.com, and I am quite satisfied. A very reliable solution.
Until two weeks ago.
I accidentally unplugged the Raspberry Pi before we went on holiday. After returning I just plugged it in, assuming it would catch up with with the inverter.
But to my surprise it did not.
So i did this to manually sync: `/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad30`
And that worked fine. After some time PVoutput was uptodate again, _almost_.
Only a part of the data of august 2 was missing, exactly from the moment it was unplugged.
How can i force it to reload all data from august 2 of the inverter? (so now 18 days ago)
Any help would be appreciated!

sbf-, 2019-08-21:
vwspotdata has no records when your RPi is off. This data is not stored by the inverter.
Try vwPvoData instead:

`select * from vwpvodata where DATE( Timestamp ) = '2019-08-02';`

ArnoldSchw, 2019-08-22:
yeah, vwpvodata does contain the records. Then I think the reason must be the PVoutput.com uploadlimit of 14 day. I restarted the RPI on aug the 16th, so 2nd of august was outside the allowed date-range....

closed 2019-08-22 by ArnoldSchw

sbf-, 2019-08-22:
You can always add generation for that day via https://pvoutput.org/addoutput.jsp
Only live data is limited.