SBFspot/SBFspot #326: Reading archived Spot Values not working properly

Gunhawk007, 2019-08-26:
Hi,
I can successfully read the archived months data providing i.e. the -am300 switch which gives me all monthly data untile installation in March 2013. What I am missing is the archived daily data using i.e. -ad300  which does not properly work for me. 
I also think that the range 0..300 for day data is insufficiant as it gives only the last 300 days, which is even not a full year. I have no problems to obtain historically day way beyond that date until installation in March 2013 usig the PC Software, but like to automate this using a Pi.
Did I missed something?

sbf-, 2019-08-26:
Monthdata is 1 value for each day of a month. These values are kept in inverter memory forever. Well, forever, it’s only a guess, I don’t know the limit.
Daydata is 1 value every 5 minutes. These values are kept in memory for about 2 months.
So what you see in sunny explorer is actually month data in sbfspot.
Everything is working fine.

Gunhawk007, 2019-08-27:
Hi,
as mentioned I don't have any problems to obtain historic daily data well beyond the 2 months i.e. up to my installation date in 2013. I can retrieve those values using the Sunny Explorer PC software w/o any probs. However it might be specific to my inverter SunnyBoy 5000

sbf-, 2019-08-27:
Sunny Explorer caches the data. The inverter keeps 62 days in memory.
You can try with startdate argument of sbfspot to get the older data, but i'm pretty sure there aren't any.     
See also sbfspot -?

Gunhawk007, 2019-08-27:
So in order to obtain the values for the last 5 days up to a specific time one would issue the command './SBFSpot -ad5 -starttime:20190701' which will result in downloading the day data for June 26th to July 1st provided that the inverter has cached those values? I'll give it a try, but as said I could access and download the spot data with 5min interval from the inverter using the Sunny-Explorer tool. So if the tool can obtain those data, the inverter should have it cached or not? In this case I still see this as a bug/issue in SBFSpot.

sbf-, 2019-08-27:
> './SBFSpot -ad5 -starttime:20190701' which will result in downloading the day data for June 26th to July 1st provided that the inverter has cached those values

Correct

> I still see this as a bug/issue in SBFSpot.

Rather a limitation that can be undone by recompiling SBFspot after changing SBFspot.cpp
`#define MAX_CFG_AD 300	// Days`

closed 2019-10-04 by sbf-