SBFspot/SBFspot #327: unknown inverter: Sunny Boy Storage 5.0 (SBS5.0-10) & SUNNY TRIPOWER 10.0 (STP10.0-3AV-40)

joanwa, 2019-09-02:
got two more unknowns: One PV and one Battery inverter

SUSyID: 361 on 192.168.7.121: Sunny Boy Storage 5.0 (SBS5.0-10)
SUSyID: 408 on 192.168.7.123: SUNNY TRIPOWER 10.0 (STP10.0-3AV-40)

sma@LXCNAME:~$ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Mon Sep  2 22:56:21 2019: INFO: Starting...
sunrise: 04:32
sunset : 17:59
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 973985127
Inverter IP address: 192.168.7.121 from SBFspot.cfg
Inverter replied: 192.168.7.121 SUSyID: 408 - Serial: 3003925xxx
Inverter IP address: 192.168.7.123 from SBFspot.cfg
Inverter replied: 192.168.7.123 SUSyID: 361 - Serial: 3005963xxx
Logon OK
Unknown Inverter Type. Report this issue at https://github.com/SBFspot/SBFspot/issues with following info:
0x00002484 and Inverter Type=<Fill in the exact type> (e.g. SB1300TL-10)
Unknown Inverter Type. Report this issue at https://github.com/SBFspot/SBFspot/issues with following info:
0x0000248E and Inverter Type=<Fill in the exact type> (e.g. SB1300TL-10)
SUSyID: 408 - SN: 3003925xxx
Device Name:      SN: 3003925xxx
Device Class:     Solar Inverters
Device Type:      UNKNOWN TYPE
Software Version: 01.01.18.R
Serial number:    3003925xxx
SUSyID: 361 - SN: 3005963xxx
Device Name:      SN: 3005963xxx
Device Class:     Battery Inverter
Device Type:      UNKNOWN TYPE
Software Version: 01.50.14.R
Serial number:    3005963xxx
SUSyID: 361 - SN: 3005963xxx
Batt. Charging Status: 71%
SUSyID: 361 - SN: 3005963xxx
Batt. Temperature: 18.0°C
Batt. Voltage    : 370.24V
Batt. Current    : 0.406A
SUSyID: 361 - SN: 3005963xxx
Grid Power Out : 0W
Grid Power In  : 0W
SUSyID: 408 - SN: 3003925xxx
Device Status:      Ok
SUSyID: 361 - SN: 3005963xxx
Device Status:      Ok
SUSyID: 408 - SN: 3003925xxx
Device Temperature: 0.0°C
SUSyID: 361 - SN: 3005963xxx
Device Temperature: 0.0°C
SUSyID: 408 - SN: 3003925xxx
GridRelay Status:      Information not available
SUSyID: 408 - SN: 3003925xxx
Pac max phase 1: 10000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 361 - SN: 3005963xxx
Pac max phase 1: 5000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 408 - SN: 3003925xxx
Energy Production:
	EToday: 0.000kWh
	ETotal: 674.793kWh
	Operation Time: 212.81h
	Feed-In Time  : 206.84h
SUSyID: 361 - SN: 3005963xxx
Energy Production:
	EToday: 0.000kWh
	ETotal: 57.032kWh
	Operation Time: 349.01h
	Feed-In Time  : 285.80h
SUSyID: 408 - SN: 3003925xxx
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
SUSyID: 361 - SN: 3005963xxx
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
SUSyID: 408 - SN: 3003925xxx
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.000kW
SUSyID: 361 - SN: 3005963xxx
AC Spot Data:
	Phase 1 Pac :   0.114kW - Uac: 236.69V - Iac:  0.698A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.114kW
SUSyID: 408 - SN: 3003925xxx
Grid Freq. : 0.00Hz
SUSyID: 361 - SN: 3005963xxx
Grid Freq. : 50.01Hz
SUSyID: 408 - SN: 3003925xxx
Current Inverter Time: 02.09.2019 18:12:10
Inverter Wake-Up Time: 02.09.2019 18:12:08
Inverter Sleep Time  : 02.09.2019 22:56:30
SUSyID: 361 - SN: 3005963xxx
Current Inverter Time: 02.09.2019 22:57:48
Inverter Wake-Up Time: 02.09.2019 08:43:47
Inverter Sleep Time  : 02.09.2019 22:57:49
********************
* ArchiveDayData() *
********************
startTime = 5D6C5B80 -> 02/09/2019 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 5D6BB2C0 -> 01/09/2019 12:00:00
Reading events: 2019-Sep-01
Mon Sep  2 22:56:22 2019: INFO: Done.

joanwa, 2019-09-03:
Also, for the devices, the temperature is not read:
Device Temperature: 0.0°C
Not sure if this temperature sensors are supported for all devices and are supposed to work

sbf-, 2019-09-03:
Thanks, i'll add them asap.
Not every inverter reports its temperature via smadata2+ protocol. Maybe via  modus?

zaziki23, 2019-10-05:
exactly the same issue for me.

is there any svn or development branch which i can check out in order to get a fix?

commit 6678265d (2019-10-08): Fixed Could not open file C\?\C:\PathToConfig\SBFspot.cfg
Fixed #327
Fixed #328
closed 2019-10-08 by sbf-