SBFspot/SBFspot #332: updating from older version without data loss

fshumeli, 2019-10-01:
Hi,

Sorry to use this "issues" section on GitHub, but I couldn't find any SBFspot forum around. (let me know if there's one).
I lack experience to dare updating the SBFspot version without asking first. I can follow instructions but have no coding skills.

I do have SBFspot **V3.3.1** up and running on Pi 3 Model B (Raspbian Stretch). It's working like a charm for over a year. I'm only using it to store data locally, I'm not uploading to PVoutput. 

My issue is how to update to SBFspot 3.5.2 (_latest to this day I think_) without loosing configuration and data, and possibly without changing directory locations. (I know logs might switch from /home/pi/smadata/logs to /var/log/sbfspot.3/ , but I can deal with that).

Also, might there be an automatic update feature in later versions?

Thanks a lot.

_PS: funny how the example Longitude and Latitude in the config match my home... so cheers from Uccle, Brussels ;-)_

sbf-, 2019-10-06:
Hi,
Best time to upgrade is outside production hours (after sunset) to have a stable database.
Before starting the upgrade, make a backup copy of SBFspot.db and SBFspot.cfg
You can use [sbfspot-config](https://github.com/SBFspot/sbfspot-config) and select your existing SBFspot.cfg
If everything goes well, you should have a working installation with the latest version.
But remember: backup your database! You should do this on a regular base, especially when you don't upload to PVOutput

> Also, might there be an automatic update feature in later versions?
Nice feature, maybe in the future.

fshumeli, 2019-10-06:
thanks for the tips.
Just to be sure, there is no specific update procedure right now? 
I have to actually delete the current version and install the new version. 
Keeping DB and CFG files allows to, respectively, avoid losing previous data and configuring SBFspot again.
Am I right, or is there a simpler way ?

thanks !

sbf-, 2019-10-07:
> I have to actually delete the current version and install the new version.

No, you don't have to remove the current version. New version will be installed over the current one.

fshumeli, 2019-10-09:
I successfully upgraded to version 3.6.0, and both DB and CFG files are restored. The database has been updated, and there's no data loss ! :-)

However I must have missed something, because **crontab** returns the following error every 5 minutes : 

_/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/local/bin/sbfspot.3/SBFspot)
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/local/bin/sbfspot.3/SBFspot)_

Can I fix this ?

Also, while testing I noticed **sbfspot-config** returns a similar error : 

_SBFspot Configuration Tool V1.0.0
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
Checking tar... Installed.
Checking wget... Installed.
architecture=armhf
Raspbian GNU/Linux 8.0 (jessie)
username=pi
userhome=/home/pi
Reading variables...
Reading defaults from /var/tmp/sbfspot.3/sbfspot.defaults... Done
Searching for existing configuration(s)...
Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
SBFspot.db exists
Copying files...
Configuring SBFspot... Done
Running SBFspot...
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/local/bin/sbfspot.3/SBFspot)
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/local/bin/sbfspot.3/SBFspot)_

Thanks

sbf-, 2019-10-10:
sbfspot-config is not yet ready for V3.6.0 - I need to make it MQTT ready
For you it doesn't make any difference. V3.5.2 is perfect

fshumeli, 2019-10-11:
Can't explain yesterday's behavior, when sometimes it was returning an error... but it's gone.
It's working well and updating every 5 minutes.
I'll stick to 3.5.2 for now.

My ultimate goal is achieved, since I needed SBFspot running smoothly for my personal project: I have now added a few lines to the daydata bash script to extract the latest production from the SQLite DB and send it to my Fibaro home controller in a virtual device to display "live" production. 

If anyone is interested, I used the following query :

`production=$(sqlite3 /home/pi/smadata/SBFspot.db "select EToday from SpotData ORDER BY TimeStamp DESC LIMIT 1;")`

I don't think there's a faster or cleaner way to get it.

sbf-, 2019-10-14:
> I don't think there's a faster or cleaner way to get it.

Yes there is:
`SELECT EToday from Inverters WHERE Serial=<your serial>;`

closed 2019-10-15 by sbf-